openwrt/openwrt #13292: High CPU usage when speedtest.

butinhi, 2023-08-15:
### Describe the bug

[image]
CPU usage is very high on latest build ( master and 23.05). I turned software and hardware offloading in firewall but not solved

### OpenWrt version

Snapshot

### OpenWrt target/subtarget

Snapshot

### Device

Xiaomi AX6d

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

butinhi, 2023-08-15:
I changed to lede with turboacc and can solve my problem
[image]

tapper82, 2023-08-15:
Lead is old and what is turboacc?

brada4, 2023-08-15:
This is turboacc: https://github.com/fullcone-nat-nftables/nft-fullcone

brada4, 2023-08-15:
First things first - please try real OpenWRT
Snapshot release for your router:
https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=mediatek%2Fmt7622&id=xiaomi_redmi-router-ax6s
Next release 23.05 will support it.

And edit versions to reflect it installed. If indeed regression is in immortalwrt and not OpenWRT, then they will take on it once you report via correct bug tracker.

tomplast, 2024-01-12:
Well I have the same problem, whenever I do a speed test on some website to see how fast I can go my CPU get's maxed out and my SSH connection to the route freezes.

Model: Archer C7 V2
Firmware: Powered by [LuCI openwrt-23.05 branch (git-23.306.39416-c86c256)

tomplast, 2024-01-12:
UPDATE: I just tested with 23.05.2, same problem there.

Something interesting though is I went from around 200 Mbit/s to 450 Mbit/s while doing iperf3 test.

brada4, 2024-01-12:
@tomplast 
If you have PPPoE uplink there is already issue #10224 on that. Try advice around thread, and post there if it helps (not opening duplicate issue)

If not - please open a new issue, this particular one is about fork and aarch64, not 100% match with yours.
Run waveform test without offload, with offload and with hw offload like:
https://www.waveform.com/tools/bufferbloat?test-id=40c15923-382c-4d61-8a7e-c17433ccfef4
As you observed - config/network and config/firewall are relevant, you can highlight manual changes or attach existing files with MAC-s and IPs partially masked.