openwrt/openwrt #13293: e2fsprogs[Ubuntu 22.04.3 LTS] failed

digictechsleepy, 2023-08-15:
### Describe the bug

make[3]: Entering directory '/home/auto/OpenWrtHome/openwrt-cn-acc-gitopenwrtorg-openwrt/openwrt-22.03/tools/e2fsprogs'
. /home/auto/OpenWrtHome/openwrt-cn-acc-gitopenwrtorg-openwrt/openwrt-22.03/include/shell.sh; xzcat /home/auto/OpenWrtHome/openwrt-cn-acc-gitopenwrtorg-openwrt/openwrt-22.03/dl/e2fsprogs-1.46.4.tar.xz | tar -C /home/auto/OpenWrtHome/openwrt-cn-acc-gitopenwrtorg-openwrt/openwrt-22.03/build_dir/host/e2fsprogs-1.46.4/.. -xf -
tar: e2fsprogs-1.46.4/RELEASE-NOTES: Cannot utime: No such file or directory
tar: Exiting with failure status due to previous errors
make[3]: *** [Makefile:57: /home/auto/OpenWrtHome/openwrt-cn-acc-gitopenwrtorg-openwrt/openwrt-22.03/build_dir/host/e2fsprogs-1.46.4/.prepared5ad4658be0bc669b5c4d7f7e7ec7da8b_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory '/home/auto/OpenWrtHome/openwrt-cn-acc-gitopenwrtorg-openwrt/openwrt-22.03/tools/e2fsprogs'
time: tools/e2fsprogs/compile#1.46#2.15#17.68
    ERROR: tools/e2fsprogs failed to build.
make[2]: *** [tools/Makefile:168: tools/e2fsprogs/compile] Error 1
make[2]: Leaving directory '/home/auto/OpenWrtHome/openwrt-cn-acc-gitopenwrtorg-openwrt/openwrt-22.03'
make[1]: *** [tools/Makefile:164: /home/auto/OpenWrtHome/openwrt-cn-acc-gitopenwrtorg-openwrt/openwrt-22.03/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyynyyyyyyyyyyyyyyyyyyynynyyyyyyyy] Error 2
make[1]: Leaving directory '/home/auto/OpenWrtHome/openwrt-cn-acc-gitopenwrtorg-openwrt/openwrt-22.03'
make: *** [/home/auto/OpenWrtHome/openwrt-cn-acc-gitopenwrtorg-openwrt/openwrt-22.03/include/toplevel.mk:230: world] Error 2
auto@auto-virtual-machine:~/OpenWrtHome/openwrt-cn-acc-gitopenwrtorg-openwrt/openwrt-22.03$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy
auto@auto-virtual-machine:~/OpenWrtHome/openwrt-cn-acc-gitopenwrtorg-openwrt/openwrt-22.03$ git log -1
commit de29f15af173e9434d11a00ffcf437bd6bc97727 (HEAD -> openwrt-22.03, origin/openwrt-22.03)

### OpenWrt version

branch openwrt-22.03

### OpenWrt target/subtarget

default

### Device

NA

### Image kind

Self-built image

### Steps to reproduce

   32  ./scripts/feeds update -a
   34  ./scripts/feeds install -a
   36  make defconfig
   37  make menuconfig
   38  make -j1 V=s menuconfig
   39  make -j32
   40  make tools/e2fsprogs/compile

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-15:
Is it some network filesystem?

digictechsleepy, 2023-08-16:
> it some network filesystem?

yes , OpenWrtHome is mounted from another vm with sshfs

brada4, 2023-08-16:
fuse usually lacks locking semantics and inode cache coherence. If you are very lucky 1-job make may complete.