openwrt/openwrt #13299: Routerstation not getting its MAC address

MarkTerMors, 2023-08-16:
### Describe the bug

When booting up 22.03.5 on Ubiquiti Routerstation (ath79) the device does not load its MAC address
[    0.997422] ag71xx 19000000.eth: invalid MAC address, using random address

### OpenWrt version

OpenWrt 22.03.5, r20134-5f15225c1e

### OpenWrt target/subtarget

ath79/generic

### Device

Ubiquiti RouterStation

### Image kind

Official downloaded image

### Steps to reproduce

Happens at boot.

### Actual behaviour

A random MAC is used.

### Expected behaviour

The MAC stored in the device flash should be used. 

### Additional info

I can find the MAC by doing 

fconfig -l -d /dev/mtd7
or
hexdump -x -s 23 -n 6 /dev/mtd7

but have no idea how to go about making the kernel find it there.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

tmn505, 2023-08-16:
The ag71xx has no capability to fetch MAC from that location (that's why the message in kernel log), therefore userspace fallback is in place. To see it run `grep -A 4 'routerstation|' /etc/board.d/02_network` on the router. This fallback sets the MAC for lan and wan interfaces, so You should have `eth0` MAC set from `ar7100_esa` value while `br-lan` has `ar7100_esa + 1`. Is that not the case in Your device?

MarkTerMors, 2023-08-17:
No, that doesn't seem to work. It looks like I'm getting random MAC addresses. Probably because I don't have "lan" and "wan" interfaces. Changing "br-lan" to "wan" doesn't provide a solution.

root@Routerstation:~# grep -A 4 'routerstation|' /etc/board.d/02_network 
	ubnt,routerstation|\
	ubnt,routerstation-pro)
		wan_mac=$(fconfig -s -r -d $(find_mtd_part "RedBoot config") -n ar7100_esa)
		lan_mac=$(macaddr_add "$wan_mac" 1)
		;;
root@Routerstation:~# fconfig -l -d /dev/mtd7
Low verbosity messages are printed.
ar7100_esa: 00:15:6d:c1:2c:a5
boot_script: TRUE
boot_script_data: fis load -d -e kernel
exec

boot_script_timeout: 1
bootp: FALSE
bootp_my_gateway_ip: 0.0.0.0
bootp_my_ip: 192.168.1.20
bootp_my_ip_mask: 255.255.255.0
bootp_server_ip: 192.168.1.42
console_baud_rate: 115200
gdb_port: 9000
info_console_force: FALSE
info_console_number: 0
net_debug: FALSE
root@Routerstation:~# ifconfig
br-lan    Link encap:Ethernet  HWaddr C2:18:E4:A4:A8:06  
          inet addr:192.168.0.36  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::c018:e4ff:fea4:a806/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:84631 errors:0 dropped:0 overruns:0 frame:0
          TX packets:18734 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:11440642 (10.9 MiB)  TX bytes:7812481 (7.4 MiB)

eth0      Link encap:Ethernet  HWaddr C2:18:E4:A4:A8:06  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:84654 errors:0 dropped:0 overruns:0 frame:0
          TX packets:18734 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:12627000 (12.0 MiB)  TX bytes:7812481 (7.4 MiB)
          Interrupt:4 

eth1      Link encap:Ethernet  HWaddr 9A:75:60:6D:0C:FE  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:70260 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:10122194 (9.6 MiB)
          Interrupt:5 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:996 errors:0 dropped:0 overruns:0 frame:0
          TX packets:996 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:76371 (74.5 KiB)  TX bytes:76371 (74.5 KiB)

root@Routerstation:~# cat /etc/config/network

config interface 'loopback'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'
	option device 'lo'

config globals 'globals'
	option ula_prefix 'fd35:713c:55d8::/48'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.0.36'
	option netmask '255.255.255.0'
	option gateway '192.168.0.1'
	option broadcast '192.168.0.255'
	list dns '192.168.0.1'
	list dns_search 'itbox.co.za'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'
	list ports 'eth1'

root@Routerstation:~#

tmn505, 2023-08-17:
I assume You've never reset the devices settings on sysupgrade, which is mandatory when going from `ar71xx` to `ath79`, because `/etc/board.d/02_network` is only sourced when `/etc/config/network` is created. Try to reset the board running `firstboot -y && reboot` (backup Your setting), and check MAC address after reboot. If You don't want to reset settings, set `macaddr` option in device section for each ethX, it should look like this in Your case:
```
config device
        option name 'eth0'
        option macaddr '00:15:6d:c1:2c:a5'
```

MarkTerMors, 2023-08-17:
Not sure about the reset. I tried many things which involved reset and force. Regardless, it fine now. Many thanks.

tmn505, 2023-08-18:
So I guess You can close the issue, as this is more of an support question rather than a bug.

closed 2023-08-19 by MarkTerMors