openwrt/openwrt #13306: Upgrade DSA, IOT devices will not authenticate

timfoley66, 2023-08-16:
### Describe the bug

After upgrade, some IOT devices will not connect to wifi if any security is turned on.  If network is left open, all devices connect successfully

### OpenWrt version

r23228-cd17d8df2a

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1200AC

### Image kind

Official downloaded image

### Steps to reproduce

Install pre-DSA firmware
Set up vlans on switch
Set up wireless on vlan - security WPA2-PSK
Connect iRobot Roomba 
Verify Roomba working

Upgrade to DSA firmware 
Set up vlans on br-lan
Set up wireless on vlan - security WPA2-PSK
Attempt to connect iRobot Roomba
Returns error code BROB25 to iRobot app

Set security to Open Network
Connect iRobot Roomba
Verify Roomba working

### Actual behaviour

iRobot Roomba s9+ returns error code BROB25 to iRobot app.

### Expected behaviour

Roomba to successfully connect to wifi and authenticate

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

neheb, 2023-08-17:
ping @jbsky

brada4, 2023-08-17:
You did not mention DHCP aka dnsmasq in your "steps"

timfoley66, 2023-08-17:
DHCP/DNS are provided external to the AP.  DHCP comes from my PFSense FW.  DNS is provided by Pihole.  All WiFi interfaces are configured as unmanaged with IPv6 disabled.

brada4, 2023-08-18:
Seems reasonable setup now.
First `logread -e MAC` on openwrt - does it authenticate? 
If it does - does it acquire lease on a DHCP server, if not - check tcpdump on bridge, then on port supposedly connected towards DHCP server. DNS as provided to everybody by PFsense should be fine as it always used to be.

e9hack, 2023-08-19:
Can you show the output of  ```bridge vlan``` and ```brctl show```. It looks like that you doesn't use dynamic vlan tagging via hostapd. This does not work. Hostapd assigns a wrong vlan id to the generated interface.

timfoley66, 2023-08-20:
When security is set to No Encryption (open network):
Sun Aug 20 07:03:14 2023 daemon.info hostapd: 40_IOT: STA 50:14:79:fc:81:ce IEEE 802.11: authenticated
Sun Aug 20 07:03:14 2023 daemon.info hostapd: 40_IOT: STA 50:14:79:fc:81:ce IEEE 802.11: associated (aid 3)
Sun Aug 20 07:03:14 2023 daemon.notice hostapd: 40_IOT: AP-STA-CONNECTED 50:14:79:fc:81:ce auth_alg=open

Nothing shows in the log when I enable WPA2-PSK.

`bridge vlan` - not found.  `find / -name bridge` only shows items in /sys/devices/

brctl show:
bridge name     bridge id               STP enabled     interfaces
br-lan          7fff.00259c13bb07       no              10_CORE
                                                        20_GAMER
                                                        30_STREAMS
                                                        40_IOT
                                                        50_GUEST
                                                        CORE_5G
                                                        lan1
                                                        lan2
                                                        lan3
                                                        lan4

brada4, 2023-08-20:
Roomba may need tkip vs (5y) recent default of ccmp only, may barf on 11w option even available. If you get pre-dsa logread you might be able to derive protocol corners it drives. Their support over freephone or website chat should help interpret 'rob' messages.