openwrt/openwrt #13309: Mediatek/Filogic Cudy WR3000: STA seen attached even when out of range for a long time (hours)

ctolzane, 2023-08-17:
### Describe the bug

2 Cudy WR3000 are connected via a WDS link (one master, one client). Both are also configured as AP for 2.4G and 5G.

Depending on the signal level STA are handled either by AP1 or AP2.

Issue: after being managed by AP1 then by AP2 (or vice versa) STA are always seen registred on both AP at the same time. Worse, even if the STA is out of range for both AP, STA is always seen as connected.
Example of what can be seen on both AP with iwinfo:

root@OpenWrt:~# iwinfo phy0-ap0 assoclist
E6:17:3B:C5:06:F9  -97 dBm / -88 dBm (SNR -9)  1150 ms ago
        RX: 6.0 MBit/s                                    48 Pkts.
        TX: 6.0 MBit/s                                  3329 Pkts.
        expected throughput: unknown

I suppose that "ms ago" is a kind of heartbeat. It's updated regularly. Values between 0ms to 4000 or 5000ms.
From arp point of view STA is still know as well.

[WDS-AP.txt](https://github.com/openwrt/openwrt/files/12367712/WDS-AP.txt)
[WDS-client.txt](https://github.com/openwrt/openwrt/files/12367714/WDS-client.txt)

### OpenWrt version

r23728-813ef01a27

### OpenWrt target/subtarget

mediatek/filogic

### Device

Cudy WR3000 v1

### Image kind

Official downloaded image

### Steps to reproduce

Configure both AP as described, with the help of the config files attached

### Actual behaviour

STA are seen registered on both AP. STA are seen registered even if out of range for hours ...

### Expected behaviour

STA should not been seen registered on both AP. STA should be disconnected automatically when out of range

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-17:
By default associations are kept for 5 minutes if low_ack condition is not reached or client explicitly disconnects.

ctolzane, 2023-08-17:
> By default associations are kept for 5 minutes if low_ack condition is not reached or client explicitly disconnects.
Yes it should be the expected behaviour. But it's not. Even after those 5 minutes association is kept. The iwinfo status shown above is 3 hours after the STA left wifi coverage (kms away) !

brada4, 2023-08-17:
I have same setup and roaming apple, android and some wifi cameras happens instantly freeing the old association.
Only thing i have PMF aka 80211w forced and the roaming protocols are enabled by usteer not by config file.

ctolzane, 2023-08-21:
Thanks for your help @brada4. Closing the ticket.

closed 2023-08-21 by ctolzane