openwrt/openwrt #13311: netifd: PPPoE username and password can be sent to syslog in cleartext

sigprof, 2023-08-17:
### Describe the bug

When the PPPoE connection can't be established, OpenWrt repeatedly tries to reconnect, and on _some_ of those attempts (not every time) `netifd` logs a message which contains the PPPoE username and password in cleartext.  This might not matter much in the default configuration (when syslog is saved only in a RAM buffer), but it is possible to configure other destinations for syslog, and if the same problem happens with some kind of VPN connection, this may result in leaking the VPN credentials.

Example of the problematic log message (here the credentials were replaced by `XXXXX`, but the actual log message which I got contained the real PPPoE username and password):
```
Thu Aug 17 19:11:59 2023 daemon.notice netifd: wan (21169): Command failed: ubus call network.interface notify_proto { "action": 1, "command": [ "\/usr\/sbin\/pppd", "nodetach", "ipparam", "wan", "ifname", "pppoe-wan", "lcp-echo-interval", "1", "lcp-echo-failure", "5", "lcp-echo-adaptive", "+ipv6", "set", "AUTOIPV6=1", "nodefaultroute", "usepeerdns", "maxfail", "1", "user", "XXXXX", "password", "XXXXX", "ip-up-script", "\/lib\/netifd\/ppp-up", "ipv6-up-script", "\/lib\/netifd\/ppp6-up", "ip-down-script", "\/lib\/netifd\/ppp-down", "ipv6-down-script", "\/lib\/netifd\/ppp-down", "mtu", "1492", "mru", "1492", "plugin", "pppoe.so", "nic-wan" ], "interface": "wan" } (Permission denied)
```

Logs for the whole reconnection attempt which contain the problematic message:
```
Thu Aug 17 19:11:43 2023 daemon.notice pppd[21157]: pppd 2.4.9 started by root, uid 0
Thu Aug 17 19:11:58 2023 daemon.warn pppd[21157]: Timeout waiting for PADO packets
Thu Aug 17 19:11:58 2023 daemon.err pppd[21157]: Unable to complete PPPoE Discovery
Thu Aug 17 19:11:58 2023 daemon.info pppd[21157]: Exit.
Thu Aug 17 19:11:58 2023 daemon.notice netifd: Interface 'wan' is now down
Thu Aug 17 19:11:58 2023 daemon.notice netifd: Interface 'wan' is disabled
Thu Aug 17 19:11:58 2023 daemon.notice netifd: Interface 'wan6' is disabled
Thu Aug 17 19:11:58 2023 kern.info kernel: [251263.687750] mtk_soc_eth 1e100000.ethernet wan: Link is Down
Thu Aug 17 19:11:58 2023 kern.info kernel: [251263.702513] mtk_soc_eth 1e100000.ethernet wan: PHY [mdio-bus:04] driver [MediaTek MT7530 PHY] (irq=POLL)
Thu Aug 17 19:11:58 2023 kern.info kernel: [251263.712222] mtk_soc_eth 1e100000.ethernet wan: configuring for phy/rgmii link mode
Thu Aug 17 19:11:58 2023 daemon.notice netifd: Interface 'wan' is enabled
Thu Aug 17 19:11:58 2023 daemon.notice netifd: Interface 'wan' is setting up now
Thu Aug 17 19:11:58 2023 daemon.notice netifd: Interface 'wan6' is enabled
Thu Aug 17 19:11:58 2023 daemon.notice netifd: Network device 'wan' link is down
Thu Aug 17 19:11:58 2023 daemon.notice netifd: Interface 'wan' has link connectivity loss
Thu Aug 17 19:11:58 2023 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Thu Aug 17 19:11:59 2023 daemon.notice netifd: wan (21169): Command failed: ubus call network.interface notify_proto { "action": 1, "command": [ "\/usr\/sbin\/pppd", "nodetach", "ipparam", "wan", "ifname", "pppoe-wan", "lcp-echo-interval", "1", "lcp-echo-failure", "5", "lcp-echo-adaptive", "+ipv6", "set", "AUTOIPV6=1", "nodefaultroute", "usepeerdns", "maxfail", "1", "user", "XXXXX", "password", "XXXXX", "ip-up-script", "\/lib\/netifd\/ppp-up", "ipv6-up-script", "\/lib\/netifd\/ppp6-up", "ip-down-script", "\/lib\/netifd\/ppp-down", "ipv6-down-script", "\/lib\/netifd\/ppp-down", "mtu", "1492", "mru", "1492", "plugin", "pppoe.so", "nic-wan" ], "interface": "wan" } (Permission denied)
Thu Aug 17 19:11:59 2023 daemon.notice netifd: Interface 'wan' is now down
Thu Aug 17 19:11:59 2023 daemon.notice netifd: Interface 'wan' is disabled
Thu Aug 17 19:11:59 2023 daemon.notice netifd: Interface 'wan6' is disabled
Thu Aug 17 19:11:59 2023 daemon.notice netifd: Interface 'wan' is enabled
Thu Aug 17 19:11:59 2023 daemon.notice netifd: Interface 'wan6' is enabled
Thu Aug 17 19:11:59 2023 kern.info kernel: [251263.964410] mtk_soc_eth 1e100000.ethernet wan: PHY [mdio-bus:04] driver [MediaTek MT7530 PHY] (irq=POLL)
Thu Aug 17 19:11:59 2023 kern.info kernel: [251263.974083] mtk_soc_eth 1e100000.ethernet wan: configuring for phy/rgmii link mode
Thu Aug 17 19:12:01 2023 kern.info kernel: [251266.053100] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control off
Thu Aug 17 19:12:01 2023 kern.info kernel: [251266.061600] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
Thu Aug 17 19:12:01 2023 daemon.notice netifd: Network device 'wan' link is up
Thu Aug 17 19:12:01 2023 daemon.notice netifd: Interface 'wan' has link connectivity
Thu Aug 17 19:12:01 2023 daemon.notice netifd: Interface 'wan' is setting up now
Thu Aug 17 19:12:01 2023 daemon.notice netifd: Interface 'wan6' has link connectivity
Thu Aug 17 19:12:01 2023 daemon.info pppd[21371]: Plugin pppoe.so loaded.
Thu Aug 17 19:12:01 2023 daemon.info pppd[21371]: PPPoE plugin from pppd 2.4.9
Thu Aug 17 19:12:01 2023 daemon.notice pppd[21371]: pppd 2.4.9 started by root, uid 0
Thu Aug 17 19:12:16 2023 daemon.warn pppd[21371]: Timeout waiting for PADO packets
```

Note that the problematic message appears only maybe in 1/10 of attempts — most of them look OK:
```
Thu Aug 17 19:12:01 2023 daemon.notice pppd[21371]: pppd 2.4.9 started by root, uid 0
Thu Aug 17 19:12:16 2023 daemon.warn pppd[21371]: Timeout waiting for PADO packets
Thu Aug 17 19:12:16 2023 daemon.err pppd[21371]: Unable to complete PPPoE Discovery
Thu Aug 17 19:12:16 2023 daemon.info pppd[21371]: Exit.
Thu Aug 17 19:12:16 2023 daemon.notice netifd: Interface 'wan' is now down
Thu Aug 17 19:12:16 2023 daemon.notice netifd: Interface 'wan' is disabled
Thu Aug 17 19:12:16 2023 daemon.notice netifd: Interface 'wan6' is disabled
Thu Aug 17 19:12:16 2023 kern.info kernel: [251281.578467] mtk_soc_eth 1e100000.ethernet wan: Link is Down
Thu Aug 17 19:12:16 2023 kern.info kernel: [251281.592045] mtk_soc_eth 1e100000.ethernet wan: PHY [mdio-bus:04] driver [MediaTek MT7530 PHY] (irq=POLL)
Thu Aug 17 19:12:16 2023 kern.info kernel: [251281.601778] mtk_soc_eth 1e100000.ethernet wan: configuring for phy/rgmii link mode
Thu Aug 17 19:12:16 2023 daemon.notice netifd: Interface 'wan' is enabled
Thu Aug 17 19:12:16 2023 daemon.notice netifd: Interface 'wan' is setting up now
Thu Aug 17 19:12:16 2023 daemon.notice netifd: Interface 'wan6' is enabled
Thu Aug 17 19:12:16 2023 daemon.notice netifd: Network device 'wan' link is down
Thu Aug 17 19:12:16 2023 daemon.notice netifd: Interface 'wan' has link connectivity loss
Thu Aug 17 19:12:16 2023 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Thu Aug 17 19:12:16 2023 daemon.notice netifd: Interface 'wan' is now down
Thu Aug 17 19:12:16 2023 daemon.notice netifd: Interface 'wan' is disabled
Thu Aug 17 19:12:16 2023 daemon.notice netifd: Interface 'wan6' is disabled
Thu Aug 17 19:12:17 2023 kern.info kernel: [251281.730740] mtk_soc_eth 1e100000.ethernet wan: PHY [mdio-bus:04] driver [MediaTek MT7530 PHY] (irq=POLL)
Thu Aug 17 19:12:17 2023 kern.info kernel: [251281.740427] mtk_soc_eth 1e100000.ethernet wan: configuring for phy/rgmii link mode
[... 25 lines cut ...]

### Image kind

Official downloaded image

### Steps to reproduce

1. Install OpenWrt 23.05.0-rc2.
2. Use LuCi to configure a PPPoE WAN connection (no special settings, just username and password).
3. Observe the log messages (e.g., in `logread -f`) while OpenWrt tries to connect to the PPPoE server when there is no working PPPoE server on the WAN interface.

### Actual behaviour

On some reconnection attempts a syslog message is logged which contains the configured PPPoE username and password in cleartext:
```
Thu Aug 17 19:11:59 2023 daemon.notice netifd: wan (21169): Command failed: ubus call network.interface notify_proto { "action": 1, "command": [ "\/usr\/sbin\/pppd", "nodetach", "ipparam", "wan", "ifname", "pppoe-wan", "lcp-echo-interval", "1", "lcp-echo-failure", "5", "lcp-echo-adaptive", "+ipv6", "set", "AUTOIPV6=1", "nodefaultroute", "usepeerdns", "maxfail", "1", "user", "XXXXX", "password", "XXXXX", "ip-up-script", "\/lib\/netifd\/ppp-up", "ipv6-up-script", "\/lib\/netifd\/ppp6-up", "ip-down-script", "\/lib\/netifd\/ppp-down", "ipv6-down-script", "\/lib\/netifd\/ppp-down", "mtu", "1492", "mru", "1492", "plugin", "pppoe.so", "nic-wan" ], "interface": "wan" } (Permission denied)
```

### Expected behaviour

Passwords should not be exposed in syslog at least in the default configuration.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2023-08-22:
@stintel - you originally added logic to expose the failed ubus command. I don't really see a good solution short of reverting it.

stintel, 2023-08-24:
Another solution could be to add a flag that allows the user to opt in for the useless logging. Off by default. But that feels kind of weird. Probably better to use a firewall rule to not allow syslog out to untrusted networks.

patrakov, 2023-12-03:
Another option, not 100% foolproof, would be to have a few regular expressions to censor sensitive data before logging.