openwrt/openwrt #13312: Associated Stations not showing

M7BAZ, 2023-08-18:
### Describe the bug

Devices are connected via wifi,on both 2.4ghz and 5ghz stations.DHCP info shows correct on overview.
Devices have no issues with wifi conections, but Associated Stations says nothing connected.

### OpenWrt version

r23228-cd17d8df2a

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT32X

### Image kind

Official downloaded image

### Steps to reproduce

Fresh install,issues maybe appears after install luci dashboard..Did not notice issues before hand.Uninstalled dashboard and still persists no information on connected devices

### Actual behaviour

Not displaying  Associated Stations

### Expected behaviour

Show Associated Stations

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lukasz1992, 2023-08-18:
Does `iwinfo phy0-ap0 assoclist` show anything ? Replace phy0-ap0 with your wireless interfaces

streetracer89101, 2023-08-20:
I too had this issue after many hours i determined you need to reset to default then install all updates minus rpcd ones these are bad packages for some reason one or more of these causes the issue your seeing plus unable to select channel  changes menu is missing putting it into auto selection of channels...hope this helps

M7BAZ, 2023-08-20:
Not updating rpcd-mod-iwinfo is the way for now

closed 2023-08-20 by M7BAZ