openwrt/openwrt #13315: ap+sta, space in wifi essid, how to solve it?

lm2lm2, 2023-08-18:
### Describe the bug

hello

I dont know if it's the right place, but since few days Im trying to experience AP+STA behaviour with openwrt
what I see is it's failing with spaces
Im trying to connect to independent networks using ap+sta, with, as follow :

config wifi-iface
        option device     'wl0'
        option network    'wan'
        option mode       'sta'
        option ssid       'First Wifi'
        option encryption 'psk2'         # WPA2 for security
        option key        'wifikey1'     # Your existing WiFi password

config wifi-iface
        option device     'wl0'
        option network    'lan'
        option mode       'ap'
        option ssid       'Second Wifi'
        option encryption 'psk2'
        option key        'wifikey2'     # Your extender's WiFi password

unfortunately with this, it appears that it fails to connect, even with the right key.
with others wifi networks tested, it worked well
in addition, is it possible, to ask to openwrt to generate anyway a specific unsecured wifi AP when it cannot reach the speific wifi network to connect to, in sta mode? where it would avoid being forced ssh to it using wired/ethernet, especially nowadays computers are almost totally without wire/copper plug..

thank you!

### OpenWrt version

see previously 

### OpenWrt target/subtarget

see previously 

### Device

see previously 

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-19:
What do you see in `logread`?
You need to fix channel in "radio" section since AP-selected channel is almost totally not matching other AP-s channel.

lm2lm2, 2023-08-19:
> What do you see in `logread`? You need to fix channel in "radio" section since AP-selected channel is almost totally not matching other AP-s channel.

hello
thank you for your answer
in a way, the logread says it could not connect, i strongly suspect the fact there are spaces within the ESSID whom makes it difficult. I even tried with 'My\ Wifi' but doing the same : I dont find out the documentation of openwrt saying specific things for spaces within ssid name for wifi.. I tried exactly the same configuration with another network, which doesnt contains spaces, it worked immediately ; unfortunately it's impossible for me to remove the spaces of the wifi ssid, I can't change it. How could I make openwrt process the wifi name with spaces into it?

in addition, I saw that when the connection could not be made to the AP from STA, the AP mode (means wifi2, the one not in sta) doesnt shows up.. problem, if I dont have a wired/copper ethernet connection, that would block me to change cofiguration on it, instead of letting "openwrt" of factory default available to configure it directly from wifi..

obsy, 2023-08-19:
wl0? What device do you have?

lm2lm2, 2023-08-19:
> wl0? What device do you have?

those are serveral tplink or others brands devices, where they can handle radio0 are two antennas : one for AP, the other for STA, but the idea is : how to manage space-names in ESSID? it's the most important ;)

obsy, 2023-08-19:
But not supported by official OpenWrt, yes?

hauke, 2023-08-19:
Which OpenWrt version are you using?

lm2lm2, 2023-08-19:
> But not supported by official OpenWrt, yes?

openwrt or openwrt, that's openwrt not ddwrt ;)

> Which OpenWrt version are you using?

no matters, that the brackets of configuration whom I try to get fixed ;)
I dont think lede does job differently : years ago when openwrt was supporting ap+sta, I guess it was doing so with space names in SSID :)

obsy, 2023-08-19:
It does matter, because maybe you have bug in this fork, as openwrt can work with spaces in ssid.

lm2lm2, 2023-08-19:
> It does matter, because maybe you have bug in this fork, as openwrt can work with spaces in ssid.

okay, well I'll tell you in few yours, but I was wondering :
1. since which version openwrt supports spaces (or special characters) in both ssid names and key?
2. I dont see on openwrt documentation how to manage them : is there a specific page, about recent openwrt versions, whom explains how to manage it?
I thank you vm

brada4, 2023-08-19:
1. [Since forever](https://w1.fi/cgit/hostap/tree/hostapd/hostapd.conf#n92) 
2. old openwrt has wlan0 wlan1 etc adapters, new in form phyX-apY etc, please fill all missing info.

lm2lm2, 2023-08-20:
> 1. [Since forever](https://w1.fi/cgit/hostap/tree/hostapd/hostapd.conf#n92)
> 
>     2. old openwrt has wlan0 wlan1 etc adapters, new in form phyX-apY etc, please fill all missing info.

on the same router, that one has been used as perfectly working for years :
config wifi-iface                         
        option device 'radio0'            
        option network 'lan'              
        option mode 'ap'                  
        option ssid     'SSID2'                   
        option encryption 'psk2'                     
        option key '123456789'                          
                                                       
config wifi-iface 'wwan'                               
        option device 'radio0'                         
        option network 'wwan'                          
        option mode 'sta'                              
        option encryption 'psk'                        
        option key '987654321'
        option ssid     'SSID1'

brada4, 2023-08-20:
Forums are a good place to dwell into archaeology. If you have top device from 10 years ago meeting ever growing linux kernel requirements https://openwrt.org/supported_devices/864_warning you might be able to jump on modern train. or with 8+32 v22.03 is still an option. Processing power wise WPA3 is same as WPA2 you are using once connected, just the storages need to be sufficient. 
You can prepend space with backslash like `option ssid 'Free\ Airport\ Wifi'`

lm2lm2, 2023-08-20:
> 

hey ;)
already tried with name\ space, with no brackers, single ' or even double "... didnt worked... like under unix : using the "\ " to tell space within the caracters.. well, didnt worked with ap+sta, I dont know what Im missing

brada4, 2023-08-20:
Also `\x20` and `\040` are other representations of SPACE character (not including tabs, page-breaks, beeps, and dozens of unicode space chars). Maybe double or quadruple backslashes are needed to get right value to hostapd configuration. Who knows.
I said archaeology... Please check if your AP can run modern OpenWRT.

hauke, 2023-08-20:
Closing this now. The user is not willing to provide the OpenWrt version or the device he is using or any logs. Maybe the user is even using a random OpenWrt fork.

Probably the best strategy is to install the vendor firmware again and ask your vendor from which you bought the device for help if the vendor firmware has any problems, they probably provide much better support than volunteer driven open source projects for you.

closed 2023-08-20 by hauke