openwrt/openwrt #13327: ucode: host compilation library error

zokl, 2023-08-21:
### Describe the bug

If ucode is used as a host build application, it cannot find libucode library. Typical use in a Makefile to compile a ucode script.

### OpenWrt version

Openwrt 22.03

### OpenWrt target/subtarget

x86_64

### Device

Host build system

### Image kind

Self-built image

### Steps to reproduce

Prepare openwrt makefile with following code:

Code in openwrt makefile:

```
PKG_BUILD_DEPENDS:=ucode/host
UCODEC:=$(STAGING_DIR_HOSTPKG)/bin/ucode -c

define Package/$(PKG_NAME)/install
        $(INSTALL_DIR) $(1)/usr/sbin
        $(UCODEC) -o $(PKG_BUILD_DIR)/test $(PKG_BUILD_DIR)/test.uc

        $(INSTALL_DIR) $(1)/usr/sbin
        $(INSTALL_BIN) $(PKG_BUILD_DIR)/test.uc $(1)/usr/sbin/test
endef
```

### Actual behaviour

Compilation crash with:

```
install -d -m0755 /mnt/data/ftester/ftester_production_22/build_dir/target-x86_64_musl/ft-led-ctl-2023082101/.pkgdir/ft-led-ctl/usr/sbin
/mnt/data/ftester/ftester_production_22/staging_dir/hostpkg/bin/ucode -c -o /mnt/data/ftester/ftester_production_22/build_dir/target-x86_64_musl/ft-led-ctl-2023082101/ft-led-ctl /mnt/data/ftester/ftester_production_22/build_dir/target-x86_64_musl/ft-led-ctl-2023082101/ft-led-ctl.uc
/mnt/data/ftester/ftester_production_22/staging_dir/hostpkg/bin/ucode: error while loading shared libraries: libucode.so.20220812: cannot open shared object file: No such file or directory
make[2]: *** [Makefile:85: /mnt/data/ftester/ftester_production_22/build_dir/target-x86_64_musl/ft-led-ctl-2023082101/.pkgdir/ft-led-ctl.installed] Error 127
make[2]: Leaving directory '/mnt/data/ftester/ftester_production_22/feeds/FTDeploy/packages/ft-led-ctl'
time: package/feeds/FTDeploy/ft-led-ctl/compile#0.30#0.32#0.59
    ERROR: package/feeds/FTDeploy/ft-led-ctl failed to build.
make[1]: *** [package/Makefile:120: package/feeds/FTDeploy/ft-led-ctl/compile] Error 1
make[1]: Leaving directory '/mnt/data/ftester/ftester_production_22'
make: *** [/mnt/data/ftester/ftester_production_22/include/toplevel.mk:230: package/ft-led-ctl/compile] Error 2
```

### Expected behaviour

Compilation finish without error.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-22:
Output of host compilation of ucode binary would be nice.

zokl, 2023-08-22:
> Output of host compilation of ucode binary would be nice.

The log is here:
[ucode_host.log](https://github.com/openwrt/openwrt/files/12407835/ucode_host.log)

If I extend calling code with path to the library, ucode works:

 `UCODEC:=LD_LIBRARY_PATH=$(LD_LIBRARY_PATH):$(STAGING_DIR_HOSTPKG)/lib/ $(STAGING_DIR_HOSTPKG)/bin/ucode -c`

zokl, 2023-08-22:
> Probably worth asking at ucode repo in regard to having optional static build for your CI purpose?

Yes, a static build would help.

The problem is even more complex. The ucode-mod-ubus ucode-mod-uloop and ucode-mod-uci packages do not compile at all under host compilation.

Maybe @jow- or @ynezz will help?

brada4, 2023-08-22:
Tracker is here: https://github.com/jow-/ucode
Please check if you can build latest version, almost certain that soon-to-be-old v22 SDK will never work your way.