openwrt/openwrt #13329: Explicit interface naming via ucidef_set_network_device_path half-baked

pprindeville, 2023-08-22:
### Describe the bug

Because `/lib/preinit/10_indicate_preinit` runs early, the only modules that have been loaded are those in `/etc/modules-boot.d/`

If you wish to explicitly bind interface names to PCI paths for ordinary or garden variety devices that are loaded by kmodloader from `/etc/modules.d/` in `/etc/init.d/boot` you're out of luck.  `boot()` doesn't handle them (but there's nothing stopping it from doing so).

If we do need to repeat some of the steps to handle renaming, that logic should probably be moved into `/lib/functions/renumber.sh` or something similar.

### OpenWrt version

r23364+189-9519a62549

### OpenWrt target/subtarget

x86/64

### Device

Silicom 80500-0214-G02-SL01A

### Image kind

Self-built image

### Steps to reproduce

Modify your `/etc/board.d/02_network` to use `ucidef_set_network_device_path` to reorder the naming of your interfaces.  If the devices aren't marked as "required for boot", it won't work.

If your device is marked as "required for boot", try to build an image where it's not.

### Actual behaviour

No interface names are renumbered for devices whose kmod is loaded via `/etc/modules.d/` (i.e. those devices using a kmod which isn't marked as "required for boot").

### Expected behaviour

Interfaces names should be renumbered regardless of whether they are loaded by `/etc/preinit` or by `/etc/init.d/boot`.

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_DEVEL=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_ATH10K-CT_LEDS=y
CONFIG_ATH_USER_REGD=y
CONFIG_BUILD_PATENTED=y
CONFIG_BUSYBOX_CONFIG_ARP=y
CONFIG_BUSYBOX_CONFIG_ARPING=y
# CONFIG_BUSYBOX_CONFIG_ASH_RANDOM_SUPPORT is not set
CONFIG_BUSYBOX_CONFIG_BASE64=y
CONFIG_BUSYBOX_CONFIG_BLKID=y
CONFIG_BUSYBOX_CONFIG_BLOCKDEV=y
CONFIG_BUSYBOX_CONFIG_BUNZIP2=y
CONFIG_BUSYBOX_CONFIG_BZCAT=y
# CONFIG_BUSYBOX_CONFIG_DATE is not set
# CONFIG_BUSYBOX_CONFIG_DD is not set
CONFIG_BUSYBOX_CONFIG_DEVMEM=y
CONFIG_BUSYBOX_CONFIG_DOS2UNIX=y
CONFIG_BUSYBOX_CONFIG_FDISK=y
CONFIG_BUSYBOX_CONFIG_FEATURE_BLKID_TYPE=y
CONFIG_BUSYBOX_CONFIG_FEATURE_BZIP2_DECOMPRESS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_CATN=y
CONFIG_BUSYBOX_CONFIG_FEATURE_CATV=y
CONFIG_BUSYBOX_CONFIG_FEATURE_CHOWN_LONG_OPTIONS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_COMPRESS_USAGE=y
CONFIG_BUSYBOX_CONFIG_FEATURE_CP_LONG_OPTIONS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_CROND_D=y
[... 832 lines cut ...]
# CONFIG_MBEDTLS_KEY_EXCHANGE_RSA_ENABLED is not set
# CONFIG_MBEDTLS_KEY_EXCHANGE_RSA_PSK_ENABLED is not set
CONFIG_MBEDTLS_NIST_KW_C=y
# CONFIG_MBEDTLS_PLATFORM_C is not set
# CONFIG_MBEDTLS_RIPEMD160_C is not set
CONFIG_MBEDTLS_RSA_NO_CRT=y
# CONFIG_MBEDTLS_SELF_TEST is not set
# CONFIG_MBEDTLS_SSL_RENEGOTIATION is not set
# CONFIG_MBEDTLS_SSL_TRUNCATED_HMAC is not set
# CONFIG_MBEDTLS_VERSION_C is not set
# CONFIG_MBEDTLS_VERSION_FEATURES is not set
# CONFIG_MBEDTLS_XTEA_C is not set
# CONFIG_OPENSSL_OPTIMIZE_SPEED is not set
# CONFIG_OPENSSL_SMALL_FOOTPRINT is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
CONFIG_PACKAGE_libmbedtls=y
# CONFIG_VERSION_FILENAMES is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pprindeville, 2023-09-03:
Fixed by PR #13411.

djarbz, 2024-07-05:
I took inspiration from @pprindeville and his PR, but it didn't quite work for me.
I have just purchased the Qotom Q20331G9 1U unit and the interface numbering was a bit jumbled compared to how it was physically laid out.

I added this case to `/etc/board.d/02_network`.
I'm not sure if `default-string-qdnv01` is only for this model or not as this is the only Qotom device I have, so this may need additional logic.
```sh
default-string-qdnv01|qotom-q20331g9-1u)
        # Set board ID and model name
        echo "qotom-q20331g9-1u" > /tmp/sysinfo/board_name
        echo "Q20331G9-1U" > /tmp/sysinfo/model
        ucidef_set_board_id "qotom-q20331g9-1u"
        ucidef_set_model_name "Q20331G9-1U"

        # Stand-alone WAN port
        ucidef_set_network_device_path "eth0" "pci0000:00/0000:00:10.0/0000:08:00.0"
        # Top/Bottom Left/Right port numbering
        ucidef_set_network_device_path "eth1" "pci0000:00/0000:00:0b.0/0000:04:00.0"
        ucidef_set_network_device_path "eth2" "pci0000:00/0000:00:0e.0/0000:06:00.0"
        ucidef_set_network_device_path "eth3" "pci0000:00/0000:00:0c.0/0000:05:00.0"
        ucidef_set_network_device_path "eth4" "pci0000:00/0000:00:0f.0/0000:07:00.0"
        ucidef_set_network_device_path "sfp1" "pci0000:00/0000:00:17.0/0000:0c:00.1"
        ucidef_set_network_device_path "sfp2" "pci0000:00/0000:00:17.0/0000:0c:00.0"
        ucidef_set_network_device_path "sfp3" "pci0000:00/0000:00:16.0/0000:0b:00.1"
        ucidef_set_network_device_path "sfp4" "pci0000:00/0000:00:16.0/0000:0b:00.0"
        #ucidef_set_interface_lan_wan "eth1 eth2 eth3 eth4 sfp1 sfp2 sfp3 sfp4" "eth0"
        ucidef_set_interface_wan "eth0" "dhcp"
        ;;
```

I created `/etc/init.d/renumber` based on the PR.
This service will rename the interfaces from eth[0-8] to match what I defined above based on PCI path.
It will also assign all interfaces that are not considered WAN to the LAN designation.
We can't assign LAN devices above as the physical devices have not been renamed yet.
```sh
#!/bin/sh /etc/rc.common
# shellcheck shell=busybox

# Enforce interface renaming per /etc/board.d/02_network and /etc/board.json

# shellcheck disable=SC2034
START=15
# shellcheck disable=SC2034
NAME=renumber
CFG=/etc/board.json

# shellcheck disable=SC1091
. /lib/functions/system.sh
# shellcheck disable=SC1091
. /lib/functions/uci-defaults.sh
# shellcheck disable=SC1091
. /usr/share/libubox/jshn.sh

# Function to check if an element is in a set
in_set() {
    local elem="${1}"
    shift
    local item

    for item in "$@"; do
        [ "${item}" = "${elem}" ] && return 0
    done
    return 1
}

# Function to get all network interfaces
get_intfs() {
    local intfs="" intf

    for intf in /sys/class/net/*; do
        intf="${intf##*/}"
[... 200 lines cut ...]
    done

    for netdev in ${netdevs}; do
        # echo "Finalizing rename for NetDev: ${netdev}"
        json_select "network_device"
        json_select "${netdev}"
        json_get_vars path
        if [ -n "${path}" ]; then
            local oldname
            oldname="$(devpath_to_intf "${path}")"
            if [ -n "${oldname}" ] && [ "${oldname}" != "${netdev}" ]; then
                echo "Renaming ${oldname} to ${netdev}"
                ip link set "${oldname}" name "${netdev}"
                ip link set "${netdev}" up
            fi
        fi
        json_select ..
        json_select ..
    done

    # Update board.json with local LAN devices
    update_board_config "${netdevs}"
}

```