openwrt/openwrt #13334: Connected devices: The device name is reported correctly.

dziugas1959, 2023-08-22:
### Describe the bug

In short, some names are not reported correctly and others that include non english characters are not translated correctly, this is mainly comparing to stock firmware and „Openwrt“ firmware. 
[image]
[image]

In this case, the Lithuanian letters that showed up correctly in „Mercusys“ firmware isn't/aren't the same as in „Openwrt“.

### OpenWrt version

OpenWrt 23.05.0-rc3 r23389-5deed175a5

### OpenWrt target/subtarget

mediatek

### Device

Mercusys MR70X v1

### Image kind

Official downloaded image

### Steps to reproduce

Connect to the network (In this case „Wi-Fi“) with a device that has standard non-english characters in this case: ąčęėįšųū„“žāēīōė̄.
And in „luci“ you will see it, I do believe that this is an „openwrt“ issue and not „luci“ (Though it might be both), since the „firewall“ mapping address show the exact same issue with the names.

### Actual behaviour

Instead of ž it becomes z and instead of „insertnamehere“ it becomes ?.
Quite a simple problem.

### Expected behaviour

Instead of ?
It should be „insertnamehere“

### Additional info

It seems to be a simple „unicode“ or „UTF-8“ compliance issue.
I would personally look at this issue more for international approval, since this sorta kills clear network management, unless you are willing to go MAC address hunting, for that one phone or what ever, since most people name their devices after themselves, and having a stupid „?“ is a dumb bug. 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pkgadd, 2023-08-22:
OpenWrt displays the active DHCP leases, which in turn represent the hostnames of the connected clients (if defined by a static lease using that information, if not, dynamically provided by the client as part of the DHCP request). The syntax for hostnames is defined and rather strict, see e.g. https://en.wikipedia.org/wiki/Hostname#Syntax for a high-level overview, special characters are invalid there and will cause various problems (and displaying issues are the least harmful ones among them).

dziugas1959, 2023-08-22:
> OpenWrt displays the active DHCP leases, which in turn represent the hostnames of the connected clients (if defined by a static lease using that information, if not, dynamically provided by the [...]

Sadly this falls apart as sites like https://raštija.lt/ work, not to meantion

dziugas1959, 2023-08-22:
> OpenWrt displays the active DHCP leases, which in turn represent the hostnames of the connected clients (if defined by a static lease using that information, if not, dynamically provided by the [...]

Sadly this falls apart as sites like https://raštija.lt/ work as expected not to mention as posted above, the original firmware supported this without issue, since this is in fact a string value.

jow-, 2023-08-22:
Provide the output of `cat /tmp/dhcp.leases`

closed 2023-08-22 by jow-

dziugas1959, 2023-08-22:
> ~Provide the output of `cat /tmp/dhcp.leases`~
> 
> See https://forum.openwrt.org/t/cyrillic-is-not-displayed-in-hostname-openwrt/109880/8 for prior discussion. OpenWrt's underlying DHCP server will reject client advertised hostnames containing non-ascii characters.
> 
> This won't be fixed.

Would an option for manually setting a name for a specific IP/MAC address be an adequate solution/compromise? Since it would be visual and remembered by your config, it shouldn't conflict with the DHCP server in this case.

pkgadd, 2023-08-22:
> Sadly this falls apart as sites like [https://raštija.lt/](https://xn--ratija-ckb.lt/) work as expected not to mention as posted above

It does not, as `raštija.lt` is indeed and invalid domain name and does not exist, what does exist however is `xn--ratija-ckb.lt` - the rest is cosmetics on converting [punycode](https://en.wikipedia.org/wiki/Punycode) back and forth (yes, this `[CENSORED]` in RFC3492 should have never been accepted).

dziugas1959, 2023-08-23:
> > Sadly this falls apart as sites like [https://raštija.lt/](https://xn--ratija-ckb.lt/) work as expected not to mention as posted above
> 
> It does not, as `raštija.lt` is indeed an invalid domain name and does not exist, what does exist however is `xn--ratija-ckb.lt` - the rest is cosmetics of converting [punycode](https://en.wikipedia.org/wiki/Punycode) back and forth host side.
> 
> Yes, this `[CENSORED]` in RFC3492 should have never been accepted, it's not surprising that the owner apparently felt a need to register `rastija.lt` alongside (paying twice, domain squatters' and scammers' (typo-domains) heaven).

Well that's sorta the point, I don't mean to say in this issue as in: „Let's completely change how DHCP and hostnames work“, all I ask for is a cosmetic change, in this case something like an edit button next to connected devices which would be remembered and for example be displayed when port forwarding or what not.
In short, the DHCP server acts and works the exact same way, no change is needed, meanwhile cosmetically you are able to name devices however you want in „UTF-8“.

I think this would even help in future reports, where a simple response like: „DHCP only supports ASCII characters because 1985 rules on the internet, but you can edit the device names right there and no problems on either end“ A win-win situation.

jow-, 2023-08-23:
Well you can simply declare static DHCP and assign arbitrary names there.

dziugas1959, 2023-08-23:
> Well you can simply declare static DHCP and assign arbitrary names there.

Doesn't work, or rather doesn't let me assign arbitrary names, it still tries to impose ASCII limits, both in „DHCP static leases“ and in „Custom hostnames“ From what I understand, the option does exist, but it means zero, if it still has the same limit even on cosmetic changes.

dziugas1959, 2023-08-24:
I have tested to check if it really is a simple „UTF-8“ compliance issue, and it seems to be, since manually changing the name appears to confirm this: 
[image]
Factually, this shouldn't appear as a problem, since it's a custom given hostname which doesn't interfere with „DHCP“, since the address still uses the original device name.

dziugas1959, 2024-01-07:
Ok, reviving old issues, since I saw something interesting in the about section of connected devices. I see that „DHCPv4“ is still reporting ? or - on device names, but v6 is 100% okay with it???
[image]
This seems 100% more, if „IPv6“ connections show up with non-English characters, then why not „IPv4“ connections?