openwrt/openwrt #13336: MikroTik RouterBOARD 750Gr3 (RB750Gr3) sysupgrade has no effect

kevin-david, 2023-08-23:
### Describe the bug

Using sysupgrade to attempt to move from 22.03 to 23.05-rc3 has not yet been successful. Have tried flashing via `sysupgrade` CLI and LuCI with no sucess - the flash seems to work, but when the router reboots it's back on 22.03. 

Haven't tried other flashing methods yet or wiping settings, in case that helps track the issue down.

### OpenWrt version

 22.03.5 r20134-5f15225c1e

### OpenWrt target/subtarget

ramips/mt7621

### Device

MikroTik RouterBOARD 750Gr3

### Image kind

Official downloaded image

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

kevin-david, 2023-08-23:
FWIW, choosing to not retain settings doesn't seem to have a change in result.

brada4, 2023-08-23:
Can you flash 22.03.4? Best capture full output calculating sha256 of file used to flash.

kevin-david, 2023-08-25:
SHA sums of all attempted flashes:
```
kevin@Kevins-MacBook-Pro-2 in openwrt/
› shasum -a 256 *                                                                                                                                                                                                        [13:27:33]
f445af0c5620e9e076c8ac9622db05590e20cb190818d893bdc56b93ceb1ae9a  openwrt-22.03.4-ramips-mt7621-mikrotik_routerboard-750gr3-squashfs-sysupgrade.bin
2d82ed1d407488b0e4378351e8953c90267bda177213924fa389721a845d9375  openwrt-22.03.5-ramips-mt7621-mikrotik_routerboard-750gr3-squashfs-sysupgrade.bin
1dc3692fcd1388172974bf23a62c57529e438d0fd35b21900617489dda0f580b  openwrt-23.05.0-rc3-ramips-mt7621-mikrotik_routerboard-750gr3-squashfs-sysupgrade.bin
```

Which appear to match the website.

Flashing 22.03.4 did not work - still on 22.03.5... what are the odds this is some kind of hardware/flash issue?

brada4, 2023-08-25:
You can flash again from mikrotik bootolader, like initial installation.
https://openwrt.org/toh/mikrotik/rb750gr3#installation
(unplugging any additional storage for installation?)

kevin-david, 2023-08-26:
Thanks for the pointer. After remembering where the reset button was on this device (not the side button you can press with a finger...) I was able to successfully flash via netboot:

```
USER=kevin
NETDEV=en13
sudo ip addr flush dev $NETDEV
sudo ip addr add 10.1.1.10/24 dev $NETDEV
sudo dnsmasq -i $NETDEV \
    --no-daemon \
    --listen-address 10.1.1.10 \
    --bind-interfaces \
    -p0 \
    --dhcp-authoritative \
    --dhcp-range=10.1.1.50,10.1.1.100 \
    --bootp-dynamic \
    --dhcp-boot=$(pwd)/openwrt-23.05.0-rc3-ramips-mt7621-mikrotik_routerboard-750gr3-initramfs-kernel.bin \
    --log-dhcp \
    --enable-tftp \
    --tftp-root=$(pwd)
``` 

After booting that image, I was able to flash `openwrt-23.05.0-rc3-ramips-mt7621-mikrotik_routerboard-750gr3-squashfs-sysupgrade.bin` and restore my previous settings. 

Any idea why flashing from the UI wouldn't work? Also happy to close this issue if it's not worth investigating.

brada4, 2023-08-26:
it is worth fixing.
Can you flash rc2 or 22.03.4 and then back rc3? I.e if only 22.3.5 is defective or more versions.

kevin-david, 2023-10-03:
Sorry, this slipped my mind today until I did an upgrade to 23.05-rc4, which worked great via attended sysupgrade. So perhaps whatever the issue was is now fixed, or there was some weird one-off issue with the flash on my device. From what I recall there were some issues with certain apps (like UPnP) before the flash via netboot.

Honestly I don't think I'll spend time trying to reproduce this any longer given the available workaround for other folks & hopefully if someone else hits this it can serve as documentation.

closed 2025-06-25 by kevin-david