openwrt/openwrt #13342: bug in device MSG1500 There is a problem with the device regarding the WiFi

Oozil, 2023-08-25:
### Describe the bug

The problem is related to wireless. When I disable Wi-Fi, it does not work unless I restart the device
Wi-Fi continues to work even after turning it off

### OpenWrt version

master

### OpenWrt target/subtarget

mt7621at

### Device

MSG1500

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-25:
Please complete:
` . /etc/openwrt_release && echo $DISTRIB_REVISION`

And try to get `logread f -e hostap` while you (try to) enable, then minute later disable the wifi.

Oozil, 2023-08-25:
> Please complete: ` . /etc/openwrt_release && echo $DISTRIB_REVISION`
> 
> And try to get `logread f -e hostap` while you (try to) enable, then minute later disable the wifi.

I have tried version 22.03.5 and it works without this problem, and after this version this problem appears

brada4, 2023-08-25:
Ok, can you compare `logread > /tmp/wifi.txt` of time-controlled experiment i.e switching it off and on minute apart and compare 2 versions?

Oozil, 2023-08-25:
> 

I will do that and send you txt

brada4, 2023-08-25:
Hou can mask mac addresses leaving cutting 2nd and 3rd youngest bytes
12:34:56:xx:yy:78
Just like public ip's 1st 2 numbers

Oozil, 2023-08-25:
> Hou can mask mac addresses leaving cutting 2nd and 3rd youngest bytes 12:34:56:xx:yy:78 Just like public ip's 1st 2 numbers

[wifi.txt](https://github.com/openwrt/openwrt/files/12443326/wifi.txt)

brada4, 2023-08-25:
Anomalous (one i never seen) message generates here:
https://github.com/openwrt/mt76/blob/b14c2351ddb8601c322576d84029e463d456caef/mt7615/mcu.c#L706C35-L706C35

Oozil, 2023-08-25:
> Anomalous (one i never seen) message generates here: https://github.com/openwrt/mt76/blob/b14c2351ddb8601c322576d84029e463d456caef/mt7615/mcu.c#L706C35-L706C35

Is this a fixable bug? 
What does the message mean

brada4, 2023-08-25:
somebody at that repo may know. But seems beacon, the packet that shows AP in client list, gets discarded without reaching the air.

Oozil, 2023-08-25:
> somebody at that repo may know. But seems beacon, the packet that shows AP in client list, gets discarded without reaching the air.

I will wait until this problem is resolved
Thank you for your interest and follow-up