openwrt/openwrt #13345: New Port Status Overview (Luci) Feedback

shauno100, 2023-08-25:
### Describe the bug

Hi, this is not an issue but more feedback, please direct me to the right area if this is the wrong place. I am really liking the new port status overview in OpenWrt 23.05 RC3 on Xiaomi AX6000 router but would it be more logical to have the red underline for ports that are not connected along with the term "no link" instead of using red to differentiate WAN from the LAN ports where the LAN ports have a green underline regardless of whether they are connected or not? 

Just didn't make much sense to me that's all to differentiate on port type as you'd expect green for connected ports regardless of type and red for disconnected ports etc.

[image]

### OpenWrt version

OpenWrt 23.05 RC3

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Redmi AX6000

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-25:
Obviously such feedback must go to ../luci repo.

shauno100, 2023-08-25:
reported in Luci sub-branch.