openwrt/openwrt #13349: GL-AR750: image needs kmod-usb-storage

therealjflf, 2023-08-25:
### Describe the bug

The GL-AR750 contains a micro-SD card slot attached to a USB Genesys Logic controller. As the `kmod-usb-storage` isn't part of the standard image, the card is inaccessible until the package is installed.

The full explanation with outputs is on the forum:
[https://forum.openwrt.org/t/sd-card-on-gl-ar750-gpio-issue/120963/2](https://forum.openwrt.org/t/sd-card-on-gl-ar750-gpio-issue/120963/2)

I have checked the solution on the latest release (23.05.0-rc3), but the original question was opened over a year ago so it's likely that the problem exists in the 22.03 branch too.

### OpenWrt version

r23389-5deed175a5

### OpenWrt target/subtarget

ath79/generic

### Device

GL.iNet GL-AR750

### Image kind

Official downloaded image

### Steps to reproduce

- install the official release
- insert an SD card

### Actual behaviour

The card cannot be found: no log messages, no device in `/dev`.

### Expected behaviour

The card is detected by the kernel.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-29:
Device page states you have to install it additionally
https://openwrt.org/toh/gl.inet/gl-ar750#basic_configuration

brada4, 2023-08-29:
You can add the driver for embedded storage device via pull request in this line:
https://github.com/openwrt/openwrt/blob/33046d2a471ed687404a617c356b9cdf5652c2d6/target/linux/ath79/image/generic.mk#L1612