openwrt/openwrt #13350: Can't update Bananapi BPi-R3 keeping settings

pmendla, 2023-08-25:
### Describe the bug

Can't update 23.05-rc2 to 23.05-rc3 without force update and loosing configuration.

Error message:
Fri Aug 25 20:38:09 2023 user.info upgrade: The device is supported, but the config is incompatible to the new image (1.0->1.1). Please upgrade without keeping config (sysupgrade -n).
Fri Aug 25 20:38:09 2023 user.info upgrade: Device tree overlay mechanism needs bootloader update

### OpenWrt version

r23389-5deed175a5

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Steps to reproduce

Try to update using Attended Sysupgrade or Flash new image option

### Actual behaviour

Can't update with retained settings and without 'Force update' option

### Expected behaviour

Update should retain configuration

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

obsy, 2023-08-26:
Yes, by https://github.com/openwrt/openwrt/commit/49bd38f01ac6927cda2dc894417994d1256673a7
So, "Please upgrade without keeping config"

im-0, 2023-10-15:
@obsy Update from 23.05-rc3 to 23.05.0 on BPI-R3 is still not possible despite the fact that https://github.com/openwrt/openwrt/commit/49bd38f01ac6927cda2dc894417994d1256673a7 exists in both 23.05-rc3 and 23.05.0:

```
BusyBox v1.36.1 (2023-09-29 09:29:36 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0-rc4, r23482-7fe85ce1f2
 -----------------------------------------------------
root@rswifi:~# cd /tmp/
root@rswifi:/tmp# sysupgrade -c -v openwrt-23.05.0-mediatek-filogic-bananapi_bpi-r3-squashfs-sysupgrade.itb 
Sun Oct  8 16:26:09 CEST 2023 upgrade: The device is supported, but the config is incompatible to the new image (1.0->1.1). Please upgrade without keeping config (sysupgrade -n).
Sun Oct  8 16:26:09 CEST 2023 upgrade: Device tree overlay mechanism needs bootloader update
Image check failed.
root@rswifi:/tmp# 
```

obsy, 2023-10-15:
Please upgrade **without** keeping config. Did you saw that?
sysupgrade -n  openwrt-23.05.0-mediatek-filogic-bananapi_bpi-r3-squashfs-sysupgrade.itb  and you can also add -F

And before: " Device tree overlay mechanism needs bootloader update" - search forum for that.

im-0, 2023-10-15:
I had an impression that this was fixed before 23.05-rc4. I did a fresh install of 23.05-rc4 from a MicroSD card, without keeping anything. Am I wrong?

popy2k14, 2023-11-17:
Hi guys, same here on an nanoPi RS4 with 23.05.0 which should be upgraded with an attended sysupgrade to 23.05.2. The image get's built and on install it errors out with:

`Fri Nov 17 19:02:08 2023 user.info upgrade: The device is supported, but the config is incompatible to the new image (1.1->1.0). Please upgrade without keeping config (sysupgrade -n).`

Any solutions or workarounds to keep the config?

Hre is my thread on the forum: https://forum.openwrt.org/t/nanopi-r4s-23-05-2-attended-sysupgrade-issues/177912

pmendla, 2023-11-17:
I've set manually in my config: compat_version=1.1 and since then sysupgrade works as expected keeping configuration.

popy2k14, 2023-11-17:
thx a lot, that did the trick also for me.
Maybe the issue was from porting/restoring my ER-X config on the nanoPi R4S on 23.05.0

Just for others, witn the same issue.
To solve it, do the following:

- Connect with ssh to the openwrt device
- nano /etc/config/system         (or with vi)
- Search for:
     config system
        option compat_version 'X.X'
- put the requested version number in from the log message, in my case it was requesting 1.0. So i changed 1.1 to 1.0
- CTRL-X > Y
- do attended sysupgrade again, this time it worked

hauke, 2023-11-26:
When you boot up a Bananapi BPi-R3 without configuration it should be set to version 1.1, like the image expects. This is handled here: 49bd38f01ac6927cda2dc894417994d1256673a7 I think this problem was fixed. 

I can not find any nanoPi RS4, you are probably not using upstream OpenWrt. 
Just coping the configuration from one device to an other is not support.

closed 2023-11-26 by hauke

erik434, 2025-04-30:
If anyone else ends up here trying to figure out how to update their Bananapi BPi-R3 to 24.10.0 or later, with an error that says `The device is supported, but the config is incompatible to the new image (1.1->1.2).`, you may have done what I did... posting about it here in case it helps somebody out in the future. Here's what I did:

1. Saw the error and that it required a full reinstall from SD card.
2. Took a backup right before doing the reinstall.
3. Did a full reinstall from SD - for me that involved changing the DIP switches, installing 24.10.0 from SD to NAND, changing DIP switches, installing from NAND to eMMC, and one final DIP switch change.
4. Router booted up and I verified 24.10.0 was installed.
5. Thought, "great! all set", and restored my backup.
6. All seemed well; time passed and 24.10.1 was released.
7. Tried using `owut upgrade` to seamlessly update to 24.10.1... and got the exact same error about going from 1.1 to 1.2. What? I already did that!

I think where I messed up was step 5 - restoring my backup from before the upgrade also restored my /etc/config/system, where the compat_version is stored! So that reset compat_version from 1.2 back to 1.1, meaning it thought I still needed the bootloader update required to reach 1.2.

In this case, since I was sure I'd already updated my bootloader, I just manually set compat_version to 1.2, and then I was able to use `owut upgrade` to get 24.10.1 without any trouble!

In the future though, it seems like I will want to think twice before restoring a backup after one of these compat_version bumps. Depending on why the compat_version bump was needed, restoring a backup might really mess things up. May still be able to manually edit a backup to *make* it compatible with the new FW, or remove some files from the backup and restore the rest, to avoid needing to do a complete manual re-config of the router; it'll all depend on why exactly compat_version was bumped.