openwrt/openwrt #13354: lantiq: Missing kmod folder and core packages?

smavridis, 2023-08-26:
### Describe the bug

Unable to build image for w8980 using imagebuilder, due to download errors.

opkg_download: Failed to download https://downloads.openwrt.org/snapshots/targets/lantiq/xrx200/kmods/5.15.111-1-cf66ca0b201d986f9127dda1b9bbc526/Packages.gz, wget returned 8.

Verified with browser, and file does not exist (404).

### OpenWrt version

r22888-0779c47be6

### OpenWrt target/subtarget

lantiq/xrx200

### Device

w8980

### Image kind

Self-built image

### Steps to reproduce

Download imagebuilder from `https://downloads.openwrt.org/snapshots/targets/lantiq/xrx200/openwrt-imagebuilder-lantiq-xrx200.Linux-x86_64.tar.xz`.

Use this command to build the image:
`make image PROFILE='tplink_tdw8980'`

### Actual behaviour

Image build, fails due to failed download of package lists, resulting in unresolved core packages (e.g. luci and kernel modules)

Attached build log:
[build.txt](https://github.com/openwrt/openwrt/files/12445347/build.txt)

### Expected behaviour

This used to work a few months ago, so perhaps some CI/upload step did not complete?

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

khanjui, 2023-08-26:
IIRC those devices were disabled.

smavridis, 2023-08-26:
@khanjui You recall correctly.

It seems in 0c117e1 lantiq was made source only.

This pull request seems to try to fix the errors #13200.

So i guess i have to try and build from source until it gets merged?

So i should follow the [Using the toolchain](https://openwrt.org/docs/guide-developer]/start#using_the_toolchain) documentation?

khanjui, 2023-08-26:
Maybe this would work if you don't mind using a bit old version.

https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=lantiq%2Fxrx200&id=tplink_tdw8980

jospezial, 2023-11-22:
@smavridis https://downloads.openwrt.org/snapshots/targets/lantiq/xrx200/ are built again. Please close.

smavridis, 2023-11-24:
Thank you to all the community.

closed 2023-11-24 by smavridis