openwrt/openwrt #13361: Waiting for root device PARTUUID=...........        and nothing else.

kipronar, 2023-08-27:
### Describe the bug

OpenWrt  stops  while booting.
official image 23.05.0-rc3 and self-built image have the same bug.

### OpenWrt version

r23389-5deed175a5

### OpenWrt target/subtarget

x86/64

### Device

AOpen DE3250

### Image kind

Official downloaded image

### Steps to reproduce

Write  .img file using MX linux live, gparted.  Then boot.

### Actual behaviour

OpenWrt stops in line:
Waiting for root device PARTUUID=...

### Additional info

lspci:
_00:00.0 Host bridge: Intel Corporation Atom Processor Z36xxx/Z37xxx Series SoC Transaction Register (rev 0e)
00:02.0 VGA compatible controller: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Graphics & Display (rev 0e)
00:13.0 SATA controller: Intel Corporation Atom Processor E3800 Series SATA AHCI Controller (rev 0e)
00:14.0 USB controller: Intel Corporation Atom Processor Z36xxx/Z37xxx, Celeron N2000 Series USB xHCI (rev 0e)
00:1a.0 Encryption controller: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Trusted Execution Engine (rev 0e)
00:1b.0 Audio device: Intel Corporation Atom Processor Z36xxx/Z37xxx Series High Definition Audio Controller (rev 0e)
00:1c.0 PCI bridge: Intel Corporation Atom Processor E3800 Series PCI Express Root Port 1 (rev 0e)
00:1c.1 PCI bridge: Intel Corporation Atom Processor E3800 Series PCI Express Root Port 2 (rev 0e)
00:1c.2 PCI bridge: Intel Corporation Atom Processor E3800 Series PCI Express Root Port 3 (rev 0e)
00:1c.3 PCI bridge: Intel Corporation Atom Processor E3800 Series PCI Express Root Port 4 (rev 0e)
00:1f.0 ISA bridge: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Power Control Unit (rev 0e)
00:1f.3 SMBus: Intel Corporation Atom Processor E3800/CE2700 Series SMBus Controller (rev 0e)
01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)_

disk:   Innodisk_Corp._mSATA_3SE_DEMSR-_02GB_

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

tmn505, 2023-08-28:
Boot log or at least the `PARTUUID` would be helpful. But judging from what You've written in "Steps to reproduce", You probably resized root partition with gparted. This in result probably changed `PARTUUID` which is appended by grub.cfg to kernel command line. Either fix Your grub.cfg to look for new `PARTUUID` (it could be named `PTUUID`) or don't run gparted on freshly written image.

kipronar, 2023-08-28:
I have not resized any partition. Gparted shows me, if  OpenWrt partions exist (after using dd command). I do not make any change in Gparted.  
This foto shows booting  with other compilation (not oficial and not mine, I did a lot of attempts ). But result is exactly the same.
[image]

tmn505, 2023-08-29:
SATA disk somehow gets disconnected? Check if connector is clean, reseat the disk and check if power supply has enough power. Aside from it, please test with official images only, so we can rule out misconfiguration. For test use latest stable images 23.05, 22.03 and 21.02.
I have a device with similar controller
`00:13.0 SATA controller [0106]: Intel Corporation Atom Processor E3800 Series SATA AHCI Controller [8086:0f23] (rev 11)`
and mSATA disk works without problems with 23.05-rc3.

tmn505, 2023-08-29:
Thank You. Post a full dmesg output from 22.03.5. Enclose it with triple backticks at beginning and at the end or attach it as txt file. I fear that's going to need git bisect of a kernel to find the cause, as I don't see anything related in config refreshing commit 12c33d99c8b37013cfe9cb36a65de08a076a6ec3.

kipronar, 2023-08-29:
dmesg in this file
[dmesg_22_03_5.txt](https://github.com/openwrt/openwrt/files/12469723/dmesg_22_03_5.txt)

kipronar, 2023-08-31:
libata.force=noncq   - did not give a result.  OpenWrt stops in the same point
I can build OpenWrt, but I can't add any patch to my built.

tmn505, 2023-08-31:
> I can build OpenWrt, but I can't add any patch to my built.

No need to patch anything, clone the users fork https://github.com/coudu/openwrt, change directory to cloned repo and run `git checkout x86-6.1`. Now select proper target, build it and write built image to the disk. Don't install anything from feeds, this might break compilation or simply extend compiling time, we want to do test if it properly boots.

tmn505, 2023-08-31:
I forgot to mention, when setting target in menuconfig don't forget to select "Use the testing kernel version" in "Global build settings" section.

kipronar, 2023-09-01:
Kernel - 6.1.46-1-237442c13aae494add39b536b012a2cc  works !
[image]

[dmesg_kernel_ 6_1_46-1.txt](https://github.com/openwrt/openwrt/files/12494347/dmesg_kernel_.6_1_46-1.txt)
Thank you for your help.

graysky2, 2023-09-02:
I am just a user not a developer but my opinion is that for x86 we should switch to the latest LTS kernel.  Many Linux distros have lots of experience running 6.1-series on x86 hardware for example.

tmn505, 2023-09-02:
@graysky2 
The default branch/snapshots will be switched to another LTS kernel at some point, just not the 23.05 release.

@kipronar 
So if it works, only option for now would be bisecting kernel, or staying with 22.03 version and waiting for a release with 6.1 kernel. Unless someone with knowledge how to debug that shows up.
If You want to compare other distros (with 5.15 kernel) if they detect the drive, You can try out https://sourceforge.net/projects/systemrescuecd/files/sysresccd-x86/9.06 or check distro list here: https://distrowatch.com/search.php?pkg=linux&relation=similar&pkgver=5.15&distrorange=InLatest#pkgsearch.

graysky2, 2023-09-02:
@tmn505 
> The default branch/snapshots will be switched to another LTS kernel at some point, just not the 23.05 release.

I understand.  My comment was relating to master not a release of OpenWrt.

kipronar, 2023-09-02:
@tmn505
I tried twice.
1. systemrescue-9.06-amd64   kernel 5.15.83-1-lts
[dmesg_SRCD_9.06.txt](https://github.com/openwrt2/openwrt/files/12504464/dmesg_SRCD_9.06.txt)
2. easy-5.4.10-amd64       kernel 5.15.120
[dmesg_easy_OS_5_4_10.txt](https://github.com/openwrt/openwrt/files/12504465/dmesg_easy_OS_5_4_10.txt)

In both cases mSATA disk was not visible to the OS.

tmn505, 2023-09-03:
Thank You, this confirms it's not isolated to OpenWrt. If You feel like to hunt the kernel commit which introduced this regression, hit me, I can write instruction how to do that (this involves compiling kernel around 13 times and You'll need time to assist it).

kipronar, 2023-09-03:
OK, let's try. Ma account on OpenWrt forum has the same name. Can you send me this instruction there?

wolfgangr, 2024-11-04:
A mere year later, installing **23.05.5**, r24106-10cc5fcd00, obviously the **problem is still there**

- still running **kernel 5.15.**, which is **widely known** for the issue
- `libata.force=noncq` on the kernel comand line (manually during boot) does not change behaviour
- it even does not work when I put the SSD on an USB adapter and try to boot from there
- **it does work when I change the type of the SSD**. So mabe it's not the kernel's culprit, but it seems like newer kernel versions are known to work around some quirky SSD obviusly

Hardware: Fujitsu Futro S900 "Thin client"
works for me: SanDisk SDSA4DH-016G
does not work: innodisk mSATA 3SE (2G)

Is there an easy way to change the kernel without a fullfledged build environment?  
On the other hand - the sell the SanDisk 16 GB for less than 20 € - so what....

gusarov, 2024-12-29:
Same problem while I was trying to install on Hyper-V via Ubuntu Live disk util to restore the image...