openwrt/openwrt #13376: cpufreq gov is missing on mt7981

ptpt52, 2023-08-30:
### Describe the bug

nothing in path: `/sys/devices/system/cpu/cpufreq/`

### OpenWrt version

xx

### OpenWrt target/subtarget

xx

### Device

xx

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dangowrt, 2023-08-30:
MT7981 and MT7986 don't support cpufreq (at least not officially and we also don't have a driver for that)

ptpt52, 2023-08-30:
> MT7981 and MT7986 don't support cpufreq (at least not officially and we also don't have a driver for that)

if it is the case, should we disable these kernel symbol in config?
```
# CONFIG_CPU_FREQ_DEFAULT_GOV_PERFORMANCE is not set
CONFIG_CPU_FREQ_DEFAULT_GOV_USERSPACE=y
CONFIG_CPU_FREQ_GOV_ATTR_SET=y
CONFIG_CPU_FREQ_GOV_COMMON=y
CONFIG_CPU_FREQ_GOV_CONSERVATIVE=y
CONFIG_CPU_FREQ_GOV_ONDEMAND=y
CONFIG_CPU_FREQ_GOV_PERFORMANCE=y
CONFIG_CPU_FREQ_GOV_POWERSAVE=y
CONFIG_CPU_FREQ_GOV_SCHEDUTIL=y
CONFIG_CPU_FREQ_GOV_USERSPACE=y
CONFIG_ARM_MEDIATEK_CPUFREQ=y
```
@dangowrt 

to save some kernel size.

dangowrt, 2023-08-30:
cpufreq is supported on MT7988 which is also in filogic subtarget