openwrt/openwrt #13377: Realtek RTL8188EUS 802.11n Wireless USBNetwork Adapter cannot work

arduino2018, 2023-08-30:
### Describe the bug

Realtek RTL8188EUS 802.11n Wireless USBNetwork Adapter cannot work

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

sunxi/cortexa8

### Device

LinkSprite pcDuino

### Image kind

Official downloaded image

### Steps to reproduce

root@OpenWrt:~# opkg list-installed | grep rtl
kmod-rtl8192c-common - 5.10.176+5.15.92-1-1
kmod-rtl8192cu - 5.10.176+5.15.92-1-1
kmod-rtl8xxxu - 5.10.176+5.15.92-1-1
kmod-rtlwifi - 5.10.176+5.15.92-1-1
kmod-rtlwifi-usb - 5.10.176+5.15.92-1-1
rtl8188eu-firmware - 20230804-1
rtl8192cu-firmware - 20230804-1

root@OpenWrt:~# lsusb 
Bus 002 Device 001: ID 1d6b:0002 Linux 5.10.176 ehci_hcd EHCI Host Controller
Bus 004 Device 001: ID 1d6b:0001 Linux 5.10.176 ohci_hcd Generic Platform OHCI controller
Bus 001 Device 002: ID 0bda:8179 Realtek 802.11n NIC
Bus 001 Device 001: ID 1d6b:0002 Linux 5.10.176 ehci_hcd EHCI Host Controller
Bus 003 Device 001: ID 1d6b:0001 Linux 5.10.176 ohci_hcd Generic Platform OHCI controller

root@OpenWrt:~# lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=, Driver=, 480M

root@OpenWrt:~# cat /sys/kernel/debug/usb/devices 
......
T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=8179 Rev= 0.00
S:  Manufacturer=Realtek
S:  Product=802.11n NIC
S:  SerialNumber=00E04C0001
C:* #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
......

iw list
show nothing

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

arduino2018, 2023-08-30:
> kmod-rtl8xxxu has that USB ID in alias list.

I have done the flowing two operation:
1 .  add “alias usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in* rtl8xxxu” to /lib/modules/5.10.176/modules.alias

arduino2018, 2023-08-30:
> kmod-rtl8xxxu has that USB ID in alias list.

I have done the flowing two operation:
1 . add “alias usb:v0BDAp8179ddcdscdpiciscipin rtl8xxxu” to /lib/modules/5.10.176/modules.alias
reboot but nothing happen

2. rmmod rtl8xxxu
    insmode rtl8xxxu  alias=usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*
when logread I get the erro msg
Wed Aug 30 18:41:01 2023 kern.warn kernel: [  648.236559] rtl8192cu: unknown parameter 'alias' ignored

Can you tell me what can I do next?

brada4, 2023-08-30:
I got that from ubuntu modinfo...

brada4, 2023-08-30:
UNTESTED option is enabled in OpenWRT, it should be auto-probed based on USB  device ID, not looking at vendor sub-codes and revisions.
 Try running `logread -f | tee -a /tmp/i_sa_this` in one window while plugging device.

arduino2018, 2023-08-30:
> UNTESTED option is enabled in OpenWRT, it should be auto-probed based on USB device ID, not looking at vendor sub-codes and revisions. Try running `logread -f | tee -a /tmp/i_sa_this` in one window while plugging device.

The wifi adapter is fixed on my board, so I cannot plugout it.

brada4, 2023-08-30:
Try removing one of 
kmod-rtl8192cu - 5.10.176+5.15.92-1-1
kmod-rtl8xxxu - 5.10.176+5.15.92-1-1
Either says supports -CUS device.

arduino2018, 2023-08-30:
I have removed rtl8192cu pkgs.

root@OpenWrt:~# opkg list-installed | grep rtl
kmod-rtl8xxxu - 5.10.176+5.15.92-1-1
kmod-rtlwifi - 5.10.176+5.15.92-1-1
kmod-rtlwifi-usb - 5.10.176+5.15.92-1-1
rtl8188eu-firmware - 20230804-1

After reboot nothing changed. The wifi adapter still have no driver.

root@OpenWrt:~# lsusb
Bus 002 Device 001: ID 1d6b:0002 Linux 5.10.176 ehci_hcd EHCI Host Controller
Bus 004 Device 001: ID 1d6b:0001 Linux 5.10.176 ohci_hcd Generic Platform OHCI controller
Bus 001 Device 002: ID 0bda:8179 Realtek 802.11n NIC
Bus 001 Device 001: ID 1d6b:0002 Linux 5.10.176 ehci_hcd EHCI Host Controller
Bus 003 Device 001: ID 1d6b:0001 Linux 5.10.176 ohci_hcd Generic Platform OHCI controller
root@OpenWrt:~# lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=, Driver=, 480M

brada4, 2023-08-30:
Fully power off the device changing drivers.

arduino2018, 2023-08-31:
> Fully power off the device changing drivers.

I have Fully powered off and reinstall dirver, nothing changed.

I read carefully the package kmod-rtl8xxxu Description    https://openwrt.org/packages/pkgdata/kmod-rtl8xxxu

Description:This is an alternative driver for various Realtek RTL8XXX\\ parts written to utilize the Linux mac80211 stack.\\ The driver is known to work with a number of RTL8723AU,\\ RL8188CU, RTL8188RU, RTL8191CU, and RTL8192CU devices\\ \\ This driver is under development and has a limited feature\\ set. In particular it does not yet support 40MHz channels\\ and power management. However it should have a smaller\\ memory footprint than the vendor drivers and benetifs\\ from the in kernel mac80211 stack.\\ \\ It can coexist with drivers from drivers/staging/rtl8723au,\\ drivers/staging/rtl8192u, and drivers/net/wireless/rtlwifi,\\ but you will need to control which module you wish to load.\\ \\ RTL8XXXU_UNTESTED is enabled\\ This option enables detection of Realtek 8723/8188/8191/8192 WiFi\\ USB devices which have not been tested directly by the driver\\ author or reported to be working by third parties.\\ \\ Please report your results!\\ \\

It mentioned RL8188CU and RTL8188RU , but not 8188EUS.

Is it the reason why the openwrt cannot detect 8188EUS?

brada4, 2023-08-31:
Here are all trade names listed [long link to debian doc](https://wiki.debian.org/rtl819x#Realtek_RTL8188CE.2C_RTL8188CUS.2C_RTL8188DE.2C_RTL8188EE.2C_RTL8188ETV.2C_RTL8188EU.2C_RTL8188SU.2C_RTL8191SE.2C_RTL8191SU.2C_RTL8192CE.2C_RTL8192CU.2C_RTL8192DE.2C_RTL8192E.2C_RTL8192EE.2C_RTL8192SE.2C_RTL8192SU.2C_RTL8192U.2C_RTL8723AE.2C_RTL8723AU.2C_RTL8723BE.2C_RTL8812AE.2C_RTL8821AE_devices) 
indeed driver in `lsusb -t` indicates success.

dmesg sgould say something about usb1-1 new high speed device, and what follows is initialisation of one of 2 drivers. Just having one means it is predictable which. Rather strange device page does not mention which of 2 is confirmed to work.

brada4, 2023-08-31:
(also check iwinfo if device came up)

arduino2018, 2023-08-31:
Thanks a lot. I have found my wifi adapter in debian doc list.
[image]
> Here are all trade names listed [long link to debian [...]

But I think the problem is openwrt kmod-rtl8xxxu driver just suport RL8188CU and RTL8188RU , but not 8188EUS.
> I have Fully powered off and reinstall dirver, nothing changed.
>
> I read carefully the package kmod-rtl8xxxu Description https://openwrt.org/packages/pkgdata/kmod-rtl8xxxu
>
> Description:This is an [...]

brada4, 2023-08-31:
The device match in both kmod-rtl8xxxu and kmod-rtl8192cu contains just the very USB ID :8179.
The dmesg should go like:
... usbcore: initialised new interface driver rtl8xx
(many lines)
... new high speed device
... new device idvendor= idproduct=
... rtl8????: initialisation 
... ieee80211 .... minstrel_ht <- wifi front end is up

arduino2018, 2023-08-31:
I can see some dmesg like this:
root@OpenWrt:~# logread | grep rtl
Thu Aug 31 14:50:28 2023 kern.info kernel: [    8.551683] usbcore: registered new interface driver rtl8xxxu
root@OpenWrt:~# logread | grep 1-1
Thu Aug 31 14:50:28 2023 kern.info kernel: [    1.493270] usb 1-1: new high-speed USB device number 2 using ehci-platform
Thu Aug 31 14:50:28 2023 kern.info kernel: [    1.694488] usb 1-1: New USB device found, idVendor=0bda, idProduct=8179, bcdDevice= 0.00
Thu Aug 31 14:50:28 2023 kern.info kernel: [    1.702700] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Thu Aug 31 14:50:28 2023 kern.info kernel: [    1.709883] usb 1-1: Product: 802.11n NIC
Thu Aug 31 14:50:28 2023 kern.info kernel: [    1.713920] usb 1-1: Manufacturer: Realtek
Thu Aug 31 14:50:28 2023 kern.info kernel: [    1.718020] usb 1-1: SerialNumber: 00E04C0001

But the device's driver is null.
root@OpenWrt:~# lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=, Driver=, 480M

> The device match in both kmod-rtl8xxxu and kmod-rtl8192cu contains just the very USB ID :8179. The dmesg should go like: ... usbcore: initialised new interface driver rtl8xx (many lines) ... new [...]

brada4, 2023-08-31:
Try installing other driver back. One of drivers should be initialising right after device ID messages you post leading to 80211 interface beig initialised and device appearing in iwinfo.

arduino2018, 2023-08-31:
kmod-rtl8192cu - 5.10.176+5.15.92-1-1 - Realtek RTL8192CU/RTL8188CU support

arduino2018, 2023-08-31:
opkg list | grep 8192cu
kmod-rtl8192cu - 5.10.176+5.15.92-1-1 - Realtek RTL8192CU/RTL8188CU support

So,  kmod-rtl8192cu just suport RTL8188CU, but not RTL8188EU.

So, just rtl8xxxu suport 8188. 
But actually, rtl8xxxu also not suport RTL8188EU.
From it's Description,  kmod-rtl8xxxu driver just suport RL8188CU and RTL8188RU , but not 8188EUS.

arduino2018, 2023-08-31:
opkg list | grep 8192cu
kmod-rtl8192cu - 5.10.176+5.15.92-1-1 - Realtek RTL8192CU/RTL8188CU support

So, kmod-rtl8192cu just suport RTL8188CU, but not RTL8188EU.

So, just rtl8xxxu suport 8188.
But actually, rtl8xxxu also not suport RTL8188EU.
From it's Description, kmod-rtl8xxxu driver just suport RL8188CU and RTL8188RU , but not 8188EUS.

> Try installing other driver back. One of drivers should be initialising right after device ID messages you post leading to 80211 interface beig initialised and device appearing in iwinfo.

brada4, 2023-08-31:
Look in debian table, kernel version is similar and the device is mentioned in both places matching aliases in driver.

arduino2018, 2023-08-31:
> Look in debian table, kernel version is similar and the device is mentioned in both places matching aliases in driver.

I hope it suport , but it dose not.

I can also find some other users also have the sam problem:
https://forum.openwrt.org/t/again-rtl8188eus/21766
https://forum.openwrt.org/t/8188eu-drivers/13194
https://forum.openwrt.org/t/rtl8188eus-chipset-with-d-link-dir-885l-not-working/24732/14

So I issue this bug track, and hope someone can fix this bug.

brada4, 2023-08-31:
You have correct driver (even two) installed, it should work but does not.
- please add pcduino2 to title, it is not exactly USB device, it is some wiring on mainboard
- try 23.05-rc3 or snapshot
- try to extract `dmesg` parts on USB device from pcduino recovery linux distribution.

arduino2018, 2023-08-31:
Yes, you are right.
My board is pcduino v2. Because of no img for this verison of board, so I use the pcduino's image.
The difference between this two version of board is just the wired Realtek RTL8188EUS wifi adapter. So I used the pcduino's image and worked well all function but without wifi.
I have changed the title of this issue.
I will to try 23.05-rc3 and get dmesg in original linux distribution.

> You have correct driver (even two) installed, it should work but does not.
> 
> * please add pcduino2 to title, it is not exactly USB device, it is some wiring on mainboard
> * try 23.05-rc3 or snapshot
> * try to extract `dmesg` parts on USB device from pcduino recovery linux distribution.

brada4, 2023-08-31:
Try to extract lsusb and dmesg parts of onboard USB getting up in pcduino image.

arduino2018, 2023-08-31:
I use this image:
ARMBIAN 5.71 user-built Debian GNU/Linux 9 (stretch) 4.14.94-sunxi

The wifi adapter is work well.
root@pcduino2:~# dmesg | grep 1-1
[    4.494070] usb 1-1: new high-speed USB device number 2 using ehci-platform
[    4.695041] usb 1-1: New USB device found, idVendor=0bda, idProduct=8179
[    4.695054] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    4.695060] usb 1-1: Product: 802.11n NIC
[    4.695066] usb 1-1: Manufacturer: Realtek
[    4.695072] usb 1-1: SerialNumber: 00E04C0001
[   17.937710] r8188eu 1-1:1.0 wlx00b605405312: renamed from wlan0

root@pcduino2:~# lsusb -t
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=musb-hdrc/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=r8188eu, 480M
> Try to extract lsusb and dmesg parts of onboard USB getting up in pcduino image.

brada4, 2023-08-31:
`iw list` ie to see radio capabilities

arduino2018, 2023-08-31:
[image]

> `iw list` ie to see radio capabilities

arduino2018, 2023-09-02:
I have try openwrt 23.05-rc3 and get the same result.
After install rtl8xxxu, the pcduino v2 board still can't recongnize the soldered RTL8188EUS wifi adapter.
> You have correct driver (even two) installed, it should work but does not.
> 
> * please add pcduino2 to title, it is not exactly USB device, it is some wiring on mainboard
> * try 23.05-rc3 or snapshot
> * try to extract `dmesg` parts on USB device from pcduino recovery linux distribution.

brada4, 2023-09-02:
vendor driver (likely proprietary) driver is called r8188eu, try rtl8192cu as that is another accepting this PCI ID.

brada4, 2023-09-02:
(looking at mainline debian it says supporting v1 and v3 of this board, i wonder if that is because of badly supported network card)

Normally it is 
echo 0bda 8179 > /proc/bus/usb/drivers/rtl8192cu/new_id
or
echo 0bda 8179 > /proc/bus/usb/drivers/rtl8xxxu/new_id

Likely yield crash mandating poweroff if driver is not compatible.

arduino2018, 2023-09-02:
I used the command but get erro of "nonexistent directory".
echo 0bda 8179 > /proc/bus/usb/drivers/rtl8192cu/new_id
-ash: can't create /proc/bus/usb/drivers/rtl8192cu/new_id: nonexistent directory
root@OpenWrt:~# echo 0bda 8179 > /proc/bus/usb/drivers/rtl8xxxu/new_id
-ash: can't create /proc/bus/usb/drivers/rtl8xxxu/new_id: nonexistent directory

I used find command find /sys/bus/usb/drivers/rtl8xxxu but get erro of "Permission denied".
echo 0bda 8179 > /sys/bus/usb/drivers/rtl8xxxu/new_id
-ash: can't create /sys/bus/usb/drivers/rtl8xxxu/new_id: Permission denied

root@OpenWrt:~# echo 0bda 8179 > /sys/bus/usb/drivers/rtl8192cu/new_id
dmseg
Sat Sep  2 09:15:15 2023 kern.info kernel: [ 4900.502555] rtl8192cu: Chip version 0x10
Sat Sep  2 09:15:15 2023 kern.err kernel: [ 4900.553766] rtl_usb: reg 0x2, usbctrl_vendorreq TimeOut! status:0xffffffb9 value=0x0
Sat Sep  2 09:15:15 2023 kern.err kernel: [ 4900.608140] rtl_usb: reg 0x8, usbctrl_vendorreq TimeOut! status:0xffffffb9 value=0x0
Sat Sep  2 09:15:15 2023 kern.err kernel: [ 4900.666765] rtl_usb: reg 0x32, usbctrl_vendorreq TimeOut! status:0xffffffb9 value=0x0
Sat Sep  2 09:15:15 2023 kern.err kernel: [ 4900.721260] rtl_usb: reg 0x33, usbctrl_vendorreq TimeOut! status:0xffffffb9 value=0x0

> (looking at mainline debian it says supporting v1 and v3 of this board, i wonder if that is because of badly supported network card)
>
> Normally it is echo 0bda 8179 > [...]

brada4, 2023-09-02:
At least no crash.....
Try xxxxu driver too (after cold boot)

request timeout usually is caused by damaged usb cable just that you dont have any involved

arduino2018, 2023-09-02:
After reboot, still this erro.
echo 0bda 8179 > /sys/bus/usb/drivers/rtl8xxxu/new_id
-ash: can't create /sys/bus/usb/drivers/rtl8xxxu/new_id: Permission denied

> At least no crash..... Try xxxxu driver too (after cold boot)
> 
> request timeout usually is caused by damaged usb cable just that you dont have any involved

brada4, 2023-09-02:
Try disabling USB powersave 
`echo 'off' | tee -a  /sys/bus/usb/devices/*/power/control`

I am out of ideas by large.

brada4, 2023-09-02:
(you need to remove id and add id to driver after powersave to repeat initialisation routines)

arduino2018, 2023-09-02:
Permission denied, :(
root@OpenWrt:~# echo 'off' | tee -a  /sys/bus/usb/devices/1-1:1.0/power/control
tee: /sys/bus/usb/devices/1-1:1.0/power/control: Permission denied
off
root@OpenWrt:~# ll /sys/bus/usb/devices/1-1:1.0/
drwxr-xr-x    6 root     root             0 Jan  1  1970 ./
drwxr-xr-x    5 root     root             0 Jan  1  1970 ../
-rw-r--r--    1 root     root          4096 Sep  2 10:15 authorized
-r--r--r--    1 root     root          4096 Sep  2 10:15 bAlternateSetting
-r--r--r--    1 root     root          4096 Sep  2 10:15 bInterfaceClass
-r--r--r--    1 root     root          4096 Sep  2 10:15 bInterfaceNumber
-r--r--r--    1 root     root          4096 Sep  2 10:15 bInterfaceProtocol
-r--r--r--    1 root     root          4096 Sep  2 10:15 bInterfaceSubClass
-r--r--r--    1 root     root          4096 Sep  2 10:15 bNumEndpoints
lrwxrwxrwx    1 root     root             0 Sep  2 10:15 driver -> ../../../../../../../bus/usb/drivers/rtl8192cu/
drwxr-xr-x    3 root     root             0 Sep  2 10:15 ep_02/
drwxr-xr-x    3 root     root             0 Sep  2 10:15 ep_03/
drwxr-xr-x    3 root     root             0 Sep  2 10:15 ep_81/
-r--r--r--    1 root     root          4096 Sep  2 10:15 modalias
drwxr-xr-x    2 root     root             0 Sep  2 10:15 power/
lrwxrwxrwx    1 root     root             0 Sep  2 10:15 subsystem -> ../../../../../../../bus/usb/
-r--r--r--    1 root     root          4096 Sep  2 10:15 supports_autosuspend
-rw-r--r--    1 root     root          4096 Sep  2 10:15 uevent
rm /sys/bus/usb/devices/1-1:1.0/driver
rm: can't remove '/sys/bus/usb/devices/1-1:1.0/driver': Operation not permitted

> Try disabling USB powersave `echo 'off' | tee -a /sys/bus/usb/devices/*/power/control`
> 
> I am out of ideas by large.

brada4, 2023-09-02:
Seems pm is not enabled/supported globally. this is good, not cause of detection failure.

arduino2018, 2023-09-02:
I remove the mod-rtl8192cu and reboot , the driver is disappeared.
root@OpenWrt:~# ll /sys/bus/usb/devices/1-1:1.0/
drwxr-xr-x    6 root     root             0 Jan  1  1970 ./
drwxr-xr-x    5 root     root             0 Jan  1  1970 ../
-rw-r--r--    1 root     root          4096 Sep  2 10:47 authorized
-r--r--r--    1 root     root          4096 Sep  2 10:47 bAlternateSetting
-r--r--r--    1 root     root          4096 Sep  2 10:47 bInterfaceClass
-r--r--r--    1 root     root          4096 Sep  2 10:47 bInterfaceNumber
-r--r--r--    1 root     root          4096 Sep  2 10:47 bInterfaceProtocol
-r--r--r--    1 root     root          4096 Sep  2 10:47 bInterfaceSubClass
-r--r--r--    1 root     root          4096 Sep  2 10:47 bNumEndpoints
drwxr-xr-x    3 root     root             0 Sep  2 10:47 ep_02/
drwxr-xr-x    3 root     root             0 Sep  2 10:47 ep_03/
drwxr-xr-x    3 root     root             0 Sep  2 10:47 ep_81/
-r--r--r--    1 root     root          4096 Sep  2 10:47 modalias
drwxr-xr-x    2 root     root             0 Sep  2 10:47 power/
lrwxrwxrwx    1 root     root             0 Sep  2 10:47 subsystem -> ../../../../../../../bus/usb/
-r--r--r--    1 root     root          4096 Sep  2 10:47 supports_autosuspend
-rw-r--r--    1 root     root          4096 Sep  2 10:47 uevent

I reinstall the mod-rtl8192cu, the dirver doesn't come back.

brada4, 2023-09-02:
The listed device is USB port on computer side.

ehmad11, 2023-11-24:
i am also facing a similar issue, tried few pkgs but it is not working 

```
root@OpenWrt:~# opkg list-installed | grep rtl
kmod-rtl8xxxu - 5.15.134+6.1.24-3
librtlsdr - 0.6.0-3
rtl-sdr - 0.6.0-3

```
```
root@OpenWrt:~# lsusb 
Bus 001 Device 004: ID 0bda:f179 Realtek 802.11n
```

cat /sys/kernel/debug/usb/devices

```
T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=02 Dev#=  4 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=f179 Rev= 0.00
S:  Manufacturer=Realtek
S:  Product=802.11n
S:  SerialNumber=002E2D10BA99
C:* #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
```

brada4, 2023-11-24:
Does it work on other linux or windows?

ehmad11, 2023-11-24:
> Does it work on other linux or windows?

yes works fine out of box on fedora and windows

brada4, 2023-11-25:
please compare ethtool -i with fedora, probably also boot messages like firmware load.

ehmad11, 2023-11-25:
> please compare ethtool -i with fedora, probably also boot messages like firmware load.

fedora

```
sudo ethtool --driver wlp0s20f0u1
driver: rtl8xxxu
version: 6.5.12-200.fc38.x86_64
firmware-version: N/A
expansion-rom-version: 
bus-info: 1-1:1.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```
openwrt

```
root@OpenWrt:~# opkg list-installed | grep rtl
kmod-rtl8xxxu - 5.15.134+6.1.24-3
```

ehmad11, 2023-11-25:
fedora dmesg 

```
[ 6686.352357] usb 1-1: new high-speed USB device number 8 using xhci_hcd
[ 6686.478901] usb 1-1: New USB device found, idVendor=0bda, idProduct=f179, bcdDevice= 0.00
[ 6686.478905] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6686.478906] usb 1-1: Product: 802.11n
[ 6686.478907] usb 1-1: Manufacturer: Realtek
[ 6686.478909] usb 1-1: SerialNumber: 002E2D10BA99
[ 6686.520826] usb 1-1: RTL8188FU rev B (SMIC) romver 0, 1T1R, TX queues 2, WiFi=1, BT=0, GPS=0, HI PA=0
[ 6686.520830] usb 1-1: RTL8188FU MAC: 00:2e:2d:10:ba:99
[ 6686.520832] usb 1-1: rtl8xxxu: Loading firmware rtlwifi/rtl8188fufw.bin
[ 6686.522660] usb 1-1: Firmware revision 4.0 (signature 0x88f1)
[ 6687.564592] rtl8xxxu 1-1:1.0 wlp0s20f0u1: renamed from wlan0
[ 6688.545343] wlp0s20f0u1: authenticate with 8a:8c:b5:44:1b:52
[ 6688.545353] wlp0s20f0u1: 80 MHz not supported, disabling VHT
[ 6688.581822] wlp0s20f0u1: send auth to 8a:8c:b5:44:1b:52 (try 1/3)
[ 6688.583298] wlp0s20f0u1: authenticated
[ 6688.584401] wlp0s20f0u1: associate with 8a:8c:b5:44:1b:52 (try 1/3)
[ 6688.588018] wlp0s20f0u1: RX AssocResp from 8a:8c:b5:44:1b:52 (capab=0x411 status=0 aid=23)
[ 6688.588618] usb 1-1: rtl8xxxu_bss_info_changed: HT supported
[ 6688.589468] wlp0s20f0u1: associated
[ 6688.613089] wlp0s20f0u1: Limiting TX power to 30 (30 - 0) dBm as advertised by 8a:8c:b5:44:1b:52
```

brada4, 2023-11-25:
i was asking for differences fedora vs openwrt if possible

ehmad11, 2023-11-25:
> i was asking for differences fedora vs openwrt if possible

openwrt dmesg

```
[11130.803577] usb 1-1.2: new high-speed USB device number 7 using dwc_otg
[11130.934533] usb 1-1.2: New USB device found, idVendor=0bda, idProduct=f179, bcdDevice= 0.00
[11130.948096] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[11130.958198] usb 1-1.2: Product: 802.11n
[11130.964743] usb 1-1.2: Manufacturer: Realtek
[11130.971635] usb 1-1.2: SerialNumber: 002E2D10BA99
```

pls check now, let me know if you need anything else, appreciate your assistance, openwrt is running on headless raspberry pi

brada4, 2023-11-25:
Try snapshot, the USB ID is not present in kernel 5.15

hauke, 2023-11-25:
Support for RTL8188FU with idVendor=0bda, idProduct=8179 was added in this Linux commit: https://github.com/torvalds/linux/commit/3dfb8e844fa30cceb4b810613e2c35f628eb3e70 for Linux 6.3.

Support for RTL8188FU with idVendor=0bda, idProduct=f179 was added in this Linux commit https://github.com/torvalds/linux/commit/c888183b21f36a247bb166ca9365705611bea847 for Linux 6.2.

Both are included in wireless backports in OpenWrt master, but not in OpenWrt 23.05 and older.

closed 2023-11-25 by hauke

Minionguyjpro, 2024-05-20:
> Support for RTL8188FU with idVendor=0bda, idProduct=8179 was added in this Linux commit: [torvalds/linux@3dfb8e8](https://github.com/torvalds/linux/commit/3dfb8e844fa30cceb4b810613e2c35f628eb3e70) for Linux 6.3.
> 
> Support for RTL8188FU with idVendor=0bda, idProduct=f179 was added in this Linux commit [torvalds/linux@c888183](https://github.com/torvalds/linux/commit/c888183b21f36a247bb166ca9365705611bea847) for Linux 6.2.
> 
> Both are included in wireless backports in OpenWrt master, but not in OpenWrt 23.05 and older.

How could we use this on OpenWRT x86 today the day? I have 23.05.2 installed.

acoul, 2025-12-30:
unfortunately TP-Link TL-WN722N v2/v3 [Realtek **RTL8188EUS**] with `rtl8188eu-firmware` installed is not working on:

`Linux version 6.12.62 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 14.3.0 r32421-349040eeb5) 14.3.0, GNU ld (GNU Binutils) 2.44) #0 SMP Mon Dec 29 10:11:29 2025`

lsusb:
```
Bus 001 Device 001: ID 1d6b:0002 Linux 6.12.62 xhci-hcd xHCI Host Controller
Bus 001 Device 003: ID 2357:010c Realtek 802.11n NIC
Bus 002 Device 001: ID 1d6b:0003 Linux 6.12.62 xhci-hcd xHCI Host Controller
```