openwrt/openwrt #13379: problem with MT7615 5G, slow on WAN but normal on LAN to LAN and 2.4G

zcy85611, 2023-08-30:
### Describe the bug

I just compiled the latest firmware and did speedtest, weird thing is, with default setting, WAN speed of 5G is only about 40mbps-100mbps(wifi connection speed is 866mbps, country AU, channel 36, 80MHz), but with iperf3 lan to lan test, it can go up to 450mbps. 2.4G, WAN speed is normal, goes up to 150mbps.

I've already tried flashing bakcup eeprom from the stock firmware, still the same

 I cannot figure out why is this happening, have anyone met this problem?

`Connecting to host 192.168.3.1, port 5201
[  4] local 192.168.3.223 port 12320 connected to 192.168.3.1 port 5201
[  6] local 192.168.3.223 port 12321 connected to 192.168.3.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  28.8 MBytes   241 Mbits/sec
[  6]   0.00-1.00   sec  25.6 MBytes   214 Mbits/sec
[SUM]   0.00-1.00   sec  54.4 MBytes   455 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   1.00-2.00   sec  13.2 MBytes   111 Mbits/sec
[  6]   1.00-2.00   sec  11.0 MBytes  92.4 Mbits/sec
[SUM]   1.00-2.00   sec  24.2 MBytes   204 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   2.00-3.00   sec  29.0 MBytes   243 Mbits/sec
[  6]   2.00-3.00   sec  25.1 MBytes   210 Mbits/sec
[SUM]   2.00-3.00   sec  54.1 MBytes   453 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   3.00-4.00   sec  28.2 MBytes   237 Mbits/sec
[  6]   3.00-4.00   sec  25.8 MBytes   216 Mbits/sec
[SUM]   3.00-4.00   sec  54.0 MBytes   454 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   4.00-5.00   sec  30.6 MBytes   256 Mbits/sec
[  6]   4.00-5.00   sec  27.4 MBytes   229 Mbits/sec
[SUM]   4.00-5.00   sec  58.0 MBytes   485 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   5.00-6.00   sec  27.2 MBytes   229 Mbits/sec
[  6]   5.00-6.00   sec  24.2 MBytes   204 Mbits/sec
[SUM]   5.00-6.00   sec  51.5 MBytes   433 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   6.00-7.00   sec  28.2 MBytes   237 Mbits/sec
[  6]   6.00-7.00   sec  25.2 MBytes   212 Mbits/sec
[SUM]   6.00-7.00   sec  53.5 MBytes   450 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   7.00-8.00   sec  29.0 MBytes   243 Mbits/sec
[  6]   7.00-8.00   sec  25.0 MBytes   210 Mbits/sec
[SUM]   7.00-8.00   sec  54.0 MBytes   453 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   8.00-9.00   sec  28.1 MBytes   235 Mbits/sec
[  6]   8.00-9.00   sec  25.4 MBytes   212 Mbits/sec
[SUM]   8.00-9.00   sec  53.5 MBytes   448 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   9.00-10.01  sec  28.8 MBytes   240 Mbits/sec
[  6]   9.00-10.01  sec  25.2 MBytes   211 Mbits/sec
[SUM]   9.00-10.01  sec  54.0 MBytes   452 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.01  sec   271 MBytes   227 Mbits/sec                  sender
[  4]   0.00-10.01  sec   271 MBytes   227 Mbits/sec                  receiver
[  6]   0.00-10.01  sec   240 MBytes   201 Mbits/sec                  sender
[  6]   0.00-10.01  sec   240 MBytes   201 Mbits/sec                  receiver
[SUM]   0.00-10.01  sec   511 MBytes   429 Mbits/sec                  sender
[SUM]   0.00-10.01  sec   511 MBytes   429 Mbits/sec                  receiver`

### OpenWrt version

r23841-33046d2a47

### OpenWrt target/subtarget

ramips/mt7621

### Device

RAISECOM MSG1500 X.00

### Image kind

Self-built image

### Expected behaviour

5G WAN speed should be normal which is around 450mpbs

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-08-30:
Run channel survey on router itself, or externally like ubiquiti wifiman on mobile)
If it is somewhat crowded run either ch 149 or auto to get away from them.
Is your wifi client also set to AU regulatory domain? Say NZ there would dictate much lower tx power than due.
Also mind the distance, 5GHz is damped by walls much more than 2GHz.

Running iperf on router is not indicative of performance, you need to run it on a separate computer in LAN via switch.

zcy85611, 2023-08-31:
> Slow LAN -> WAN - enable firewall offload.

I've already enabled the NAT offload, I forgot to mention.

brada4, 2023-08-31:
Tune wifi environment first - almost free channel + proper country code in both ends.

Measuring CPU load like
```
while sleep 5 ; do
 uptime
done
```

1/ test like fast.com / speedtest.net / www.waveform.com/tools/bufferbloat (should be half gig or subscribed rate)
2/ iperf faulty LAN to device on ethernet switch (at least half gig)
3/ iperf LAN<->LAN (should be near gigabit both ways as no CPU is involved)

The description is quite vague on what you measured.

benoe77, 2024-01-23:
I have the same issue on an Asus AC65P, latest snapshot. My WAN connection 1000 / 300 Mbps, PPPoE. WAN-LAN download is around 970Mbps. WLAN-LAN througput is fast on AC (~500Mbps) so the wifi coverage is good, WLAN-WAN is much slower (maximum 120Mbps). Can be that the CPU is not beefy enough to handle PPPoE with more than 100Mbps. WAN-LAN?

brada4, 2024-01-23:
Can you copy waveform link? You have two beefs for CPU, should be ok.

benoe77, 2024-01-23:
mobile wifi 5g:
https://www.waveform.com/tools/bufferbloat?test-id=37d8d96a-d5f4-448b-9fe6-eb57ba49a366

mobile ethernet lan:
https://www.waveform.com/tools/bufferbloat?test-id=98b0f0db-dced-4a64-b2fb-ac28d5945fc0

notebook wifi 5g:
https://www.waveform.com/tools/bufferbloat?test-id=2f089c36-c05c-4adb-b422-3869cc4dac67

notebook ethernet lan:
https://www.waveform.com/tools/bufferbloat?test-id=ade9f9bf-3ed5-46fe-8f14-ae94dfb13a07

brada4, 2024-01-23:
So it actually forwards fine to pppoe gpon but gets super slow using backup mobile network connection?

benoe77, 2024-01-23:
> So it actually forwards fine to pppoe gpon but gets super slow using backup mobile network connection?

No. I tested the behaviour on two devices: a mobile phone and a notebook. I connected both devices to the router first via 5gwifi, did a measure then via ethernet cable, did a second measure. For mobile I used an usb-ethernet dongle.

The gpon is bridged, so the ac65p is doing the pppoe connection.

brada4, 2024-01-23:
So wifi is slow? According to spec maximum is 433Mbps per antenna indicated as link speed in either device (433 or 866), also you se VHT-NSS 1 or VHT-NSS 2 in association list by end of first page of luci.
Acceptable TCP speed should be 69/70 usable link -5% for TCP ACKs like 400Mbps 50MB/s (and double thereof)

Basic setup:
Set country code
Set channel to automatic
Check `iw reg get` - if you have DFS channels weaker than no-DFS ones in your area you might want to add 5GHz radio setting to keep away from them:
        list hostapd_options 'min_tx_power=20'
If you ever changed reg domain from one country to other while device is still powered you need to power device off completely for first programming attempt on poweron to set up radio correctly.
You can try software flow offload but that often damages wifi roaming 2.4 <-> 5 

Now check that tx-power is at least 20 (iwinfo or in luci/network/wireless/edit AP), and check that signal bar and link speed toggles betweeen 6mbps and max on the router associations list.
Now try speed test - ? Since devices behave same, just use one easier for you.
Enable soft offload - another test.
Go system/interfaces/global and flip packet steering flag - test again
Disable flow offload - test again.
Flip back packet steering.

Draw conclusion which is faster. Less flipped boxes better.

benoe77, 2024-01-24:
There should be no problem with the 5g connection itself, because when I run an iperf test to a local server I get perfectly fine results:

```
iperf3 -c 192.168.1.10 -R
Connecting to host 192.168.1.10, port 5201
Reverse mode, remote host 192.168.1.10 is sending
[  5] local 192.168.1.2 port 51070 connected to 192.168.1.10 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  68.2 MBytes   571 Mbits/sec                  
[  5]   1.00-2.00   sec  68.8 MBytes   578 Mbits/sec                  
[  5]   2.00-3.00   sec  58.6 MBytes   492 Mbits/sec                  
[  5]   3.00-4.00   sec  53.1 MBytes   446 Mbits/sec                  
[  5]   4.00-5.00   sec  56.2 MBytes   472 Mbits/sec                  
[  5]   5.00-6.00   sec  62.2 MBytes   522 Mbits/sec                  
[  5]   6.00-7.00   sec  68.1 MBytes   571 Mbits/sec                  
[  5]   7.00-8.00   sec  67.2 MBytes   564 Mbits/sec                  
[  5]   8.00-9.00   sec  68.2 MBytes   572 Mbits/sec                  
[  5]   9.00-10.00  sec  68.5 MBytes   575 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.04  sec   642 MBytes   537 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   639 MBytes   536 Mbits/sec                  receiver

iperf Done.
```

Only when I measure to a remote server through the pppoe connection I get the slowdown:

remote iperf connected through ethernet:

`iperf3 -c [rychlost.poda.cz](http://rychlost.poda.cz/) -R
Connecting to host rychlost.poda.cz, port 5201
Reverse mode, remote host [rychlost.poda.cz](http://rychlost.poda.cz/) is sending
[  5] local 192.168.1.2 port 45082 connected to 85.135.32.105 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  6.50 MBytes  54.4 Mbits/sec                  
[  5]   1.00-2.00   sec  24.2 MBytes   204 Mbits/sec                  
[  5]   2.00-3.00   sec  76.4 MBytes   641 Mbits/sec                  
[  5]   3.00-4.00   sec   112 MBytes   936 Mbits/sec                  
[  5]   4.00-5.00   sec  92.8 MBytes   778 Mbits/sec                  
[  5]   5.00-6.00   sec  84.9 MBytes   712 Mbits/sec                  
[  5]   6.00-7.00   sec  88.5 MBytes   743 Mbits/sec                  
[  5]   7.00-8.00   sec  90.2 MBytes   757 Mbits/sec                  
[  5]   8.00-9.00   sec  93.2 MBytes   783 Mbits/sec                  
[  5]   9.00-10.00  sec  94.9 MBytes   796 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   770 MBytes   646 Mbits/sec   87             sender
[  5]   0.00-10.00  sec   763 MBytes   640 Mbits/sec                  receiver

iperf Done.`

remote iperf connected through wifi5g:

`iperf3 -c [rychlost.poda.cz](http://rychlost.poda.cz/) -R
Connecting to host rychlost.poda.cz, port 5201
Reverse mode, remote host [rychlost.poda.cz](http://rychlost.poda.cz/) is sending
[  5] local 192.168.1.2 port 35340 connected to 85.135.32.105 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.03   sec  8.38 MBytes  67.9 Mbits/sec                  
[  5]   1.03-2.01   sec  15.5 MBytes   134 Mbits/sec                  
[  5]   2.01-3.01   sec  16.5 MBytes   138 Mbits/sec                  
[  5]   3.01-4.01   sec  14.9 MBytes   125 Mbits/sec                  
[  5]   4.01-5.01   sec  13.6 MBytes   113 Mbits/sec                  
[  5]   5.01-6.00   sec  12.6 MBytes   107 Mbits/sec                  
[  5]   6.00-7.00   sec  13.4 MBytes   112 Mbits/sec                  
[  5]   7.00-8.00   sec  14.1 MBytes   119 Mbits/sec                  
[  5]   8.00-9.00   sec  14.5 MBytes   122 Mbits/sec                  
[  5]   9.00-10.01  sec  15.5 MBytes   129 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec   141 MBytes   118 Mbits/sec    3             sender
[  5]   0.00-10.01  sec   139 MBytes   117 Mbits/sec                  receiver

iperf Done.`

brada4, 2024-01-24:
So forwarding wifi->pppoe-wan is slow. Latency seems good.
What if you substitute WAN adapter like eth0 to pppoe-wan in firewall zone?
Seems very related to #10224 it is old issue still not addressed.

benoe77, 2024-01-24:
When I disabled sw and hw off-load, I get similarly slow results through lan as I got through wifi5g->pppoe-wan

brada4, 2024-01-24:
Check `fw4 print` - what are WAN interface(s) vs offload ones?

benoe77, 2024-01-24:
`	flowtable ft {
		hook ingress priority 0;
		devices = { "lan1", "lan2", "lan3", "lan4", "phy0-ap0", "phy1-ap0", "pppoe-wan", "wan" };
		counter;
		flags offload;
	}
`

...

`define wan_devices = { "pppoe-wan", "wan" }`

benoe77, 2024-01-24:
So basically I need to remove 'wan' from this list?

`        flowtable ft {
                hook ingress priority filter
                devices = { lan1, lan2, lan3, lan4, phy0-ap0, phy1-ap0, pppoe-wan, wan }
                flags offload
                counter
        }
`
I'm not at the router right now, so cannot really test atm, but will try to find some time later

brada4, 2024-01-24:
That is one of points in linked thread.

benoe77, 2024-01-25:
> So basically I need to remove 'wan' from this list?
> 
> ` flowtable ft { hook ingress priority filter devices = { lan1, lan2, lan3, lan4, phy0-ap0, phy1-ap0, pppoe-wan, wan } flags offload counter }` I'm not at the router right now, so cannot really test atm, but will try to find some time later

I went through and removed all the devices one by one: no change in speed on wifi5g. The only change was when I removed lan1, my notebooks ethernet speed that I used as a reference went down from gigabit to around 150mbps.

I even tried to add eth0 and to replace pppoe-wan with eth0, no change.

brada4, 2024-01-25:
Try to permit offload between br-lan and pppoe-wan only (the "internet" and the "lan"). Omit `flags offload;` because both are software interfaces, it will play no role here whatsoever.
According to documentation they should reasonably pick up packets on 1 step lower more encapsulated interfaces.

benoe77, 2024-01-25:
> Try to permit offload between br-lan and pppoe-wan only (the "internet" and the "lan"). Omit `flags offload;` because both are software interfaces, it will play no role here whatsoever. According to documentation they should reasonably pick up packets on 1 step lower more encapsulated interfaces.

Please elaborate exactly what I need to do by commands or luci? Should I put only br-lan and pppoe-wan into the above line and remove everything else?

brada4, 2024-01-25:
Disable hardware offload.
Remove wan interface from wan firewall zone leaving pppoe-wan only.
Edit ruleset dump and limit offload group to 2 interfaces.
pppoe-wan and  br-lan (2nd is not there) 
```
        flowtable ft {
                hook ingress priority filter
                devices = {pppoe-wan,br-lan }
                counter
        }
```

benoe77, 2024-01-25:
In my wan firewall zone "wan" and "wan6" zones are covered (checked in luci). I cannot select pppoe-wan there, but put it as "custom"

brada4, 2024-01-25:
Then it will be a bit more complicated editing of ruleset dump removing wan from every place it  is paired with pppoe-wan.
But first try the soft offload between wan and lan.

benoe77, 2024-01-25:
By disabling hw offload and putting only pppoe-wan and br-lan into the device list, I saw no change in wifi5g speed, gigabit ethernet speed has decreased to around 140Mbps

brada4, 2024-01-25:
Check with `ip ro sh` what is the route to pppoe gateway. `src some ip` type is somewhat slower than `via some ip` or/together `dev wan`.

brada4, 2024-01-25:
(the guess-pinching comes from hard to tracke ksoftirq doing added work from interrupts like nftables and qdisc and kworker dealing with rule expiration and other cleanup)

benoe77, 2024-01-26:
> Check with `ip ro sh` what is the route to pppoe gateway. `src some ip` type is somewhat slower than `via some ip` or/together `dev wan`. EDIT this goes besides normal firewall rules with offload restored, as offloads are able (not guaranteed) to bypass routing tables.

`ip ro sh
default via 10.0.0.1 dev pppoe-wan 
10.0.0.1 dev pppoe-wan scope link  src xxx.xxx.xxx.xxx 
192.168.1.0/24 dev br-lan scope link  src 192.168.1.1`

brada4, 2024-01-26:
`ip route add PPPOE_SERVER_IP/32  (dev wan) (src LOCAL_IP_ON_WAN)`

brada4, 2024-01-26:
Try both parentheses, I think dev is used by wg & co.

benoe77, 2024-01-26:
PPPOE_SERVER_IP is 10.0.0.1 or 192.168.1.0 from my example? And LOCAL_IP_ON_WAN?

brada4, 2024-01-26:
`ip route add 10.0.0.1/32 dev wan src (what you have there as src xyz.abc.....)`

benoe77, 2024-01-26:
`ip: RTNETLINK answers: File exists`

brada4, 2024-01-26:
```
ip ro del 10.0.0.1 dev pppoe-wan scope link
ip ro add 10.0.0.1 dev wan src XXX.YYY.ZZZ scope link
```
And use 3x backticks to format code blocks

brada4, 2024-01-26:
Another cpu eater is qdisc
(package tc-tiny)
```
tc -s qdisc 
should be fq_codel at pppoe-wan and noqueua at wan
tc qdisc replace  dev wan  root noqueue
```

benoe77, 2024-01-26:
Thanks, I went through all these settings, but unfortunately none of them gave significant (if any) results.

brada4, 2024-01-26:
You need to restart firewall to re-detect offload interfaces and destroy ancient states with `conntrack -D`

brada4, 2024-01-27:
@jow- there is obvious wrong that pppoe lower route is wrongly added to upper interface. Does not impact (offload) performance though.

brada4, 2024-01-29:
@benoe77 small discovery - as long as hw_offload states are up - fw4 is not able to detect if interface can offload at all, so you need to disable offload , then wait 60 seconds for offload states to get back to conntrack , and then enable soft offload. (or more accurately wait for hw_offload states to disappear in `conntrack -L -u offload`) add route correction any time before re-enabling offload.

benoe77, 2024-01-29:
@brada4  thanks for all your help, I went and started over setting up my router from zero. Now I can get ~300mbps via 5g wifi and 1gbps via ethernet, with hw offload on. Hopefully it will stay like this.

brada4, 2024-01-29:
Can you observe with htop, if CPU is fully used it may speed up wifi to full 400-few Mbps by adjusting route then restarting (service restart or disable/apply/enable/apply) soft offload firewall (hard offload is not trivial to get around as I mentioned)

khanjui, 2024-02-03:
> Another cpu eater is qdisc (package tc-tiny)
> 
> ```
> tc -s qdisc 
> ```
> 
> should be fq_codel at pppoe-wan and noqueue at wan
> 
> ```
> tc qdisc replace  dev wan  root noqueue
> ```

Just wondering, do we really need fq_codel on any other interfaces if we are using CAKE on pppoe-wan? Setting them as noque should be better in theory or not?

brada4, 2024-02-03:
Default is good for 99% of devices, but for other 1% the option to change default would be helpful.
sysctl sometimes helps sometimes (pppoe) not