openwrt/openwrt #13384: Less than ideal advice in the OpenVPN Extras "DNS and domain" section

marti4d, 2023-08-31:
### Describe the bug

Hey there!

I'm using OpenWRT to create a wireless hotspot that tunnels all traffic to a VPN server, and I found the advice very helpful from the "OpenVPN extas" > "DNS and domain" section [here](https://openwrt.org/docs/guide-user/services/vpn/openvpn/extras#dns_and_domain).

Unfortunately, an issue I ran into is that the approach relies on UCI maintaining a pending change that never gets committed. This means that if the user does anything that invokes a "uci commit" later on, they will accidentally persist the setting that points DNSMasq resolvfile at the OpenVPN resolver across reboots.

That causes DNS to be unusable after reboot, which is a problem if the VPN server is identified by domain name.

I actually wrote a little init script below that works with the scripts above to ensure that a reboot also restores the "auto" resolver.

`/etc/init.d/resolv_restore`

```sh
#!/bin/sh /etc/rc.common

START=18

boot() {
  uci set dhcp.@dnsmasq[0].resolvfile="/tmp/resolv.conf.d/resolv.conf.auto"
  uci commit
}
```

And it has to be enabled using:

```sh
> /etc/init.d/resolv_restore enable
```

### OpenWrt version

22.03.5

### OpenWrt target/subtarget

all

### Device

all

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

marti4d, 2023-09-06:
@egc112 This seems like it would also have the same issue though? If the user just powers off the router without a proper shutdown, won't DNSMasq be stuck pointing to the VPN resolvfile after next boot?

egc112, 2023-09-07:
I have tested it by cutting the power and it does not seem to have the problem you described.