openwrt/openwrt #13391: Problem with WAN ports going down when network restarts in fortigate 50e.

odkrys, 2023-09-01:
### Describe the bug

When I run "service network restart" in CLI, wan ports go down.
@musashino205  Could you take a look?

### OpenWrt version

r23389-5deed175a5

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Fortinet FortiGate 50E

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mrkiko, 2023-09-02:
May you detail little bit more the Actual Behaviour and the Expected one?
Thanks!

musashino205, 2023-09-02:
I tested on my FortiGate 50E and I found an interrupt issue on 88E1512 PHYs...
Interrupts on both PHYs work without problems after booting, but if you restart "network" service, interrupts stop working, won't be sumitted to CPU and kernel can't detect link changing on PHYs.

- SNAPSHOT, r23856-9a4bd3b1e4

```
root@OpenWrt:/# cat /proc/interrupts 
           CPU0       CPU1       
 25:          0          0     GIC-0  27 Edge      gt
 26:      16478      31673     GIC-0  29 Edge      twd
 27:          0          0      MPIC   5 Level     armada_370_xp_per_cpu_tick
 29:        158          0     GIC-0  34 Level     mv64xxx_i2c
 30:        241          0     GIC-0  44 Level     ttyS0
 40:          0          0     GIC-0  41 Level     f1020300.watchdog
 44:          0          0     GIC-0  96 Level     f1020300.watchdog
 45:         17          0      MPIC   8 Level     eth0
 46:          6          0      MPIC  10 Level     eth1
 47:          6          0      MPIC  12 Level     eth2
 48:          0          0     GIC-0  51 Level     f1090000.crypto
 49:          0          0     GIC-0  52 Level     f1090000.crypto
 50:          0          0     GIC-0  53 Level     f10a3800.rtc
 53:          2          0     GIC-0  54 Level     f1060800.xor
 54:          2          0     GIC-0  97 Level     f1060900.xor
 55:          2          0  f1018100.gpio  20 Level     f1072004.mdio-mii:00
 56:          2          0  f1018140.gpio   9 Level     f1072004.mdio-mii:01
 60:          0          0  mv88e6xxx-g1   3 Edge      mv88e6xxx-f1072004.mdio-mii:02-g1-atu-prob
 62:          0          0  mv88e6xxx-g1   5 Edge      mv88e6xxx-f1072004.mdio-mii:02-g1-vtu-prob
 64:          0          0  mv88e6xxx-g1   7 Edge      mv88e6xxx-f1072004.mdio-mii:02-g2
 66:          0          0  mv88e6xxx-g2   0 Edge      mv88e6xxx-1:00
 67:          0          0  mv88e6xxx-g2   1 Edge      mv88e6xxx-1:01
 68:          0          0  mv88e6xxx-g2   2 Edge      mv88e6xxx-1:02
 69:          0          0  mv88e6xxx-g2   3 Edge      mv88e6xxx-1:03
 70:          0          0  mv88e6xxx-g2   4 Edge      mv88e6xxx-1:04
 81:          0          0  mv88e6xxx-g2  15 Edge      mv88e6xxx-f1072004.mdio-mii:02-watchdog
 82:          0          0  f1018140.gpio  22 Edge      gpio-keys
IPI0:          0          1  CPU wakeup interrupts
IPI1:          0          0  Timer broadcast interrupts
IPI2:        496        516  Rescheduling interrupts
IPI3:       7753       7204  Function call interrupts
IPI4:          0          0  CPU stop interrupts
IPI5:          0          0  IRQ work interrupts
IPI6:          0          0  completion interrupts
Err:          0
root@OpenWrt:/# grep "f1072004.mdio-mii" /proc/interrupts
 55:          2          0  f1018100.gpio  20 Level     f1072004.mdio-mii:00
 56:          2          0  f1018140.gpio   9 Level     f1072004.mdio-mii:01
 60:          0          0  mv88e6xxx-g1   3 Edge      mv88e6xxx-f1072004.mdio-mii:02-g1-atu-prob
 62:          0          0  mv88e6xxx-g1   5 Edge      mv88e6xxx-f1072004.mdio-mii:02-g1-vtu-prob
 64:          0          0  mv88e6xxx-g1   7 Edge      mv88e6xxx-f1072004.mdio-mii:02-g2
 81:          0          0  mv88e6xxx-g2  15 Edge      mv88e6xxx-f1072004.mdio-mii:02-watchdog
root@OpenWrt:/# [  107.724448] mv88e6085 f1072004.mdio-mii:02 lan3: Link is Up - 100Mbps/Full - flow control rx/tx
[  107.828782] br-lan: port 3(lan3) entered blocking state
[  107.891412] br-lan: port 3(lan3) entered forwarding state
[  107.957221] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
root@OpenWrt:/# grep "f1072004.mdio-mii" /proc/interrupts
 55:          2          0  f1018100.gpio  20 Level     f1072004.mdio-mii:00
 56:          2          0  f1018140.gpio   9 Level     f1072004.mdio-mii:01
 60:          0          0  mv88e6xxx-g1   3 Edge      mv88e6xxx-f1072004.mdio-mii:02-g1-atu-prob
 62:          0          0  mv88e6xxx-g1   5 Edge      mv88e6xxx-f1072004.mdio-mii:02-g1-vtu-prob
 64:          0          1  mv88e6xxx-g1   7 Edge      mv88e6xxx-f1072004.mdio-mii:02-g2
 81:          0          0  mv88e6xxx-g2  15 Edge      mv88e6xxx-f1072004.mdio-mii:02-watchdog
root@OpenWrt:/# [  123.611472] br-lan: port 3(lan3) entered disabled state
[  123.674870] mv88e6085 f1072004.mdio-mii:02 lan3: Link is Down
[  125.162399] mvneta f1030000.ethernet eth1: Link is Up - 100Mbps/Full - flow control rx/tx
[  125.260477] br-wan: port 1(eth1) entered blocking state
[  125.323101] br-wan: port 1(eth1) entered forwarding state

root@OpenWrt:/# grep "f1072004.mdio-mii" /proc/interrupts
 55:          3          0  f1018100.gpio  20 Level     f1072004.mdio-mii:00
 56:          2          0  f1018140.gpio   9 Level     f1072004.mdio-mii:01
[... 83 lines cut ...]
 64:          0          2  mv88e6xxx-g1   7 Edge      mv88e6xxx-f1072004.mdio-mii:02-g2
 81:          0          0  mv88e6xxx-g2  15 Edge      mv88e6xxx-f1072004.mdio-mii:02-watchdog
root@OpenWrt:/# grep "f1072004.mdio-mii" /proc/interrupts
 55:          4          0  f1018100.gpio  20 Level     f1072004.mdio-mii:00
 56:          3          0  f1018140.gpio   9 Level     f1072004.mdio-mii:01
 60:          0          0  mv88e6xxx-g1   3 Edge      mv88e6xxx-f1072004.mdio-mii:02-g1-atu-prob
 62:          0          0  mv88e6xxx-g1   5 Edge      mv88e6xxx-f1072004.mdio-mii:02-g1-vtu-prob
 64:          0          2  mv88e6xxx-g1   7 Edge      mv88e6xxx-f1072004.mdio-mii:02-g2
[... cut ...]
 55:          4          0  f1018100.gpio  20 Level     f1072004.mdio-mii:00
 56:          3          0  f1018140.gpio   9 Level     f1072004.mdio-mii:01
 60:          0          0  mv88e6xxx-g1   3 Edge      mv88e6xxx-f1072004.mdio-mii:02-g1-atu-prob
 62:          0          0  mv88e6xxx-g1   5 Edge      mv88e6xxx-f1072004.mdio-mii:02-g1-vtu-prob
 64:          0          2  mv88e6xxx-g1   7 Edge      mv88e6xxx-f1072004.mdio-mii:02-g2
 81:          0          0  mv88e6xxx-g2  15 Edge      mv88e6xxx-f1072004.mdio-mii:02-watchdog
root@OpenWrt:/#
```

musashino205, 2023-09-02:
I'll remove interrupts from both PHYs and use polling instead.

odkrys, 2023-09-02:
Thanks for fixing it.

musashino205, 2023-09-02:
I did some additional tests. Then, both PHYs don't seem to be re-configured correctly after HW resetting by RESET line of each PHYs. So this is not a interrupt issue.
At least the interface mode (page 18,  register 20) that should have been set to SGMII (```0b001```) was set to "Reserved" (```0b111```, default of HW Reset).

musashino205, 2023-09-02:
By adding ```reset-(de)assert-us = <10000>;``` to each PHYs, this issue is solved on my FortiGate 50E.

@odkrys Can you test attached firmware on your 50E?

[openwrt-mvebu-cortexa9-fortinet_fg-50e.zip](https://github.com/openwrt/openwrt/files/12504158/openwrt-mvebu-cortexa9-fortinet_fg-50e.zip)

odkrys, 2023-09-02:
I just installed it and tested it. I confirmed that WAN was reconnected.

commit 5c65224d (2023-09-03): mvebu: add reset delays of PHYs for Fortinet FortiGate 50E

Add reset-(de)assert-us to ethernet PHYs on Fortinet FortiGate 50E to
solve instability after HW resetting of PHYs. (ex.: restarting "network"
service, etc...)

Fixes: #13391
Fixes: 102dc5a62506 ("mvebu: add support for Fortinet FortiGate 50E")
commit 68972704 (2023-09-16): mvebu: add reset delays of PHYs for Fortinet FortiGate 50E

Add reset-(de)assert-us to ethernet PHYs on Fortinet FortiGate 50E to
solve instability after HW resetting of PHYs. (ex.: restarting "network"
service, etc...)

Fixes: #13391
Fixes: 102dc5a62506 ("mvebu: add support for Fortinet FortiGate 50E")

(cherry picked from commit 5c65224d8f5787f6bdfaec9547637f1f2f46354f)
PR #13450 (2023-09-16): [23.05] mvebu: add reset delays of PHYs for Fortinet FortiGate 50E
Add reset-(de)assert-us to ethernet PHYs on Fortinet FortiGate 50E to solve instability after HW resetting of PHYs. (ex.: restarting "network" service, etc...)

Fixes: #13391
Fixes: 102dc5a62506 ("mvebu: add support for Fortinet FortiGate 50E")

(cherry picked from commit 5c65224d8f5787f6bdfaec9547637f1f2f46354f)
closed 2023-09-02 by odkrys