openwrt/openwrt #13395: Issues with WCN6855 / NFA765

jordipalet, 2023-09-02:
### Describe the bug

Trying to use the NFA765 M.2 in a mini PC (386/64) with Proxmox. I got other WiFi cards working so it seems related to this one specifically, not the passthru config, etc.

Here is the relevant info:

`[    0.275842] pci 0000:06:03.0: [1af4:1002] type 00 class 0x00ff00
[    0.282123] pci 0000:06:03.0: reg 0x10: [io  0x9040-0x907f]
[    0.284034] pci 0000:06:03.0: reg 0x20: [mem 0x383800004000-0x383800007fff 64bit pref]
[    0.292119] pci 0000:06:08.0: [1af4:1003] type 00 class 0x078000
[    0.292119] pci 0000:06:08.0: reg 0x10: [io  0x9000-0x903f]
[    0.292119] pci 0000:06:08.0: reg 0x14: [mem 0xc1d0c000-0xc1d0cfff]
[    0.302128] pci 0000:06:08.0: reg 0x20: [mem 0x383800000000-0x383800003fff 64bit pref]
[    0.302128] pci 0000:06:10.0: [17cb:1103] type 00 class 0x028000
[    0.302574] pci 0000:06:10.0: reg 0x10: [mem 0xc1800000-0xc19fffff 64bit]
[    0.312122] pci 0000:06:11.0: [8086:125c] type 00 class 0x020000
[    0.312122] pci 0000:06:11.0: reg 0x10: [mem 0xc1c00000-0xc1cfffff]
[    0.313690] pci 0000:06:11.0: reg 0x1c: [mem 0xc1d08000-0xc1d0bfff]
[    0.322139] pci 0000:06:11.0: reg 0x30: [mem 0xfff00000-0xffffffff pref]
[    0.322139] pci 0000:06:1b.0: [8086:125c] type 00 class 0x020000
[    0.322139] pci 0000:06:1b.0: reg 0x10: [mem 0xc1b00000-0xc1bfffff]
[    0.332121] pci 0000:06:1b.0: reg 0x1c: [mem 0xc1d04000-0xc1d07fff]
[    0.332121] pci 0000:06:1b.0: reg 0x30: [mem 0xfff00000-0xffffffff pref]
[    0.332121] pci 0000:06:1c.0: [8086:125c] type 00 class 0x020000
[    0.332121] pci 0000:06:1c.0: reg 0x10: [mem 0xc1a00000-0xc1afffff]
[    0.342132] pci 0000:06:1c.0: reg 0x1c: [mem 0xc1d00000-0xc1d03fff]
[    0.342132] pci 0000:06:1c.0: reg 0x30: [mem 0xfff00000-0xffffffff pref]
[    0.504796] pci 0000:06:11.0: can't claim BAR 6 [mem 0xfff00000-0xffffffff pref]: no compatible bridge window
[    0.505523] pci 0000:06:1b.0: can't claim BAR 6 [mem 0xfff00000-0xffffffff pref]: no compatible bridge window
[    0.506242] pci 0000:06:1c.0: can't claim BAR 6 [mem 0xfff00000-0xffffffff pref]: no compatible bridge window
[    0.536322] pci 0000:06:11.0: BAR 6: no space for [mem size 0x00100000 pref]
[    0.536843] pci 0000:06:11.0: BAR 6: failed to assign [mem size 0x00100000 pref]
[    0.537393] pci 0000:06:1b.0: BAR 6: no space for [mem size 0x00100000 pref]
[    0.537910] pci 0000:06:1b.0: BAR 6: failed to assign [mem size 0x00100000 pref]
[    0.538452] pci 0000:06:1c.0: BAR 6: no space for [mem size 0x00100000 pref]
[    0.538969] pci 0000:06:1c.0: BAR 6: failed to assign [mem size 0x00100000 pref]
[    4.293071] ath11k_pci 0000:06:10.0: BAR 0: assigned [mem 0xc1800000-0xc19fffff 64bit]
[    4.296003] ath11k_pci 0000:06:10.0: MSI vectors: 1
[    4.296941] ath11k_pci 0000:06:10.0: wcn6855 hw2.0
[    4.297742] ath11k_pci 0000:06:10.0: FW memory mode: 0
[    5.181522] ath11k_pci 0000:06:10.0: chip_id 0x2 chip_family 0xb board_id 0xff soc_id 0x400c0200
[    5.182213] ath11k_pci 0000:06:10.0: fw_version 0x110b196e fw_build_timestamp 2022-12-22 12:54 fw_build_id WLAN.HSP.1.1-03125-QCAHSPSWPL_V1_V2_SILICONZ_LITE-3.6510.23
[    5.250981] ath11k_pci 0000:06:10.0: leaving PCI ASPM disabled to avoid MHI M2 problems
    6.282238] ath11k_pci 0000:06:10.0: failed to receive control response completion, polling..
[    7.322393] ath11k_pci 0000:06:10.0: Service connect timeout
[    7.323314] ath11k_pci 0000:06:10.0: failed to connect to HTT: -110
[    7.324425] ath11k_pci 0000:06:10.0: failed to start core: -110
[    7.534311] ath11k_pci 0000:06:10.0: ignore reset dev flags 0x4000
[    7.642295] ath11k_pci 0000:06:10.0: ignore reset dev flags 0x4000
[   17.802376] ath11k_pci 0000:06:10.0: failed to wait wlan mode request (mode 4): -110
[   17.803426] ath11k_pci 0000:06:10.0: qmi failed to send wlan mode off: -110
`

It looks to me that the last few lines tell why, but don't have the driver knowledge to be able to debug it ...

Any hints?

### OpenWrt version

DISTRIB_REVISION='r23389-5deed175a5'

### OpenWrt target/subtarget

DISTRIB_REVISION='r23389-5deed175a5'

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Steps to reproduce

Just install ath11 packages and download firmware into 
 /lib/firmware/ath11k/WCN6855/hw2.0/

### Actual behaviour

see logs in 1st section

### Expected behaviour

wifi config doesn't work, device not identified, but I can see it in lspci

06:10.0 Network controller: Qualcomm QCNFA765 Wireless Network Adapter (rev 01)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-02:
> can't claim BAR

It us exactly incomplete passthrough. Try various APIC settings like x2 etc in bios, maybe you succeed with BAR remapping.

jordipalet, 2023-09-02:
However, it seems after several attempts this worked or I'm missing something?

[    4.293071] ath11k_pci 0000:06:10.0: BAR 0: assigned [mem 0xc1800000-0xc19fffff 64bit] [    4.296003] ath11k_pci 0000:06:10.0: MSI vectors: 1 [    4.296941] ath11k_pci 0000:06:10.0: wcn6855 hw2.0 [    4.297742] ath11k_pci 0000:06:10.0: FW memory mode: 0 [    5.181522] ath11k_pci 0000:06:10.0: chip_id 0x2 chip_family 0xb board_id 0xff soc_id 0x400c0200 [    5.182213] ath11k_pci 0000:06:10.0: fw_version 0x110b196e fw_build_timestamp 2022-12-22 12:54 fw_build_id WLAN.HSP.1.1-03125-QCAHSPSWPL_V1_V2_SILICONZ_LITE-3.6510.23 [    5.250981]

jordipalet, 2023-09-02:
Adding something else, in the same M.2 slot, an MT7922 is working (but only using one radio channel in AP mode) and also the passthrough works with the embedded i226V 2.5Gbps ports.

brada4, 2023-09-02:
Does the mentioned device work on
- clean port without moxprox
- debian in a vm
-

cristian-ciobanu, 2023-11-12:
I think this is a known bug in ath11k driver. There are 2 issues opened on the bugzilla bug tracker for this.
```
https://bugzilla.kernel.org/show_bug.cgi?id=216055
https://bugzilla.kernel.org/show_bug.cgi?id=217570
```
Currently there is no solution or workaround. I also have a WCN6855 / NFA765 card which cannot be used in a virtual machine. I also tested a while ago in Proxmox without luck. I have another MT7916 card which seems to work so the passthrough problem is related only to ath11k driver.

Second bug reports a similar issue for another QCN9074 card so I think the problem is not specific only for WCN6855 / NFA765 card but more generic for all cards which use ath11k driver.

brada4, 2023-11-13:
Known bug of what? That proxmox or kvm may not be able to remap every pcie slot or sometimes anything on a consumer grade hardware?

roolrz, 2023-11-16:
Here are some discussions that might be related, it seems to be an upstream bug. 

https://patchwork.kernel.org/project/linux-wireless/patch/20230601033840.2997-1-quic_bqiang@quicinc.com/

https://bugzilla.kernel.org/show_bug.cgi?id=217570

https://forum.openwrt.org/t/qcn9074-doesnt-initialize-on-x86-64/163288

I'm gonna to try out the solutions provided by the discussion. I'll back if any works.

roolrz, 2023-11-18:
Some updates:

After applying the patch https://patchwork.kernel.org/project/linux-wireless/patch/20230601033840.2997-1-quic_bqiang@quicinc.com/ to the self-compiled linux-6.6.1 kernel in Debian 12 (it's not been merged to 6.6.1 yet). I managed to get ath11k working inside Proxmox VE. The 6.1.0 kernel that comes with official Debian 12 doesn't work either.

Here's some log for the working kernel:
```
root@debian:/# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: enp6s18: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP mode DEFAULT group default qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff
3: wlp1s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DORMANT group default qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff permaddr xx:xx:xx:xx:xx:xx
root@debian:/# lspci -nn | grep Network
01:00.0 Network controller [0280]: Qualcomm Technologies, Inc QCNFA765 Wireless Network Adapter [17cb:1103] (rev 01)
root@debian:/# dmesg | grep ath11k
[    4.494954] ath11k_pci 0000:01:00.0: BAR 0: assigned [mem 0xfe800000-0xfe9fffff 64bit]
[    4.502295] ath11k_pci 0000:01:00.0: MSI vectors: 32
[    4.747332] ath11k_pci 0000:01:00.0: wcn6855 hw2.0
[    5.756418] ath11k_pci 0000:01:00.0: chip_id 0x2 chip_family 0xb board_id 0xff soc_id 0x400c0200
[    5.756426] ath11k_pci 0000:01:00.0: fw_version 0x11090c35 fw_build_timestamp 2022-04-18 20:23 fw_build_id WLAN.HSP.1.1-03125-QCAHSPSWPL_V1_V2_SILICONZ_LITE-3.6510.9
[    6.130561] ath11k_pci 0000:01:00.0 wlp1s0: renamed from wlan0
```

It's definitely an issue of upstream ath11k driver. 

For OpenWRT, I failed to get it working after changing some code in the 5.15 kernel for some unknown reason. And the ath11k driver in 5.15 is very different from in 6.6, I don't know if there's some other bugs in 5.15 that prevent the firmware being load, or I just apply the patch in the wrong way.

roolrz, 2023-11-19:
Final update:

In the stable 23.05 branch, it uses backported driver from 6.1 kernel, I tried again for applying the patch https://patchwork.kernel.org/project/linux-wireless/patch/20230601033840.2997-1-quic_bqiang@quicinc.com/ for the newer ath11k driver.

Unfortunately, it acts different from Debian, here's some log in OpenWRT 23.05.2

```
[    4.219976] NET: Registered PF_QIPCRTR protocol family
[    4.224095] xt_time: kernel timezone is -0000
[    4.238067] PPP generic driver version 2.4.2
[    4.240294] NET: Registered PF_PPPOX protocol family
[    4.245922] ath11k_pci 0000:06:10.0: BAR 0: assigned [mem 0xc1600000-0xc17fffff 64bit]
[    4.249247] Patched ath11k driver
[    4.250350] ath11k_pci 0000:06:10.0: MSI vectors: 1
[    4.251323] ath11k_pci 0000:06:10.0: wcn6855 hw2.0
[    4.252011] ath11k_pci 0000:06:10.0: FW memory mode: 0
[    4.411729] mhi mhi0: Requested to power ON
[    4.415032] mhi mhi0: Power on setup success
[    4.534915] mhi mhi0: Wait for device to enter SBL or Mission mode
[    5.145512] kmodloader: done loading kernel modules from /etc/modules.d/*
[    5.171411] ath11k_pci 0000:06:10.0: chip_id 0x2 chip_family 0xb board_id 0xff soc_id 0x400c0200
[    5.172458] ath11k_pci 0000:06:10.0: fw_version 0x110b196e fw_build_timestamp 2022-12-22 12:54 fw_build_id WLAN.HSP.1.1-03125-QCAHSPSWPL_V1_V2_SILICONZ_LITE-3.6510.23
[    5.259366] ath11k_pci 0000:06:10.0: leaving PCI ASPM disabled to avoid MHI M2 problems
[    6.314684] ath11k_pci 0000:06:10.0: failed to receive control response completion, polling..
[    6.802607] 8021q: adding VLAN 0 to HW filter on device eth0
[    6.805937] br-lan: port 1(eth0) entered blocking state
[    6.806707] br-lan: port 1(eth0) entered disabled state
[    6.807593] device eth0 entered promiscuous mode
[    6.812601] br-lan: port 1(eth0) entered blocking state
[    6.813417] br-lan: port 1(eth0) entered forwarding state
[    6.819878] 8021q: adding VLAN 0 to HW filter on device eth1
[    7.354711] ath11k_pci 0000:06:10.0: Service connect timeout
[    7.355646] ath11k_pci 0000:06:10.0: failed to connect to HTT: -110
[    7.356846] ath11k_pci 0000:06:10.0: failed to start core: -110
[    7.566788] ath11k_pci 0000:06:10.0: ignore reset dev flags 0x4000
[    7.674016] ath11k_pci 0000:06:10.0: ignore reset dev flags 0x4000
[    7.834730] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
root@OpenWrt:~# uname -a
Linux OpenWrt 5.15.137 #0 SMP Tue Nov 14 13:38:11 2023 x86_64 GNU/Linux
```

Due to lack of knowledge of ath11k, wireless driver and OpenWRT source tree, I'm not gonna to dig in this issue. Hope these info will be helpful for the dev team.

brada4, 2023-11-19:
You can try snapshot every now and then (kernel 6.1 often with newer wifi backported)