openwrt/openwrt #13403: [ZTE MF286R] Kernel Panic on Boot

cunyiles, 2023-09-03:
### Describe the bug

After a successful OpenWrt installation yesterday, I reverted to the original to update the modem firmware. But today I get kernel panic error when booting OpenWrt. 

- There is no problem with the original firmware
- initramfs and squashfs give the same error
- I can access failsafe mode
- I'm sure I'm using the correct files: `openwrt-22.03.5-ath79-nand-zte_mf286r-squashfs-sysupgrade.bin` `openwrt-22.03.5-ath79-nand-zte_mf286r-initramfs-kernel.bin`

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ath79/nand

### Device

ZTE MF286R

### Image kind

Official downloaded image

### Additional info

```
[    0.346780] printk: console [ttyS0] enabled
[    0.346780] printk: console [ttyS0] enabled
[    0.355842] printk: bootconsole [early0] disabled
[    0.355842] printk: bootconsole [early0] disabled
[    0.382257] spi-nand spi0.1: GigaDevice SPI NAND was found.
[    0.388031] spi-nand spi0.1: 128 MiB, block size: 128 KiB, page size: 2048, OOB size: 128
[    0.398845] 6 fixed-partitions partitions found on MTD device (null)
[    0.405501] OF: Bad cell count for /ahb/spi@1f000000/flash@1/partitions
[    0.412385] OF: Bad cell count for /ahb/spi@1f000000/flash@1/partitions
[    0.419929] OF: Bad cell count for /ahb/spi@1f000000/flash@1/partitions
[    0.426832] OF: Bad cell count for /ahb/spi@1f000000/flash@1/partitions
[    0.434406] Creating 6 MTD partitions on "(null)":
[    0.439364] 0x000000000000-0x0000000a0000 : "fota-flag"
[    0.448902] 0x0000000a0000-0x000000120000 : "art"
[    0.456204] 0x000000120000-0x0000001a0000 : "mac"
[    0.463580] 0x0000001a0000-0x000001800000 : "ubiconcat0"
[    0.500452] 0x000001800000-0x000001c00000 : "kernel"
[    0.512701] 0x000001c00000-0x000008000000 : "ubiconcat1"
[    0.659362] spi-nor spi0.0: gd25q16 (2048 Kbytes)
[    0.664321] 3 fixed-partitions partitions found on MTD device spi0.0
[    0.670884] Creating 3 MTD partitions on "spi0.0":
[    0.675863] 0x000000000000-0x0000000a0000 : "u-boot"
[    0.682755] 0x0000000a0000-0x0000000c0000 : "u-boot-env"
[    0.689170] 0x0000000c0000-0x000000200000 : "reserved1"
[    0.698249] Concatenating MTD devices:
[    0.702198] (0): "ubiconcat0"
[    0.705259] (1): "ubiconcat1"
[    0.708323] into device "ubi-concat"
[    0.712056] 1 fixed-partitions partitions found on MTD device ubi-concat
[    0.718971] Creating 1 MTD partitions on "ubi-concat":
[    0.724298] 0x000000000000-0x000007a60000 : "ubi"
[    1.425862] switch0: Atheros AR8337 rev. 2 switch registered on mdio.0
[    2.083353] ag71xx 19000000.eth: connected to PHY at mdio.0:00 [uid=004dd036, driver=Atheros AR8216/AR8236/AR8316]
[    2.094631] eth0: Atheros AG71xx at 0xb9000000, irq 4, mode: sgmii
[    2.103343] NET: Registered protocol family 10
[    2.115439] Segment Routing with IPv6
[    2.119347] NET: Registered protocol family 17
[    2.124074] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[    2.137468] 8021q: 802.1Q VLAN Support v1.8
[    2.145761] PCI host bridge /ahb/pcie-controller@18250000 ranges:
[    2.152132]  MEM 0x0000000012000000..0x0000000013ffffff
[    2.157529]   IO 0x0000000000000000..0x0000000000000000
[... 164 lines cut ...]
[   24.793638] [<80413594>] 0x80413594
[   24.797247] [<80413518>] 0x80413518
[   24.800841] [<80410564>] 0x80410564
[   24.804458] [<80411c7c>] 0x80411c7c
[   24.808068] [<800ebd40>] 0x800ebd40
[   24.811667] [<80413d78>] 0x80413d78
[   24.815270] [<8201f000>] 0x8201f000 [ath9k@827bab61+0x192b0]
[   24.821111] [<8201f018>] 0x8201f018 [ath9k@827bab61+0x192b0]
[   24.826952] [<80060970>] 0x80060970
[   24.830553] [<800ea170>] 0x800ea170
[   24.834154] [<801a4164>] 0x801a4164
[   24.837772] [<800ea190>] 0x800ea190
[   24.841369] [<800ec150>] 0x800ec150
[   24.844979] [<8006e1ac>] 0x8006e1ac
[   24.848590]
[   24.850128] Code: 0000000f  8c820000  0000000f <03e00008> 00000000  0000000f  94820000  3042ffff  0000000f
[   24.860208]
[   24.861794] ---[ end trace 196c38740f66603d ]---
[   24.866581] Kernel panic - not syncing: Fatal exception
[   24.871979] Rebooting in 1 seconds..
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

maxxxed123, 2024-02-28:
@cunyiles merhaba ben maxxxed fiverden bana yazabilirmisin, cok acil yardim ihtiyacim var kika99@hotmail.de

maxxxed123, 2024-02-28:
ya da size nereden yazabilirim @cunyiles