openwrt/openwrt #13404: openwrt on Sinovoip BananaPi BPi R3 is not stable

jcdutton, 2023-09-03:
### Describe the bug

After a period of use, the BPI-R3 stops forwarding packets over wifi.
A software "reboot" does not fix the problem. A power cycle is needed to recover it.
Sometimes, even after a power cycle, the wifi needs disable/enable cycles in Luci to get it working again.
Router Model: Sinovoip BananaPi BPi R3
Image downloaded from:
https://openwrt.org/toh/sinovoip/bananapi_bpi-r3
Luci status:
OpenWrt SNAPSHOT r23797-ff2379662c / LuCI Master git-23.236.53281-98e3743

### OpenWrt version

r23797-ff2379662c

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Steps to reproduce

It is not easily reproducible. It occurs after a days of usage.
I am not sure if the problem is wifi related, or switch port related or something else.

### Actual behaviour

After a period of use, the BPI-R3 stops forwarding packets over wifi.
A software "reboot" does not fix the problem. A power cycle is needed to recover it.
Sometimes, even after a power cycle, the wifi needs disable/enable cycles in Luci to get it working again.
In summary: The unit is not stable over periods of > 24hrs.

### Expected behaviour

The router should be stable for weeks.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-03:
Can you share /etc/config/wireless without `key` and `ssid` lines?
Which band stops?
What do you see in `logread`?

jcdutton, 2023-09-03:
Nothing appears in the logread when the problem happens. I.e. no extra lines appear.
It was using the 2G wifi at the time, but the problem happens on both 2G and 5G.
/etc/config/wireless
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi'
	option channel 'auto'
	option band '2g'
	option cell_density '0'
	option htmode 'HE20'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option encryption 'sae-mixed'
	option macfilter 'allow'
	list maclist 'xx:xx:xx:xx:xx:xx'
	list maclist 'xx:xx:xx:xx:xx:xx'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi+1'
	option channel 'auto'
	option band '5g'
	option htmode 'HE80'
	option cell_density '0'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option encryption 'sae-mixed'
	option macfilter 'allow'
	list maclist 'xx:xx:xx:xx:xx:xx'
	list maclist 'xx:xx:xx:xx:xx:xx'

config wifi-iface 'wifinet2'
	option device 'radio0'
	option mode 'ap'
	option encryption 'sae-mixed'
	option network 'lan'
	option macfilter 'allow'
	list maclist 'xx:xx:xx:xx:xx:xx'
	list maclist 'xx:xx:xx:xx:xx:xx'

config wifi-iface 'wifinet3'
	option device 'radio1'
	option mode 'ap'
	option encryption 'sae-mixed'
	option network 'lan'
	option macfilter 'allow'
	list maclist 'xx:xx:xx:xx:xx:xx'
	list maclist 'xx:xx:xx:xx:xx:xx'

config wifi-iface 'wifinet4'
	option device 'radio0'
	option mode 'ap'
	option encryption 'sae-mixed'
	option network 'guest'
	option macfilter 'allow'
	list maclist 'xx:xx:xx:xx:xx:xx'
	list maclist 'xx:xx:xx:xx:xx:xx'

config wifi-iface 'wifinet5'
	option device 'radio1'
	option mode 'ap'
	option encryption 'sae-mixed'
	option network 'guest'
	option macfilter 'allow'
	list maclist 'xx:xx:xx:xx:xx:xx'
	list maclist 'xx:xx:xx:xx:xx:xx'

brada4, 2023-09-03:
`option country 'GB'` or alike is missing on both radios, though that should be only disabling 5GHz radio suddenly.
Any thermal issues?

zuowei8, 2023-09-05:
please try to add one CPU fin or one fan to reduce the CPU's tempreture.

judyhuang88, 2023-09-05:
hello, friend, this problem may caused by overheating, suggest you can try our heatsink with fan, you can check the heatsink here:
https://www.aliexpress.us/item/3256805581539217.html?spm

brada4, 2023-09-05:
Interesting question. Depends if you put asbestos or copper heatsink.

hagier, 2023-09-13:
Can you still connect through cable when it stops forwarding WiFi packets?

I have issue where it completely freezes and I cannot connect to it until complete reboot through power cycle. #13198
Heat sinks and additional fan helped but it still happens.

ne20002, 2023-09-21:
We had the full freeze already and it seemed to have something to do with iPhones leaving the WiFi.
I fixed it for me by disabling 2.4 WiFi and go with only 5 GHz. If this is an option for you, you may try this.

brada4, 2023-09-21:
@ne20002 are you in position to crash 2.4ghz wifi again and look at logread around your activities?

ne20002, 2023-09-22:
@brada4 
We have a discusion already in #13198.
For me freezes disapeared when I disabled 2.4G Wifi completely. As stated in #13198 it only occured when an iPhone left the wifi range and both Wifi have been on. It did never happen (for me) with Android or /e/OS devices.
As I don't have an iPhone (a friend has one) I just can't reproduce it without her ;)
I have a technical background but I don't know how to use logread here. Logs have been gone with the freeze. I may need advice on how to do this properly.

brada4, 2023-09-22:
So this is duplicate of other?

jcdutton, 2023-09-22:
Hi,
I am the original poster for this bug.
It is not a duplicate of #13198 as I do not see any stack traces and for me it does not freeze, it just requires a power cycle, but sometimes a simple disconnect/reconnect at the client side works.
When it fails for me, I can still ssh and logread to the BPi R3, it is just the wifi bits that stop working.
Unfortunately, when I do ssh in and do logread or dmesg, no extra log messages appear between before the problem to the problem happening.

brada4, 2023-09-22:
Best explanation for random chips flopping is temperature, next coming burnt dried capacitors.
Does 2.4ghz alone fail?

If you wave hand over the board - is there any chip running hot(ter)? They should certainly be under 70C. Normally few minutes of any kind of stress test should bring chips to their max.

jcdutton, 2023-09-23:
Hi,

Using "sensors" from lm-sensors, the temps are all less than 70C and have heat sinks on them.
I don't think the problem is heat related.

brada4, 2023-09-23:
So genuinely wifi firmware crashes on its own. Is it just one when its up or one when both are up. Might also relate to pci bus limitation when both are needed for crash. Please add `ethtool -i` for reference.

jcdutton, 2023-09-24:
Note: phy0 is 2.4G, phy1 is 5G
A question, is "DEV_000000-20221012174937" the latest and best firmware?
Are there any instructions on how to compile and build one's own openwrt image.
The Linux wifi code is not very well written, so I would like to add some fixes to it, and test to see if I can get any new log messages to be raised when things go wrong. For example, a fair amount of the Linux wifi code has no error paths.

root@OpenWrt:~# ethtool -i phy0-ap0
driver: mt798x-wmac
version: 5.15.132
firmware-version: DEV_000000-20221012174937
expansion-rom-version: 
bus-info: 18000000.wifi
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
root@OpenWrt:~# ethtool -i phy1-ap0
driver: mt798x-wmac
version: 5.15.132
firmware-version: N/A
expansion-rom-version: 
bus-info: 18000000.wifi
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

brada4, 2023-09-24:
Did/does vendor openwrt 21/22 images work?
If they do - what is the distinction...

jcdutton, 2023-10-18:
I have maybe narrowed down the problem a bit.
I got to a situation where I could, from the wifi client, ping some devices on the fixed network and not others. Looking at the arp table showed this sort of thing:
The client is IP address D.
All the others, A, B, C are on the ethernet LAN, only D is on wifi.
arp -an
? (IP Address A) at MAC-Address-A [ether] on wlp1
? (IP Address B) at <incomplete> on wlp1
? (IP Address C) at <incomplete> on wlp1

It this scenario, ping from D to A worked, ping D to B or D to C failed.
ping from A to B worked, ping from A to C worked.
So, the wifi itself looks OK.
It appears to be an ARP forwarding / resolution problem within the BPi R3.
So, this might point to a problem with the switching within the BPi R3, and not a problem with the wifi within the R3.

brada4, 2023-10-19:
Are  you using openwrt?

jcdutton, 2023-10-19:
@brada4 
Why do you ask?
I stated the openwrt image I was downloading in the original post.
I am currently compiling openwrt from source and using that, and it has similar problems.

brada4, 2023-10-19:
Interface na,e wlp is strange. Should be lanX ethX br-X sfpX normally. If you renamed default bridges you dhould say so.
By default clients are on br-lan arp table.

jcdutton, 2023-10-19:
The interface name wlp1 arp table is from the client PC, not the arp table of the openwrt.
I was wishing to highlight that the client PC is not getting responses to ARP requests correctly, when the problem occurs.
With more investigation, I installed tcpdump on the openwrt and am finding that it is not capturing all the ARP packets that should be seen. E.g. I am seeing ARP replies to a wifi client, but not seeing the ARP request from that client.
So, in this case it is forwarding the ARP packets, but not all of them are getting to tcpdump.
The problem with this, is that when the stability problem occurs next time, I won't know if it was a tcpdump problem or an actual missing packet.
Also, if I configure openwrt to send syslog messages, if it uses tcp all the message get through, if I use udp, hardly any of the messages get through. I can maybe infer from that that the openwrt is dropping/missing forwarding some udp packets. I will do some more investigations to provide some more accurate conclusions.

lu-zero, 2023-10-25:
It seems related to the wired side of the bridge, I can reproduce it on a slightly different hardware so the problem doesn't seem only for the bpi-r3.

In my case the wired device trying to join using dhcp had its packets reach dnsmasq but the replies from dnsmasq seem to not reach the wired network.

jcdutton, 2023-11-04:
I saw this in another post to a different bug:
"I finally managed to reproduce this (or a similar) problem on MT7981, but as the Ethernet part of MT7986 is more or less identical it should affect both SoCs: Up on receiving a corrupted Ethernet frame, RX DMA hangs. MediaTek actually reverted to use the old DMA and we may have to do the same:
https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/8ecd51bed4f7b011c2d59cf93627ca598d4e9172
https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/231d3ac3e990b788564973c9a37ea850accce6e4"

So, there might be some progress coming in the future.

jcdutton, 2023-11-05:
Previous I was plugged into lan4 ethernet
Looking and the block diagram in the schematic, it looks like lan1-4 goes via an extra chip, MT7986A -> MT7531AE -> RJ45
The "wan" interface goes direct, MT7986A -> RJ45
I have now plugged into the "wan" interface, to see if this is any more stable. Thus narrowing down the problem to maybe the MT7531AE chip. If the "wan" interface is stable, it is good enough for me.

jcdutton, 2023-11-06:
@lu-zero 
Which hardware do you have?
Do you have a block diagram for it, showing which chips are being used for what?
My setup seems more stable now that I have bypassed the MT7531AE switch chip.
I don't know if your hardware uses a switch chip or not.

lu-zero, 2023-11-07:
I saw the behavior on a GL.iNet GL-MT3000, I did not experience it again so far.

jcdutton, 2023-11-11:
Using the wan interface, I have not seen any examples of the stability problems.
I therefore suspect the MT7531AE chip as being the problem.
Does anyone have any documentation for the MT7531AE ?
Without the documents, it is difficult to diagnose any further.

hagaygo, 2023-11-18:
Hi

Got the BPI-R3 up and running OpenWrt for about 2.5 month now as dumb AP.

Over that time i had 4 "Crashes". (The longest uptime without any issue was 29 days, other restarts were due to the upgrades or power outage)

My install is on the NAND memory and today's crash was on the 23.05.0 official release (second crash on this version)

Will use this opportunity to upgrade to 23.05.2.

My device is accessible using ssh but service like dnsmasq and wifi are not responding.

Attaching my dmesg output , i hope it will help to resolve this issue.

`
[182525.608601] ------------[ cut here ]------------
[182525.613304] Timeout waiting for MCU reset state 10
[182525.618215] WARNING: CPU: 0 PID: 7479 at 0xffffffc000adb2a8 [mt7915e@0000000055157591+0x21000]
[182525.626905] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_re                     ject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_fl                     ow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib                      mt76 mac80211 cfg80211 slhc sfp nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcrc32c crc_ccitt                      compat crypto_safexcel pwm_fan i2c_gpio i2c_algo_bit cifs oid_registry cifs_md4 cifs_arc4 asn1_decoder dns_resolver sha1_generic seqiv md5 des_g                     eneric libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common
[182525.700941] CPU: 0 PID: 7479 Comm: kworker/u8:0 Not tainted 5.15.134 #0
[182525.707622] Hardware name: Bananapi BPI-R3 (DT)
[182525.712221] Workqueue: mt76 0xffffffc000add364 [mt7915e@0000000055157591+0x21000]
[182525.719771] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[182525.726798] pc : 0xffffffc000adb2a8 [mt7915e@0000000055157591+0x21000]
[182525.733391] lr : 0xffffffc000adb2a8 [mt7915e@0000000055157591+0x21000]
[182525.739985] sp : ffffffc0113ebc50
[182525.743370] x29: ffffffc0113ebc50 x28: ffffff8003dc1ec0 x27: ffffff8003dc25f8
[182525.750572] x26: ffffff8003df2020 x25: ffffff8003df2590 x24: ffffff8003dc2020
[182525.757775] x23: 0000000000000000 x22: ffffff8003dc6020 x21: 0000000000000010
[182525.764977] x20: ffffff8003dc9ee0 x19: 0000000000000000 x18: ffffffc008a4a338
[182525.772179] x17: 000000007871b926 x16: 00000000208ebc88 x15: 00000000000003c9
[182525.779381] x14: 0000000000000143 x13: ffffffc0113eb988 x12: ffffffc008aa2338
[182525.786583] x11: 434d20726f662067 x10: ffffffc008aa2338 x9 : 0000000000000000
[182525.793785] x8 : ffffffc008a4a2e8 x7 : ffffffc008a4a338 x6 : 0000000000000001
[182525.800987] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[182525.808188] x2 : ffffff807fb8a090 x1 : ffffffc07716c000 x0 : 0000000000000026
[182525.815390] Call trace:
[182525.817908]  0xffffffc000adb2a8 [mt7915e@0000000055157591+0x21000]
[182525.824157]  0xffffffc000addff0 [mt7915e@0000000055157591+0x21000]
[182525.830405]  0xffffffc008050910
[182525.833617]  0xffffffc008050c20
[182525.836829]  0xffffffc008058e1c
[182525.840040]  0xffffffc00801581c
[182525.843252] ---[ end trace 0320bc83c365f310 ]---
[182556.293752] ------------[ cut here ]------------
[182556.298454] Timeout waiting for MCU reset state 20
[182556.303347] WARNING: CPU: 0 PID: 7479 at 0xffffffc000adb2a8 [mt7915e@0000000055157591+0x21000]
[182556.312035] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_re                     ject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_fl                     ow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib                      mt76 mac80211 cfg80211 slhc sfp nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcrc32c crc_ccitt                      compat crypto_safexcel pwm_fan i2c_gpio i2c_algo_bit cifs oid_registry cifs_md4 cifs_arc4 asn1_decoder dns_resolver sha1_generic seqiv md5 des_g                     eneric libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common
[182556.386072] CPU: 0 PID: 7479 Comm: kworker/u8:0 Tainted: G        W         5.15.134 #0
[182556.394139] Hardware name: Bananapi BPI-R3 (DT)
[182556.398738] Workqueue: mt76 0xffffffc000add364 [mt7915e@0000000055157591+0x21000]
[182556.406289] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[182556.413316] pc : 0xffffffc000adb2a8 [mt7915e@0000000055157591+0x21000]
[182556.419910] lr : 0xffffffc000adb2a8 [mt7915e@0000000055157591+0x21000]
[182556.426504] sp : ffffffc0113ebc50
[182556.429889] x29: ffffffc0113ebc50 x28: ffffff8003dc1ec0 x27: ffffff8003dc25f8
[182556.437091] x26: ffffff8003df2020 x25: ffffff8003df2590 x24: ffffff8003dc2020
[182556.444293] x23: 0000000000000000 x22: ffffff8003dc6020 x21: 0000000000000020
[182556.451495] x20: ffffff8003dc9ee0 x19: 0000000000000000 x18: ffffffc008a4a338
[182556.458697] x17: 0000000083512952 x16: 00000000589160b1 x15: 0000000000000420
[... cut ...]
[183497.631154]  0xffffffc00865c710
[183497.634365]  0xffffffc00865ca28
[183497.637576]  0xffffffc00866b80c
[183497.640788]  0xffffffc00866b8a0
[183497.644000]  0xffffffc00874290c
[183497.647212]  0xffffffc0087463fc
[183497.650423]  0xffffffc00863d994
[183497.653634]  0xffffffc0081bb8cc
[183497.656845]  0xffffffc00801f60c
[183497.660057]  0xffffffc00801f6e0
[183497.663268]  0xffffffc008830844
[183497.666480]  0xffffffc0088313e0
[183497.669692]  0xffffffc0080115cc
[183497.672904] ---[ end trace 0320bc83c365f316 ]---
`