openwrt/openwrt #13407: Netgear wax220 drops lots op packets on 1000baseT/Full Link

jmthill, 2023-09-03:
### Describe the bug

Gbit speed doesn't seem to be properly working

### OpenWrt version

23.05-SNAPSHOT r23404-56827dac01

### OpenWrt target/subtarget

mediatek/filogic

### Device

Netgear WAX220

### Image kind

Official downloaded image

### Steps to reproduce

Iperf3 has lots of retransmits and ethtool shows no gbit supported/advertised

### Actual behaviour

Iperf3:
iperf3 -c xxx
Connecting to host xxx, port 5201
[  5] local xxxy port 48116 connected to xxx port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  96.1 MBytes   806 Mbits/sec  624   65.0 KBytes       
[  5]   1.00-2.00   sec  99.7 MBytes   836 Mbits/sec  671   62.2 KBytes       
[  5]   2.00-3.00   sec   100 MBytes   841 Mbits/sec  759   60.8 KBytes       
[  5]   3.00-4.00   sec  97.3 MBytes   816 Mbits/sec  583   56.6 KBytes       
[  5]   4.00-5.00   sec   102 MBytes   857 Mbits/sec  700   58.0 KBytes       
[  5]   5.00-6.00   sec  98.4 MBytes   825 Mbits/sec  721   48.1 KBytes       
[  5]   6.00-7.00   sec  96.6 MBytes   810 Mbits/sec  703   48.1 KBytes       
[  5]   7.00-8.00   sec  96.6 MBytes   811 Mbits/sec  570   63.6 KBytes       
[  5]   8.00-9.00   sec  96.2 MBytes   807 Mbits/sec  465   53.7 KBytes       
[  5]   9.00-10.00  sec  99.0 MBytes   830 Mbits/sec  676   46.7 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   982 MBytes   824 Mbits/sec  6472             sender
[  5]   0.00-10.04  sec   981 MBytes   820 Mbits/sec                  receiver

ethtool:
ethtool eth0
Settings for eth0:
	Supported ports: [ TP MII ]
	Supported link modes:   2500baseT/Full 
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  2500baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	                                     100baseT/Half 100baseT/Full 
	                                     1000baseT/Full 
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Unknown! (255)
	Port: Twisted Pair
	PHYAD: 6
	Transceiver: external
	Auto-negotiation: on
	MDI-X: Unknown
	Current message level: 0x000000ff (255)
			       drv probe link timer ifdown ifup rx_err tx_err
	Link detected: yes

### Expected behaviour

gbit speed advertised/supported
maybe less drops in iperf3

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-03:
You can add your own advertised link modes confirm it works, at least 100 a
`ethtool -s eth0 advertise 0x80000000002c`

From `man ethtool`
```
         0x001                          10baseT Half /// change c to e or f to enable 10 mbps rates not in vendors marketing paper.
         0x002                          10baseT Full
         0x004                          100baseT Half
         0x008                          100baseT Full
         0x020                          1000baseT Full
0x800000000000                 2500baseT Full
```

jmthill, 2023-09-03:
Thanks for the quick answer. "ethtool -s eth0 advertise 0x80000000002c" corrects the advertised speeds but of course not the supported speed, and the retransmits of iperf3 stay the same. Iperf3 on an nwa50ax-pro works like a charm with only a handfull of restransmits.

ethtool eth0
Settings for eth0:
	Supported ports: [ TP MII ]
	Supported link modes:   2500baseT/Full 
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	                        2500baseT/Full 
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	                                     100baseT/Half 100baseT/Full 
	                                     1000baseT/Full 
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 6
	Transceiver: external
	Auto-negotiation: on
	MDI-X: Unknown
	Current message level: 0x000000ff (255)
			       drv probe link timer ifdown ifup rx_err tx_err
	Link detected: yes

brada4, 2023-09-03:
ping @dangowrt 

Could you test if phy works at 10Mbps i.e if it goes up if advertised link speed is 0x1 for sake of completeness, lets wait for main porter to get fix permanent, in the meantime you can add ethtool command in /etc/rc.local.

jmthill, 2023-09-03:
The switch changed to 10Mbit/Half but the AP was not reachable anymore, which should be normal since the lowest speed it supports is 100Mbit?

brada4, 2023-09-03:
Thanks for confirming, set it back and enjoy

brada4, 2023-09-05:
Some packet drops are expected along with retransmits and lowering transmit rate. There is no more (chance of) duplex mismatch and it kind of must work fine.
You can try looking at ethtool -S and the likes to see where it goes under 1Gbps, it should be maybe few drops if at all and full gigabit. Or if you test via wifi - varied radio conditions can cause such drops and still normal.

jmthill, 2023-09-05:
There's no wifi involved in the iperf3 test. And under the same conditions my nwa50ax-pro manages to do the iperf3 with really just around 10 retransmits compared to over 6400 on the wax220. ethtool -S show no drops neither on the wax220 nor on the iperf3 server side. There's also no drops on the switch in between.

brada4, 2023-09-05:
Check ethtool -S again for any pauses employed. Really should be no reason for such impairment. In the worst case every 1-4 1500byte packets gets 64 byte ack and IF there is gigabit CPU port traversed 2x it should be still be around 950Mbps/120MBps with like you said 10 drops during test (maybe 11, but really not a lot)

jmthill, 2023-09-05:
What value of ethtool is "the pauses employed"?
"iperf3 -R" in receiving direction works perfectly (0 retransmits and full speed)

brada4, 2023-09-05:
Something asymmetric
In ethtool -S -if any error or pause number diffets from zero... It is disabled for good, realy no pauses shall be emitted or acted upon.
Try enabling either offload in firewall. Very unlikely with real cpu your router has.

jmthill, 2023-09-05:
Neither hardware nor software offload nor packet steering changes anything. I can't see anything weird in the ethtool -S output.

brada4, 2023-09-05:
You have one port. Are you sure you test network foraarding speed?

closed 2023-09-06 by jmthill

brada4, 2023-09-08:
My lower end router does 600Mbps either way on single thread and gigabit on 4 threads matching device CPU cores. It drops to 100Mbps in --bidi mode. On the other hand AX laptop from wifi to VM ethernet on itself does full gigabit in br-lan with any setting. I think it is limitation on process on CPU generating packets vs same CPU almost-zero-copy forwarding them.