openwrt/openwrt #13409: OnHub devices do not upgrade properly due to improper code

dadogroove, 2023-09-03:
### Describe the bug

The Google OnHub devices (ASUS and TP-Link) do not upgrade properly when using the sysupgrade files. This has been occurring since these devices have been supported. It is not a complete upgrade. It will report the new version on the overview page but in the installed software tab it reports the previous base-files and previous kmod versions if a kernel change has occured. Thanks to user regae on the OpenWRT forums, the problem has been located.  The wrong brackets were used in the /target/linux/ipq806x/base-files/lib/upgrade/platform.sh file. 

platform_copy_config() {
	case "${board_name}" in  <-- this should be (board_name) {wrong brackets used}
	asus,onhub |\
	tplink,onhub)
		emmc_copy_config
		;;
	esac
	return 0

I have tested the change and can report that it fixes the upgrade issue.

### OpenWrt version

23.05.0-rc3

### OpenWrt target/subtarget

ipq806x/chromium

### Device

Google OnHub (ipq806x/chromium)

### Image kind

Official downloaded image

### Steps to reproduce

try to upgrade device with save current settings option enabled

### Actual behaviour

 It is not a complete upgrade. It will report the new version on the overview page but in the installed software tab it reports the previous base-files and previous kmod version if a kernel change has occured.

### Expected behaviour

Device should upgrade normally with saved settings.

### Additional info

/target/linux/ipq806x/base-files/lib/upgrade/platform.sh (at the bottom of the file)

platform_copy_config() {
	case "${board_name}" in  <-- this should be (board_name) {wrong brackets used}
	asus,onhub |\
	tplink,onhub)
		emmc_copy_config
		;;
	esac
	return 0

I have tested the change and can report that it fixes the upgrade issue.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

tmn505, 2023-09-03:
cc @computersforpeace

regae, 2023-09-04:
in [here](https://github.com/openwrt/openwrt/commit/ef649b0b14814a84a480e9b7805e4a401e9e06b6#r116354413)

commit 7b78a19e (2023-09-12): ipq806x: Correct OnHub sysupgrade config logic

There's a typo in here: board_name is a function, not a variable. This
issue was pointed out on the OpenWrt forum.

Closes: #13409

Reviewed-by: Robert Marko <robimarko@gmail.com>
PR #13434 (2023-09-12): ipq806x: Correct OnHub sysupgrade config logic
There's a typo in here: board_name is a function, not a variable. This issue was pointed out on the OpenWrt forum.

Closes: #13409
commit 10aee603 (2023-09-16): ipq806x: Correct OnHub sysupgrade config logic

There's a typo in here: board_name is a function, not a variable. This
issue was pointed out on the OpenWrt forum.

Closes: #13409

Reviewed-by: Robert Marko <robimarko@gmail.com>

(cherry picked from commit 7b78a19e6a16f5c05bfc6d7925b9981048c508d7)
closed 2023-09-12 by openwrt-bot