openwrt/openwrt #13410: firewall4 offload disconnects tcp when wifi roaming

brada4, 2023-09-03:
### Describe the bug

@jow- 
If firewall offload gets enabled it adds wifi interfaces to soft offload group and when roaming occurs all connections get severed while without offload they keep going. Seems this happens by design of flowtables that offloaded connections are stuck on interfaces without chance to migrate as opposed to normal path where wifi-s would be in bridge and all connections would follow the customer.
Is it possible to remove wifis, all or having same ssid for firewall offload consideration?

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

any

### Device

with multiple radios

### Image kind

Official downloaded image

### Steps to reproduce

Establish 2 identically authenticated AP-s on 5GHz and 2.4GHz radios.
Enable firewall flow offload, verify that wifi adapters are in offload group.
Connect SSH to somewhere outside and check with conntrack that connection is offloaded.
Now walk outside 5GHz range for client to connect to 2.4GHz

### Actual behaviour

SSH connection disconnects with timeouts

### Expected behaviour

SSH connection stays undisturbed

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

closed 2024-05-31 by openwrt-bot

nitdn, 2025-06-07:
Man I have been trying to fix this problem for months, but I had hardware flow control on. Switching it back to software finally fixed the issue.

brada4, 2025-06-07:
For those stumbling here, fresh attempt to document pros ànd cons of offload
https://openwrt.org/docs/guide-user/perf_and_log/flow_offloading
After curent 6.12 kernel mainline gained ebpf based offload that is faster than soft ofload on high function netcards, and bridge/switch firewall ewill gain offloading too, functionally superseding bridger package.