openwrt/openwrt #13412: Problem with the USB port not working in the snapshot image of fortigate 50e.

odkrys, 2023-09-04:
### Describe the bug

fortinet_fg-50e-squashfs-sysupgrade.bin / Mon Sep 4 05:09:15 2023

As the title says, the USB port does not work.

@musashino205 
Hello again.
Could you please test it?

### OpenWrt version

r23880-8554a4a7e3

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Fortinet FortiGate 50E

### Image kind

Official downloaded image

### Additional info

syslog after booting.

[rc3.txt](https://github.com/openwrt/openwrt/files/12512043/rc3.txt)
[snapshot.txt](https://github.com/openwrt/openwrt/files/12512044/snapshot.txt)

### Diffconfig

```text
rc3

kernel: [    8.923046] 0x000000a50000-0x000002000000 : "rootfs_data"
```
snapshot
```
kernel: [    8.769543] 0x000000a00000-0x000002000000 : "rootfs_data"
```

rc3
```
kernel: [  112.632602] mount_root: switching to jffs2 overlay
```
snapshot
```
mount_root: failed to sync jffs2 overlay
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

odkrys, 2023-09-05:
I reverted two commits, and now the usb works on my 6.1 test kernel.

https://github.com/openwrt/openwrt/commit/3eebb91317c64864b86cea21ae07ab84d4f4bcc1
https://github.com/openwrt/openwrt/commit/5356462ce54734afd32fce83ad118c58cfeb2a55

closed 2023-09-05 by openwrt-bot

musashino205, 2023-09-05:
Adding ```fw_devlink=permissive" to bootargs may be available as a workaround.
ex.: https://github.com/musashino-build/openwrt/commit/a5970eb031edc9be473710254fb74f5cba14bb20

mpratt14, 2023-09-14:
https://github.com/openwrt/openwrt/pull/11302 is rebased and ready for testing again

note the last commit "strict as default". if you still have issues try dropping that last commit

commit 703d667a (2023-09-05): kernel: switch back to fw_devlink=permissive

This reverts commit 5356462ce54734afd32fce83ad118c58cfeb2a55.

Kernel switching to fw_devlink=on as default broke probing some devices.
Revert it until we get a proper fix.

It seemed that mtd OF_POPULATED hack resolved probing issues but
apparently not all of them. We got reports about reading MAC using NVMEM
not working and USB controllers not working.

Ref: #10232
Fixes: #13412