openwrt/openwrt #13413: Missing Wireless Function on latest snapshot build(R6220)

muzadull, 2023-09-04:
### Describe the bug

Wireless Tab is missing in luci and both wireless radios does not show up in interfaces.
Checking kernel log shows the wireless hardware ID being detected.

### OpenWrt version

SNAPSHOT (r23880-8554a4a7e3) 

### OpenWrt target/subtarget

ramips/mt7621

### Device

NETGEAR R6220

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-04:
Snapshot has no luci included. Plese explain what you did on top of installing snapshot and the output of `iwinfo`, seeing anything here warrants you to adjust report title to reflect the truth.

muzadull, 2023-09-04:
I downloaded the firmware from here https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ramips%2Fmt7621&id=netgear_r6220.
Additional packages include : tc-tiny kmod-sched-cake kmod-veth luci-ssl luci-app-opkg luci-app-adblock luci-app-https-dns-proxy luci-app-ttyd
[image]
[image]
[image]
[image]

DragonBluep, 2023-09-05:
https://forum.openwrt.org/t/wifi-issue-with-r23769-on-mt7621/169087

muzadull, 2023-09-05:
Thanks.
I was trying the snapshot because of the latest stability fixes for MT7603E which affects the R6220.

DragonBluep, 2023-09-05:
This issue can be closed since the fix has been push to the main branch just now. https://github.com/openwrt/openwrt/commit/703d667a0cdf6dfa402c08e72d0c77a257ca5009