openwrt/openwrt #13417: AMD Geode LX hardware RNG driver broken

timur-davletshin, 2023-09-06:
### Describe the bug

AMD Geode LX hardware RNG driver is broken since OpenWRT 19.07.

Driver path: drivers/char/hw_random/geode-rng.c

### OpenWrt version

r23389-5deed175a5

### OpenWrt target/subtarget

x86/geode

### Device

Alix board 2d13

### Image kind

Official downloaded image

### Steps to reproduce

1. Install any OpenWRT image later than 19.07.9
2. Open console
3. cat /dev/hwrng

### Actual behaviour

Output in console shows regular pattern (see bellow):

��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��`I��^C

Package rng-tools (rngd daemon) refuses to run because of low entropy from /dev/hwrng

### Expected behaviour

True random output like this:

                     #) R��j���eB�m�c�a@=�ێ�#��E,�0�s�R
ö�2a��n`3t8O(bb�F�	um��^����A�*�p(��`-`�)O���V`�͂JM��	%H����ȴ�i'�Q�s�Nn�����Xj
                                                                               $7�hV�j��yʊBT�I���gh�����Gt������Žjb3C-��T
                                         ڥڝ��C/zw�
                                                  �T(C��Ĥo��-�����*MVח�r�=Z�`_���),��4v^�E���3�1���([��X��@A�׋�+Ex��t��^C

### Additional info

Problem started in OpenWRT releases later than 19.07.9 (last working release).

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-07:
looks like faulty hardware?
Driver really just passes hw reg to userspace.
drivers/char/hw_random/geode-rng.c

timur-davletshin, 2023-09-07:
> looks like faulty hardware?

Or faulty firmware? Rolling back to 19.07.9 suddenly fixes it.

brada4, 2023-09-07:
Date:   Wed Feb 16 12:44:53 2022 +0100
   Linux 4.14.267

If you are into bisecting - it is only last 2 commits potentially affecting the upstream driver:
https://github.com/torvalds/linux/commits/master/drivers/char/hw_random/geode-rng.c

Does `haveged` relieve the issue in the meantime?

timur-davletshin, 2023-09-07:
> Does `haveged` relieve the issue in the meantime?

Well, I believe it works fine with the default settings at the moment. I believe it's urngd which feeds entropy via some sort of jitter-based algorithm.

But, nevertheless, original driver should be fixed.

jow-, 2023-09-07:
This should be reported upstream at kernel.org, afair OpenWrt neither patches, nor develops this driver, so there will be no capacity to investigate and fix it.

timur-davletshin, 2023-09-07:
I will try compiling openwrt with those changes rolled back during weekend.

brada4, 2023-09-07:
You should remove asterix here:
https://github.com/torvalds/linux/blob/7ba2090ca64ea1aa435744884124387db1fac70f/drivers/char/hw_random/geode-rng.c#L108

Having no hardware i cannot confirm if that is only problem. Good luck reporting/testing.

timur-davletshin, 2023-09-07:
FYI: https://bugzilla.kernel.org/show_bug.cgi?id=217882

timur-davletshin, 2023-09-07:
> You should remove asterix here

Can I ask you kindly to prepare a patch (or even two, with those couple changes rolled back), so I just quickly compile and test those? It's been two or three years since I made my own patches and I don't remember the procedure well.

brada4, 2023-09-07:
You can do `git diff -R revOLD revNEW` to get inverse patch.

You can try editing that single line before compiling kmod as shown here:
https://github.com/openwrt/openwrt/issues/13417#issuecomment-1709946099
(blame points to latest commit to file, but if it is "only this" then you can fix it yourself)

```diff
---        priv = kzalloc(sizeof(*priv), GFP_KERNEL);
+++        priv = kzalloc(sizeof(priv), GFP_KERNEL);
```

timur-davletshin, 2023-09-07:
> You can do `git diff -R revOLD revNEW` to get inverse patch.
> 
> You can try editing that single line before compiling kmod as shown here: [#13417 (comment)](https://github.com/openwrt/openwrt/issues/13417#issuecomment-1709946099) (blame points to latest commit to file, but if it is "only this" then you can fix it yourself)
> 
> ```diff
> ---        priv = kzalloc(sizeof(*priv), GFP_KERNEL);
> +++        priv = kzalloc(sizeof(priv), GFP_KERNEL);
> ```

AFAIR there was this quill (?) based system of making patches in SDK preventing you from direct editing and which I don't remember how to use. But anyway, it will just take longer :)

brada4, 2023-09-08:
just that pointer is 4 bytes to 8 byte struct....

KanjiMonster, 2023-09-08:
> I think the actual fix could be something like this:

Yepp, that analysis and proposed solution looks correct, and should be sent upstream if confirmed (I can also take care of that if you want, else have a `Reviewed-by: Jonas Gorski <jonas.gorski@gmail.com>`).

jow-, 2023-09-08:
@KanjiMonster - it would be great if you could take care of it and send a patch on my behalf (maybe with a Suggested-by or similar)

KanjiMonster, 2023-09-08:
Let's first get a confirmation that it fixes stuff (: @timur-davletshin can you give it a try?

timur-davletshin, 2023-09-08:
> Let's first get a confirmation that it fixes stuff (: @timur-davletshin can you give it a try?

Yep, I will try to compile it on Saturday or Sunday :)

KanjiMonster, 2023-09-09:
I created a commit at https://github.com/KanjiMonster/linux/commit/a8c7560dd393fcb6942f77f367784151774459ea (branch https://github.com/KanjiMonster/linux/tree/geode-rng-fix). I needed to add some casts to avoid warnings during compilation.

Will create a draft PR now based on that.

@timur-davletshin If want, you can have

```
Reported-by: Your Name Here <you@your.email>
Tested-by: Your Name Here <you@your.email>
```

in the patch if you fill that out for me ;-) (optional, not mandatory).

timur-davletshin, 2023-09-09:
Having troubles compiling 23.05 branch from git...

 make[1] world
 make[2] tools/compile
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile
 make[3] -C tools/sed compile
 make[3] -C tools/patch compile
 make[3] -C tools/libdeflate compile
 make[3] -C tools/tar compile
 make[3] -C tools/ninja compile
 make[3] -C tools/meson compile
 make[3] -C tools/pkgconf clean-build
 make[3] -C tools/pkgconf compile
    ERROR: tools/pkgconf failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/timur/Downloads/Temporary/git/openwrt-openwrt-23.05/include/toplevel.mk:232: world] Error 1

timur-davletshin, 2023-09-09:
@KanjiMonster well, if you compile it for me I'd be glad to flash it and test it. Then it will be fair to add my email... which is: timur.davletshin at gmail dot com

KanjiMonster, 2023-09-09:
> Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on

Please do this.

Also I have no idea which image file you need, as I have never worked with x86.

timur-davletshin, 2023-09-09:
It didn't work but I managed to fix it already :)

timur-davletshin, 2023-09-09:
> Please do this.

Well, I failed. I fixed couple simple problems of similar nature. But now I have libubox compilation error and it is beyond my powers. Probably the right way to fix it would be to update my debian oldstable but I don't have time right now for that. If you have working development environment maybe you could compile this target for me https://downloads.openwrt.org/releases/23.05.0-rc3/targets/x86/geode/config.buildinfo and then I'll test it gladly.

KanjiMonster, 2023-09-09:
> Well, I failed. I fixed couple simple problems of similar nature. But now I have libubox compilation error and it is beyond my powers. Probably the right way to fix it would be to update my debian [...]

Do these work? http://www.jonas-gorski.de/geode/ these were the results of my compile test. Note that they are main builds, with no optional anything, and only intended for verification of the fix.

timur-davletshin, 2023-09-10:
> Do these work?

Indeed it worked fine. I was able to feed output to rngd without problems.

BTW, Thanks a lot for spending your time for making those builds!

timur-davletshin, 2023-09-10:
Well, after three reboots and one hour testing I may confirm that your patch is functional.

KanjiMonster, 2023-09-10:
Sent the patch upstream. Found out I could have gotten your email address also from kernel bugzilla, once I found out I already had an account and logged in.

closed 2023-09-11 by openwrt-bot

brada4, 2023-09-21:
amd-rng.c

commit 09d13cd8 (2023-09-11): x86: geode: fix hwrng register accesses

When the membase and pci_dev pointer were moved to a new struct in priv,
the actual membase users were left untouched, and they started reading
out arbitrary memory behind the struct instead of registers. This
unfortunately turned the RNG into a constant number generator, depending
on the content of what was at that offset.

To fix this, update geode_rng_data_{read,present}() to also get the
membase via amd_geode_priv, and properly read from the right addresses
again.

Closes #13417.

Reported-by: Timur I. Davletshin <timur.davletshin@gmail.com>
Tested- [... cut]
PR #13429 (2023-09-11): x86: geode: fix hwrng register accesses
When the membase and pci_dev pointer were moved to a new struct in priv, the actual membase users were left untouched, and they started reading out arbitrary memory behind the struct instead of registers. This unfortunately turned the RNG into a constant number generator, depending on the content of what was at that offset.

To fix this, update geode_rng_data_{read,present}() to also get the membase via amd_geode_priv, and properly read from the right addresses again.

Closes #13417.

Suggested-by: Jo-Philipp Wich <jo@mein.io>
commit 3fe28753 (2023-09-14): x86: geode: fix hwrng register accesses

When the membase and pci_dev pointer were moved to a new struct in priv,
the actual membase users were left untouched, and they started reading
out arbitrary memory behind the struct instead of registers. This
unfortunately turned the RNG into a constant number generator, depending
on the content of what was at that offset.

To fix this, update geode_rng_data_{read,present}() to also get the
membase via amd_geode_priv, and properly read from the right addresses
again.

Closes #13417.

Reported-by: Timur I. Davletshin <timur.davletshin@gmail.com>
Tested- [... cut]
commit ac52e201 (2023-09-14): x86: geode: fix hwrng register accesses

When the membase and pci_dev pointer were moved to a new struct in priv,
the actual membase users were left untouched, and they started reading
out arbitrary memory behind the struct instead of registers. This
unfortunately turned the RNG into a constant number generator, depending
on the content of what was at that offset.

To fix this, update geode_rng_data_{read,present}() to also get the
membase via amd_geode_priv, and properly read from the right addresses
again.

Closes #13417.

Reported-by: Timur I. Davletshin <timur.davletshin@gmail.com>
Tested- [... cut]