openwrt/openwrt #13427: Zbtlink ZBT-WE3526 WiFi 5GHz issue - low output power

sloniupl, 2023-09-08:
### Describe the bug

Wireless driver for 5GHz wifi on this device is showing max 3dBm power to select and as a result 5GHz wifi is not workig (very very weak signal).
It is working correctly on 22.03. Issue only after updating to 23.05
[image]
[image]
[image]

### OpenWrt version

r23389-5deed175a5

### OpenWrt target/subtarget

ramips/mt7621

### Device

Zbtlink ZBT-WE3526

### Image kind

Official downloaded image

### Steps to reproduce

Update to 23.05 and go to wifi section to configure 5GHz wifi

### Actual behaviour

Wifi 5GHz signal very week, unusable for normal use.

### Expected behaviour

Wifi 5GHz signal strong and working correctly on 22.03

### Additional info

BusyBox v1.36.1 (2023-08-19 14:01:06 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0-rc3, r23389-5deed175a5
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# . /etc/openwrt_release && echo $DISTRIB_REVISION
r23389-5deed175a5
root@OpenWrt:~# . /etc/openwrt_release && echo $DISTRIB_TARGET
ramips/mt7621
root@OpenWrt:~# cat /tmp/sysinfo/model
Zbtlink ZBT-WE3526
root@OpenWrt:~# cat /etc/config/wireless 

config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option disabled '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option cell_density '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-08:
You missdd out setting country code in 'advanced' tab for each radio

sloniupl, 2023-09-10:
I've tested before setting up country code but still can't get power higher than 3dBm:

```
BusyBox v1.36.1 (2023-08-19 14:01:06 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0-rc3, r23389-5deed175a5
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# cat /etc/config/wireless 

config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option country 'PL'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option cell_density '0'
        option country 'PL'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
```

brada4, 2023-09-10:
Another more modern mt76.
pci id different as it is different SoC, and AX mode is supported here, your AC setting is correct for your device.
tx power suggested to vote down 13dbm channels in EU, for diagnostics you need to follow `logread -f` as radio comes up for 1-2 minutes.
full powerdown might be necessary if radio cannot take "stronger" regdom after 00 was set once.
```
config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
        option band '5g'
        option cell_density '0'
        option min_tx_power '21'
        option htmode 'HE80'
        option country 'BE'
        option channel 'auto'
```

brada4, 2023-09-10:
(you might be able to catch 3dBm signal 2cm from antenna, but it could be placeholder for TX off as well)

sloniupl, 2023-09-10:
I was able to connect my phone after setting min_tx_power '21' but that does not solve the issue. Wireless power is still low and can't be changed via luci.

`BusyBox v1.36.1 (2023-08-19 14:01:06 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0-rc3, r23389-5deed175a5
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# logread -f
Sat Aug 19 14:11:49 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy phy1-ap0) --> new PHY
Sat Aug 19 14:11:49 2023 kern.info kernel: [   55.534087] br-lan: port 5(phy1-ap0) entered blocking state
Sat Aug 19 14:11:49 2023 kern.info kernel: [   55.539709] br-lan: port 5(phy1-ap0) entered disabled state
Sat Aug 19 14:11:49 2023 kern.info kernel: [   55.546025] device phy1-ap0 entered promiscuous mode
Sat Aug 19 14:11:49 2023 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat Aug 19 14:11:49 2023 daemon.notice hostapd: phy1-ap0: interface state COUNTRY_UPDATE->HT_SCAN
Sat Aug 19 14:11:50 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sat Aug 19 14:11:50 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Sat Aug 19 14:11:50 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Sat Aug 19 14:11:50 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sat Aug 19 14:11:50 2023 kern.info kernel: [   56.412013] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
Sat Aug 19 14:11:50 2023 kern.info kernel: [   56.419103] br-lan: port 5(phy1-ap0) entered blocking state
Sat Aug 19 14:11:50 2023 kern.info kernel: [   56.424753] br-lan: port 5(phy1-ap0) entered forwarding state
Sat Aug 19 14:11:50 2023 daemon.notice hostapd: phy1-ap0: interface state HT_SCAN->ENABLED
Sat Aug 19 14:11:50 2023 daemon.notice hostapd: phy1-ap0: AP-ENABLED
Sat Aug 19 14:11:50 2023 daemon.notice netifd: Wireless device 'radio1' is now up
Sat Aug 19 14:11:50 2023 daemon.notice netifd: Network device 'phy1-ap0' link is up
Sat Aug 19 14:11:51 2023 daemon.warn odhcpd[1558]: No default route present, overriding ra_lifetime!
root@OpenWrt:~# cat /etc/config/wireless 

config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option country 'PL'
        option cell_density '0'
        option disabled '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
        option disabled '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option cell_density '0'
        option country 'PL'
        option txpower '21'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

root@OpenWrt:~# `

brada4, 2023-09-10:
Please show `iwinfo` (actual tx power) and `iw reg get` (pivot of country rules and radio calibration data)
seems wrong radio calibration data is read from vendor partitions.
If you have chance and time  make same 2 reports with 22.03 to have side by side comparison.

brada4, 2023-09-11:
It was initially ported by @nbd168

sloniupl, 2023-09-11:
I noticed that in 22.03 interface is called wlan1 and on 23.05 phy1-ap0 also I noticed that wireless AP mac addresses are correct in 22.03 and not correct in 23.05 (don't know if that helps or not).

I have this device for tests as I use 2 exactly the same devices at my and my parents house so I want  to figure out the issue and help solve it for 23.05 so I can perform any tests necessary to help figure it out.

brada4, 2023-09-11:
Is MAC address of ethernet matching the label?
https://github.com/openwrt/openwrt/commits/main/target/linux/ramips/dts/mt7621_zbtlink_zbt-we3526.dts

sloniupl, 2023-09-11:
Not sure how to check it but my MAC for LAN and wan is:
78:A3:51:63:38:EA
78:A3:51:63:38:EB

[image]

brada4, 2023-09-11:
Should be on a sticker on underside https://openwrt.org/toh/zbtlink/zbt_we3526
Yhat MAC does not seem registeted to a manufacturer.

sloniupl, 2023-09-11:
Yep that 20:55:3F is just something that the WiFi 5GHz (phy1-ap0) is getting on OpenWRT 23.05. I Imaged again 23.05 on this device for future testing and it got a different MAC for WiFi interface than it had before on 23.05 (phy1-ap0  ESSID: "OpenWrt" Access Point: 24:6C:AA:9F:2B:02) so i belive it can be part of the problem.

brada4, 2023-09-11:
Wifi calibration data is obtained from wrong place following one of latest changes...

sloniupl, 2023-09-11:
I've tested rc1 and rc2 of 23.05 to check if I have the same issue on those versions of 23.05 and looks like the issue is there.
Is there any way I can help solve it before stable 23.05 release?

brada4, 2023-09-11:
If you want to build image you may try to bisect dts and dtsi changes i e count changes apply first half try if it works, like number guess game till you find commit breaking device.
Problem is isolated to 5ghz radio only that mac and cal data are wrong.

sloniupl, 2023-09-11:
Will do, so if i understand correctly you want me to build a 23.05 rc3 build for testing with different version of openwrt/target/linux/ramips/dts/mt7621_zbtlink_zbt-we3526.dts ?

For example replacing most up to date comit with the older one in build directory and than build it, flash to the router and see if it will work?
[image]

Sorry for all the questions but i'll do it first time :)

sloniupl, 2023-09-11:
Funny thing, i build my own image for 23.05-rc3 using this guide:
https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem

As a target i used ZBT-WG3526 16M:
[image]
and it is working correctly:
[image]
[image]

```
BusyBox v1.36.1 (2023-08-19 14:01:06 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
sloniupl.eu test image for ramips/mt7621 - Zbtlink ZBT-WE3526
 -----------------------------------------------------
 OpenWrt 23.05.0-rc3, r23389-5deed175a5
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# iwinfo
phy1-ap0  ESSID: "OpenWrt"
          Access Point: 78:A3:51:63:38:E9
          Mode: Master  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 21 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

wlan0     ESSID: unknown
          Access Point: 62:E5:F8:11:60:4E
          Mode: Client  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 35 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7603 14C3:7603 [MediaTek MT7603E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

root@OpenWrt:~# iw reg get
global
country PL: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5875 @ 80), (N/A, 13), (N/A)
        (5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

root@OpenWrt:~# 
```

sloniupl, 2023-09-11:
I tested doing the same thing with ZBT-WG3526 32M but after flashing i did not have the wireless section under network despite Status page showing Wireless interfaces (but not active - no way to enable it via gui):

[image]
[image]

Checking console commands you provided me i found that on this 32M version 5GHz wifi is showing as 3dBm 

```
BusyBox v1.36.1 (2023-08-19 14:01:06 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
sloniupl.eu test image for ramips/mt7621 - Zbtlink ZBT-WE3526
 -----------------------------------------------------
 OpenWrt 23.05.0-rc3, r23389-5deed175a5
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# iwinfo
wlan0     ESSID: unknown
          Access Point: EE:21:78:EA:B0:59
          Mode: Client  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 35 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7603 14C3:7603 [MediaTek MT7603E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: unknown
          Access Point: 78:A3:51:63:38:E9
          Mode: Client  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 3 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

root@OpenWrt:~# iw reg get
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

root@OpenWrt:~# 
```

Any idea how to build a Generic image for ZBT-WE3526 similar how the official images are build?
When doing make menuconfig i only have 16M or 32M to choose from:

[image]

sloniupl, 2023-09-11:
Ok disregard my last two posts. I created images for WG3526 and not WE3526. I'll do one now and test it.

DragonBluep, 2023-09-12:
@sloniupl Can you show us the output of the `dmesg`?

sloniupl, 2023-09-12:
Sure, Here is dmesg from official 23.05 rc3 (i configured country code on 5GHz wifi and enabled it):

```
BusyBox v1.36.1 (2023-08-19 14:01:06 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0-rc3, r23389-5deed175a5
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# dmesg
[    0.000000] Linux version 5.15.127 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23389-5deed175a5) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Sat Aug 19 14:01:06 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Zbtlink ZBT-WE3526
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s14736 r8192 d22128 u45056
[    0.000000] pcpu-alloc: s14736 r8192 d22128 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00008bc0
[    0.000000] Readback ErrCtl register=00008bc0
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 249124K/262144K available (7296K kernel code, 627K rwdata, 880K rodata, 1232K init, 226K bss, 13020K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.008053] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066217] pid_max: default: 32768 minimum: 301
[    0.071648] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.078860] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.090582] rcu: Hierarchical SRCU implementation.
[    0.096076] smp: Bringing up secondary CPUs ...
[    0.101564] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.101595] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.101610] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.101657] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.160867] Synchronize counters for CPU 1: done.
[    0.193276] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.193298] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.193310] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.193339] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.252336] Synchronize counters for CPU 2: done.
[    0.283007] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[... 234 lines cut ...]
[   30.274478] mt7530-mdio mdio-bus:1f lan4: configuring for phy/gmii link mode
[   30.283937] br-lan: port 4(lan4) entered blocking state
[   30.289252] br-lan: port 4(lan4) entered disabled state
[   30.297510] device lan4 entered promiscuous mode
[   30.326433] mtk_soc_eth 1e100000.ethernet wan: PHY [mdio-bus:04] driver [MediaTek MT7530 PHY] (irq=POLL)
[   30.336001] mtk_soc_eth 1e100000.ethernet wan: configuring for phy/rgmii link mode
[   33.603846] mt7530-mdio mdio-bus:1f lan4: Link is Up - 1Gbps/Full - flow control rx/tx
[   33.611869] br-lan: port 4(lan4) entered blocking state
[   33.617160] br-lan: port 4(lan4) entered forwarding state
[   33.623553] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   65.443380] done.
[   65.445396] jffs2: notice: (2207) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[   65.759384] overlayfs: upper fs does not support tmpfile.
[   88.886843] br-lan: port 5(phy1-ap0) entered blocking state
[... cut ...]
[   89.724901] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
[   89.732328] br-lan: port 5(phy1-ap0) entered blocking state
[   89.738136] br-lan: port 5(phy1-ap0) entered forwarding state
[  100.663327] nf_conntrack: nf_conntrack: table full, dropping packet
[  100.684391] nf_conntrack: nf_conntrack: table full, dropping packet
[  101.695069] nf_conntrack: nf_conntrack: table full, dropping packet
[  103.708476] nf_conntrack: nf_conntrack: table full, dropping packet
root@OpenWrt:~# 
```

DragonBluep, 2023-09-12:
Are you sure your device model is `Zbtlink ZBT-WE3526`? The image for `Zbtlink ZBT-WE1326` should work perfectly for you. Please test it.

sloniupl, 2023-09-12:
Thank you very much, indeed it looks like i have WE1326.
After your message i disassembled the device i have and removed MAC address sticker that is on motherboard. Under that was a model number:
[image]
And it works correctly on the correct firmware:
[image]

Those devices i have and use are rebranded versions of Zbtlink models. And i was basing the model name on the original OpenWRT firmware that was installed on it when i got my first unit:
[image]
[Router Korbox Gigabitus R1219](https://blog.sloniupl.eu/2020/10/24/router-korbox-gigabitus-r1219-zbt-we3526-nowy-firmware/)

closed 2023-09-12 by sloniupl