openwrt/openwrt #13428: toolchain/binutils link error on ubuntu 18.04

ptpt52, 2023-09-09:
### Describe the bug

```
make V=s toolchain/binutils/compile
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/scripts/config'
make[1]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt'
+ mkdir -p /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-mips_24kc_musl
+ cd /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-mips_24kc_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mips_24kc_musl/stamp
touch /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-mips_24kc_musl/.prepared
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/toolchain/binutils'
make  -C /mnt/Data/Sources/openwrt/x-wrt/build_dir/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40 all
make[3]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40'
make[4]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40'
make[5]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/libiberty'
make[6]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/libiberty/testsuite'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/libiberty/testsuite'
make[5]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/libiberty'
make[5]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/intl'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/intl'
make[5]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/libsframe'
make  all-am
make[6]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/libsframe'
make[6]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/libsframe'
make[5]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/libsframe'
make[5]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/bfd'
Making info in po
make[6]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/bfd/po'
make[6]: Nothing to be done for 'info'.
make[6]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/bfd/po'
make[6]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/bfd'
make[6]: Circular doc/bfd.stamp <- doc/bfd.texi dependency dropped.
make[6]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/bfd'
make  all-recursive
make[6]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/bfd'
Making all in po
make[7]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/bfd/po'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/bfd/po'
make[7]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/bfd'
make[7]: Circular doc/bfd.stamp <- doc/bfd.texi dependency dropped.
make[7]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/bfd'
make[6]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/bfd'
make[5]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/bfd'
make[5]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/opcodes'
make  all-recursive
make[6]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/opcodes'
Making all in .
make[7]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/opcodes'
make[7]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/opcodes'
Making all in po
make[7]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/opcodes/po'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/opcodes/po'
make[6]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/opcodes'
make[5]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/opcodes'
make[5]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/gas'
make  all-recursive
make[6]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/gas'
Making all in po
make[7]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/gas/po'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/gas/po'
make[7]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/toolchain-mips_24kc_gcc-12.3.0_musl/binutils-2.40/gas'
/usr/bin/env bash ./libtool  --tag=CC   --mode=link /mnt/Data/Sources/openwrt/x-wrt/staging_dir/host/bin/gcc -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wstack-usage=262144 -Wwrite-strings  -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-mips_24kc_gcc-12.3.0_musl/include -pthread -O2 -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/host/include       -L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/host/lib  -o as-new app.o as.o atof-generic.o codeview.o compress-debug.o cond.o depend.o dwarf2dbg.o dw2gencfi.o ecoff.o ehopt.o expr.o flonum-copy.o flonum-konst.o flonum-mult.o frags.o gen-sframe.o hash.o input-file.o input-scrub.o listing.o literal.o macro.o messages.o output-file.o read.o remap.o sb.o sframe-opt.o stabs.o subsegs.o symbols.o write.o config/tc-mips.o config/obj-elf.o config/atof-ieee.o itbl-parse.o itbl-lex-wrapper.o itbl-ops.o  config/e-mipself.o ../opcodes/libopcodes.la ../bfd/libbfd.la ../libiberty/libiberty.a   -lz -L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-mips_24kc_gcc-12.3.0_musl/lib -lzstd -pthread -ldl 
[... 77 lines cut ...]
/mnt/Data/Sources/openwrt/x-wrt/include/toplevel.mk:230: recipe for target 'toolchain/binutils/compile' failed
make: *** [toolchain/binutils/compile] Error 2
```

but if I remove `rm -rf staging_dir/toolchain-mips_24kc_gcc-12.3.0_musl/` then rebuild, it pass.

### OpenWrt version

xx

### OpenWrt target/subtarget

xx

### Device

xx

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Entropy512, 2023-09-12:
18.04 has been EOL for a few months - https://ubuntu.com/blog/18-04-end-of-standard-support

t-8ch, 2023-09-18:
Same for me when building for ipq40xx on up-to-date ArchLinux.
This worked fine before.

t-8ch, 2023-09-18:
Went away after a full rebuild.