openwrt/openwrt #13430: ath10k + ipq4018 does not pass multicast to client station

Npeca75, 2023-09-09:
### Describe the bug

https://forum.openwrt.org/t/hap-ac2-ipq4018-hostapd-dynamic-vlans/170666/

ipv4 dhcp and ipv6 slaac does not work on ATH10K bonded to DSA bridge
only static ipv4 will work
description and debugging in linked thread

### OpenWrt version

r23899-5294b358aa

### OpenWrt target/subtarget

ipq40xx/mikrotik

### Device

HaP ac2 ipq4018

### Image kind

Official downloaded image

### Steps to reproduce

configs are in linked post

### Actual behaviour

no multicast dst 255.255.255.255 are send to client device

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-09:
Does /etc/config/dhcp enable dhcp and slaac for each "interface" from /etc/config/network ?

Npeca75, 2023-09-09:
> Does /etc/config/dhcp enable dhcp and slaac for each "interface" from /etc/config/network ?

hi @brada4 
as i wrote in forum, this is a dumb AP, only purpose for this device is to get vlans on trunk port and server VLANs on WIFI
my choice was to use dynamic VLANs with hostapd, because i want to push about 10 pcs ov VLANs to WIFI, so it is out of question to use 10 different SSID

this dynamic VLANs config working well with swconfig/ramips and swconfig/ath97 devices

but fail with ipq+ath10k

Npeca75, 2023-09-09:
to be clear, here is the screenshots from wireshark from client PC
it sendind discover, but never get multicast (255.255.255.255) response

[image]

Npeca75, 2023-09-09:
and this is screenshot from OpenWRT wifi interface bonded on vlan100
here everything looks good
[image]

Npeca75, 2023-09-09:
and this is wireshark from OpenWRT captured on DSA bridge
also looks good
vlanid=100 
[image]

brada4, 2023-09-09:
Does cable connected to that vlan untagged get addressing?

Npeca75, 2023-09-09:
> Does cable connected to that vlan untagged get addressing?

yes, as i state in forum post, ethernet side of this setup is working as expected
i am able to tag/untag every VLAN on any DSA port

Npeca75, 2023-09-09:
Tried wit CT and nonCT drivers

further information
```
bridge vlan show
port              vlan-id  
lan4              255 PVID Egress Untagged
lan3              200 PVID Egress Untagged
lan2              100 PVID Egress Untagged
lan1              2 PVID Egress Untagged
wan               1 PVID Egress Untagged
                  2
                  100
                  200
                  255
switch            1
                  2
                  100
                  200
                  255
wlan1-vl200       200 PVID Egress Untagged
wlan1-vl100       100 PVID Egress Untagged
wlan1-vl2         2 PVID Egress Untagged
wlan0-vl200       200 PVID Egress Untagged
wlan0-vl100       100 PVID Egress Untagged
wlan0-vl2         2 PVID Egress Untagged
```

```
lsmod
ath                    24576  1 ath10k_core
ath10k_core           307200  1 ath10k_pci
ath10k_pci             40960  0 
cfg80211              512000  3 ath10k_core,ath,mac80211
cmac                   16384  0 
compat                 16384  3 ath10k_pci,mac80211,cfg80211
crc_ccitt              16384  0 
crc32c_generic         16384  1 
dwc3                   45056  0 
dwc3_qcom              16384  0 
ghash_arm_ce           20480  0 
gpio_button_hotplug    16384  0 
leds_gpio              16384  0 
libcrc32c              16384  0 
mac80211              610304  1 ath10k_core
sha512_arm             24576  0 
xhci_hcd              139264  2 xhci_plat_hcd,xhci_pci
xhci_pci               16384  0 
xhci_plat_hcd          16384  0 
```

Npeca75, 2023-09-09:
```
cat /var/run/hostapd-wlan0.psk 
vlanid=2  00:00:00:00:00:00 test2-passw
vlanid=100  00:00:00:00:00:00 test100-passw
vlanid=200  00:00:00:00:00:00 test200-passw
```

```
cat /var/run/hostapd-wlan0.vlan 
2 wlan0-vl2
100 wlan0-vl100
200 wlan0-vl200
```

```
cat /var/run/hostapd-phy0.conf 
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=HU
ieee80211d=1
hw_mode=g
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 120 240
beacon_int=100
channel=6
chanlist=6
noscan=1

ieee80211n=1
ht_coex=0
ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40]

interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
ap_max_inactivity=120
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=0
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
nas_identifier=0e0199000199
wpa_passphrase=SomeUnusedPass
wpa_psk_file=/var/run/hostapd-wlan0.psk
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=t-AP
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
dynamic_vlan=0
vlan_naming=1
vlan_no_bridge=1
vlan_file=/var/run/hostapd-wlan0.vlan
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
bssid=0e:01:99:00:01:99
```

brada4, 2023-09-09:
What do you get with tcpdump -p -c 100 -w /tmp/something.pcap 
on wlan1-vl100 wlan0-vl100 ?

According to hostapd.conf you do not use dynamic vlans https://w1.fi/cgit/hostap/tree/hostapd/hostapd.conf#n1580
Where is network channel/bonding involved?

Npeca75, 2023-09-09:
screenshot of wlan0-vl100 is second screnshot
but, as you wish, i could make pcap file

yes, i saw this 
> dynamic_vlan=0

but on other ramips/swconfig and ath79/swconfig it is same
looks like openwrt netifd work some magic ....

here is config
```
cat /etc/config/wireless 

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option phy 'phy0'
        option country 'HU'
        option legacy_rates '0'
        option noscan '1'
        option txpower '20'
        option htmode 'HT20'
        option disabled '0'
        option distance 'auto'
        option channel '5'

config wifi-iface 'wifinet0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'ele-AP'
        option encryption 'psk2+aes'
        option key 'SomeUnusedPass'
        option wmm '1'
        option short_preamble '1'
        option disassoc_low_ack '0'
        option max_inactivity '120'
        option isolate '1'
        option disabled '0'
        option ifname 'wlan0'
        option macaddr '0e:01:99:00:01:01'

config wifi-vlan
        option name 'vl2'
        option network 'vlan2'
        option vid '2'

config wifi-station
        option key 'test2-passw'
        option vid '2'

config wifi-vlan
        option name 'vl100'
        option network 'vlan100'
        option vid '100'

config wifi-station
        option key 'test100-passw'
        option vid '100'

config wifi-vlan
        option name 'vl200'
        option network 'vlan200'
        option vid '200'

config wifi-station
        option key 'test200-passw'
        option vid '200'
```

Npeca75, 2023-09-09:
ok @brada4 
here is dump 

```
tcpdump -i wlan0-vl100 -p -c 100 -w /tmp/wlan0-100.pcap
tcpdump: listening on wlan0-vl100, link-type EN10MB (Ethernet), snapshot length 262144 bytes
100 packets captured
101 packets received by filter
0 packets dropped by kernel
```
[wlan0-100.zip](https://github.com/openwrt/openwrt/files/12566816/wlan0-100.zip)

brada4, 2023-09-09:
You say frame 91 never reaches through ath10k wifi?
Try `option multicast_to_unicast '1'` for that test AP?

Npeca75, 2023-09-09:
> You say frame 91 never reaches through ath10k wifi?

yes @brada4 
there is NO ipv4 dhcp and NO ipv6 slaac over wifi, which depend on broadcast ? multicast ? sry it is late night here
only static v4 working because ARP is unicast if i remember right

>Try option multicast_to_unicast '1' for that test AP?
ok, will try

brada4, 2023-09-09:
ARP is broadcast too just at lower level. No ARP before DHCP though.

Npeca75, 2023-09-09:
> Try `option multicast_to_unicast '1'` for that test AP?

tried with 0 and 1
same thing, no dhcp :(

```
config wifi-iface 'wifinet0'
        option device 'radio0'
        option mode 'ap'
        option ssid 't-AP'
        option encryption 'psk2+aes'
        option wmm '1'
        option short_preamble '1'
        option disassoc_low_ack '0'
        option max_inactivity '120'
        option isolate '1'
        option disabled '0'
        option key 'SomeUnusedPass'
        option ifname 'wlan0'
        option macaddr '0e:01:99:00:01:99'
        option multicast_to_unicast '1'
```

Npeca75, 2023-09-09:
> ARP is broadcast too just at lower level. No ARP before DHCP though.

hmmm, i read that ARP is unicast
https://serverfault.com/questions/409865/why-arp-request-is-unicast

brada4, 2023-09-09:
Default is zero, only non-default things i see no explicit "network" and non-default isolation in AP config, first should render to "bridge=" in hostapd.conf

ARP is multicast for initial discovery for obvious reasons. That is invalidation/refresh targetted probe like hey gateway you still there? You see few lines later broadcast discovery.

Npeca75, 2023-09-09:
> i see no explicit "network" and non-default isolation in AP config, first should render to "bridge=" in hostapd.conf

because it is dynamic vlan config
i don't need default network, my idea is to use only vlans 2,100,200, etc
same config WORK on other devices
and as you see from bridge vlan show, wifi interfaces are correctly assigned to "switch" bridge

but ... ok, i will try same config with default network tomorrow

Npeca75, 2023-09-09:
> You see few lines later broadcast discovery.

broadcast is working in ONE direction, from PC to OpenWRT
but opposite direction not, broadcast is lost somewhere in hostapd/wifi ...

brada4, 2023-09-10:
Yes, I see, like tags in dumps look correct too, at least even frame 91 gets to adapter....
It works when wireless adapter is connected to bridge+vlan interface static ?

Npeca75, 2023-09-10:
so, after all, this bugs is same as
https://github.com/openwrt/openwrt/issues/9555
and the "solution" / workaround is

option multicast_to_unicast_all '1'

but, it is still workaround
:(

brada4, 2023-09-10:
when settimg fixed bridge as network it sets 'bridge' and 'snoop'  options to that bridge  in hostapd.conf. later smells mcast...

brada4, 2023-09-10:
Since other issue is mt76 it is software or configuration issue.

yogo1212, 2024-10-16:
i'm currently facing a similar problem except i broke it down to (what i hope is) the minimal configuration.

cut out the vlan-to-psk thing, cut out dynamic vlans on the wifi side, removed dhcp, and reduced it to this:

```
config interface 'd_vlan12'                  
        option force_link '1'             
        option device 'br-vlan12'          
        option ip4table '8'                    
        option ip6table '8'     
                                  
config device 'vlan12'            
        option type 'bridge'      
        option isolate '1'        
        option bridge_empty '1'           
        option name 'br-vlan12'    
        list ports 'lan.12' 
        list ports 'wan.12'
```

wireless:
```
config wifi-device 'radio5_0'
        option type 'mac80211'
        option path 'platform/soc/18000000.wifi+1'
        option band '5g'
        option channel '36'
        option htmode 'HE160'
        option country 'DE'
        
config wifi-iface 'w5_0-vlan12'                
        option mode 'ap'                     
        option ifname 'w5_0-vlan12'                 
        option device 'radio5_0'             
        option network 'vlan12'               
        option encryption 'psk2'             
        option isolate '1'                       
        option key '12345678'                
        option ieee80211r '1'              
        option ft_over_ds '0'                
        option ft_psk_generate_local '0'     
        option ieee80211w '1'                
        option ssid 'vlan test'                   
```

`option multicast_to_unicast_all '1'` doesn't do anything, afaics. i added a static neighbour to rule out multicast being an issue (`ip n add 10.10.0.2 lladdr d0:7e:35:b5:e6:86 dev br-vlan12`) but that also changes nothing.

my test is a simple `ping 10.10.0.1` from the wifi client.
the router sees the response from the gateway:
```
08:23:07.490639 ARP, Request who-has 10.10.0.1 tell 10.10.0.2, length 28
08:23:07.491026 ARP, Reply 10.10.0.1 is-at c4:54:44:b9:16:99, length 46
```

the client only sees the request - but no responses. similar to the dhcp response missing for @Npeca75 .

another thing i noticed is that the router also doesn't see the outgoing traffic on the wifi interface:

```
08:26:19.007251 ARP, Request who-has 10.10.0.1 tell 10.10.0.2, length 28
08:26:20.020600 ARP, Request who-has 10.10.0.1 tell 10.10.0.2, length 28
```
but i'm skeptical about the responses passing through that part of the linux network stack even if everything was working.

so my conclusion is also that the responses are lost somewhere between the bridge and the .

originally, i encountered this with ath10k + ipq4028 (compex,wpj428) in 2022. since then, i switched to mt76 (yuncore,ax835) and it also doesn't work there either.
my last known-to-be-working setup was using ath9k running the same version of openwrt in 2022. newer versions of openwrt don't fit on my 8mb flash and testing years-old software is kinda useless.

hoping to help resolve this bug :blush: 
regards

yogo1212, 2024-10-16:
i forgot to mention:

pinging the client from the ap works (after add an ip: `ip a add 10.10.0.3/24 dev br-vlan12; ping 10.10.0.2`).
pinging from the cable network doesn't (no request seen on the client).
the client can also ping the AP.

it's just the path **through** the ap that doesn't work.

yogo1212, 2024-10-18:
`should_deliver` (br_forward.c) returns false?!

need more time for this

yogo1212, 2024-10-19:
`nbp_switchdev_allowed_egress(p, skb)` is false.

```
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.595576] br_forward dev w5_0-13c2 ok: 1, local rcv 0
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.601158] should_deliver      p->flags & BR_HAIRPIN_MODE: 0
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.606890] should_deliver   or skb->dev != p->dev: 1
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.611940] should_deliver and
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.614982] should_deliver      br_mst_is_enabled(p->br): 0
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.620544] should_deliver   or p->state == BR_STATE_FORWARDING: 1
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.626709] should_deliver and
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.629755] should_deliver      br_allowed_egress(vg, skb): 1
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.635486] should_deliver and
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.638531] nbp_switchdev_allowed_egress      !test_bit(p->hwdom, &cb->fwd_hwdoms): 1
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.646344] nbp_switchdev_allowed_egress and
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.650603] nbp_switchdev_allowed_egress     !skb->offload_fwd_mark: 0
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.657116] nbp_switchdev_allowed_egress  or cb->src_hwdom != p->hwdom: 0
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.663892] should_deliver      nbp_switchdev_allowed_egress(p, skb): 0
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.670494] should_deliver and
Sat Oct 19 00:16:40 2024 kern.err kernel: [  649.673534] should_deliver      !br_skb_isolated(p, skb): 1
```

yogo1212, 2024-10-19:
does this have to do with https://lists.linuxfoundation.org/pipermail/bridge/2021-July/014340.html?

hwdom is not not in sysfs.. sigh.
it takes ~40 minutes for me to swap the kernel (with new debug statements) because of this crazy old laptop..

yogo1212, 2024-10-19:
yup. both hwdom 0...
let's try a newer kernel with the patch. maybe there's actually no more dev work here :crossed_fingers:

yogo1212, 2024-10-19:
i think what happens is:

- hwdom is not populated for the switchdev belonging to the tagged ethernet interface (i assume, because it's tagged)
- hwdom is not populated for the wifi interface but `tx_fwd_offload` is enabled

then:

- a packet arrives with vlan tag 20
- (kernel finds the correct egress interface using the mac address, which has a static entry in the fdb)
- (`br_handle_vlan` strips the vlan tag from the skb because the vlan is not used inside the bridge (no vlan filtering) and `tx_fwd_offload` is not set. this, i assume, also reflects the inability of the offloader to handle tagged packets)
- the skb is marked for `tx_fwd_offload` in `__br_forward`, because the egress port has it enabled.
- finally, the actual check `nbp_switchdev_allowed_egress` looks at the hwdoms (both 0) and, because they're equal, assumes the underlying switch already did the egress **but there is no such switch containing `lan.12` and `w5_0-vlan12`, so that's not the case**

if i'm correct (which i'm not entirely convinced of), the hwdom needs to be checked for being `!= 0` as well.
if i'm wrong about hwdom=0 being different from other values, there is a bug that causes both interfaces being in hwdom=0 despite belonging to different phys.

**BUT** this sort of setup should be fairly common while these kind of bug reports are relatively sporadic, i'd assume somebody would have noticed already, or not?

i'm not sure what the right fix is. my best guess is that `hwdom = 0` should never be set in `fwd_hwdoms` and the egress check should account for that. or at least it shouldn't be set if it has been forwarded to all interfaces (haven't looked into broadcast/multicast yet).
also, this seems a trivial case which (the much more skilled than me) upstream developers will have thought of?

nbd168, 2024-10-19:
@yogo1212 I think your analysis is mostly correct - nice job tracking this one down! One reason why fewer people might have run into this issue is because the setup you're describing is typically configured through a VLAN filtering bridge instead of adding VLAN devices as bridge members.

Unless I'm missing something, I think you might have misread the tx_fwd_offload / fwd_hwdoms situation.
nbp_switchdev_allowed_egress checks if the port hwdom bit is *NOT* set in cb->fwd_hwdoms, so I don't think that tx_fwd_offload is actually set for the wifi or vlan device. However, that makes your theory regarding the missing hwdom != 0 check even more likely, since fewer conditions are needed to trigger it.

Regading your proposed fix - I do agree with adding a check for hwdom = 0. I've taken a look at the code, and I believe a valid hwdom should never have zero id.
One other thing that would also prevent the bug is clearing skb->offload_fwd_mark whenever the skb is received on a non-bridge interface. However, that would be slightly more expensive, since it would have to be added in a commonly called core network stack codepath.

My suggestion would be proposing something like this upstream: https://nbd.name/p/1b11e320
Assuming it works in your tests, do you want to send it upstream, or should I?

yogo1212, 2024-10-19:
@nbd168 Thanks for praise!

> Unless I'm missing something, I think you might have misread the tx_fwd_offload / fwd_hwdoms situation.
nbp_switchdev_allowed_egress checks if the port hwdom bit is NOT set in cb->fwd_hwdoms, so I don't think that tx_fwd_offload is actually set for the wifi or vlan device.

Sorry for the confusion. I'm tumbling over my own thoughts. My concern was with multicast traffic.
That the zero bit should never be set for hwdom=0 because only the first interface with hwdom=0 would send the packet and the others would miss out (`nbp_switchdev_frame_mark_tx_fwd_to_hwdom` sets bit 0 `fwd_hwdoms`).

> Regading your proposed fix - I do agree with adding a check for hwdom = 0. I've taken a look at the code, and I believe a valid hwdom should never have zero id.

I'm glad you agree because I'm still unsure.
To retain multicast functionality, the `!p->hwdom` check would have be placed before `!test_bit(..)`:

```diff,c
-	return !test_bit(p->hwdom, &cb->fwd_hwdoms) &&
+	return !p->hwdom || !test_bit(p->hwdom, &cb->fwd_hwdoms) &&
 		(!skb->offload_fwd_mark || cb->src_hwdom != p->hwdom);
 }
 ```
 
Going by your statement about the vlan filtering being more typical, there is probably a discussion to be had about what lies on the 'hot' code path (one that I feel out of place in). I suspect the `!test_bit(..)` was placed first to efficiently deal with multicast traffic on devices with multiple 'chip' switches (like big 'rack' switches), so I would go with your recommendation.
However, at least one other function needs to be changed as well for multicast to work:

```diff,c
 void nbp_switchdev_frame_mark_tx_fwd_to_hwdom(const struct net_bridge_port *p,
                 struct sk_buff *skb)
 { 
-	if (nbp_switchdev_can_offload_tx_fwd(p, skb))
+	if (p->hwdom && nbp_switchdev_can_offload_tx_fwd(p, skb))
 		set_bit(p->hwdom, &BR_INPUT_SKB_CB(skb)->fwd_hwdoms);
 }
```

Here, I'm putting the `p->hwdom` first because it's significantly cheaper than `nbp_switchdev_can_offload_tx_fwd`. `nbp_switchdev_frame_mark_tx_fwd_to_hwdom` is called by `maybe_deliver` which, in turn, is called by `br_flood` and `br_multicast_flood`.

With your permission, I would submit your proposal and my change to `nbp_switchdev_frame_mark_tx_fwd_to_hwdom`, adding you in a `Co-Developed-By:` - once I'm finished testing the change.

nbd168, 2024-10-19:
@yogo1212, Why does `nbp_switchdev_frame_mark_tx_fwd_to_hwdom` need to be changed? From what I can tell, `nbp_switchdev_can_offload_tx_fwd` can only be true if `p->flags & BR_TX_FWD_OFFLOAD` is true.
Unless I'm misinterpreting what you wrote, I didn't see anything in your logs which proves that `BR_TX_FWD_OFFLOAD` is actually set for a port with hwdom=0. The only call chain that I can find that would set the flag is a call to `switchdev_bridge_port_offload`, which is only called from dsa and a few drivers.
Any port that has `BR_TX_FWD_OFFLOAD` set should have a non-zero hwdom, since `nbp_switchdev_hwdom_set` allocates them starting at index 1.
Either way, placing the `!p->hwdom` check before the `!test_bit(...)` and using `Co-developed-by` is fine with me as well.

yogo1212, 2024-10-19:
> Any port that has BR_TX_FWD_OFFLOAD set should have a non-zero hwdom.

I missed this :+1: My mistake was thinking that `BR_TX_FWD_OFFLOAD` and `br_input_skb_cb.tx_fwd_offload` are both implied by `dsa_notifier_bridge_info.tx_fwd_offload`.

commit 2a1daeaa (2024-10-20): kernel: fix forwarding broadcast packets between vlan-on-dsa and non-dsa netdevs

Fix conditions for handling offloaded packets

Fixes: #13430
closed 2024-10-20 by openwrt-bot