openwrt/openwrt #13435: mac80211：Updated to 6.5 Build Errors！

zxlhhyccc, 2023-09-11:
### Describe the bug

````
make[3]: Entering directory '/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5'
make[4]: 'Kconfig.versions' is up to date.
make[6]: 'Kconfig.versions' is up to date.
make[7]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/compat/main.o
In file included from ./arch/arm64/include/asm/pointer_auth.h:7,
                 from ./arch/arm64/include/asm/processor.h:43,
                 from ./include/asm-generic/qrwlock.h:18,
                 from ./arch/arm64/include/generated/asm/qrwlock.h:1,
                 from ./arch/arm64/include/asm/spinlock.h:9,
                 from ./include/linux/spinlock.h:94,
                 from /home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/backport-include/linux/spinlock.h:3,
                 from ./include/linux/kref.h:16,
                 from ./include/linux/mm_types.h:8,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from /home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/backport-include/linux/module.h:3,
                 from /home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/compat/main.c:1:
/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/backport-include/linux/random.h:27:19: error: static declaration of '__get_random_u32_below' follows non-static declaration
   27 | static inline u32 __get_random_u32_below(u32 ceil)
      |                   ^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/backport-include/linux/random.h:3:
./include/linux/random.h:54:5: note: previous declaration of '__get_random_u32_below' with type 'u32(u32)' {aka 'unsigned int(unsigned int)'}
   54 | u32 __get_random_u32_below(u32 ceil);
      |     ^~~~~~~~~~~~~~~~~~~~~~
/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/backport-include/linux/random.h:66:19: error: redefinition of 'get_random_u32_below'
   66 | static inline u32 get_random_u32_below(u32 ceil)
      |                   ^~~~~~~~~~~~~~~~~~~~
./include/linux/random.h:61:19: note: previous definition of 'get_random_u32_below' with type 'u32(u32)' {aka 'unsigned int(unsigned int)'}
   61 | static inline u32 get_random_u32_below(u32 ceil)
      |                   ^~~~~~~~~~~~~~~~~~~~
/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/backport-include/linux/random.h:104:19: error: redefinition of 'get_random_u32_above'
  104 | static inline u32 get_random_u32_above(u32 floor)
      |                   ^~~~~~~~~~~~~~~~~~~~
./include/linux/random.h:99:19: note: previous definition of 'get_random_u32_above' with type 'u32(u32)' {aka 'unsigned int(unsigned int)'}
   99 | static inline u32 get_random_u32_above(u32 floor)
      |                   ^~~~~~~~~~~~~~~~~~~~
/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/backport-include/linux/random.h:116:19: error: redefinition of 'get_random_u32_inclusive'
  116 | static inline u32 get_random_u32_inclusive(u32 floor, u32 ceil)
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/random.h:111:19: note: previous definition of 'get_random_u32_inclusive' with type 'u32(u32,  u32)' {aka 'unsigned int(unsigned int,  unsigned int)'}
  111 | static inline u32 get_random_u32_inclusive(u32 floor, u32 ceil)
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:250: recipe for target '/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/compat/main.o' failed
make[8]: *** [/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/compat/main.o] Error 1
scripts/Makefile.build:500: recipe for target '/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/compat' failed
make[7]: *** [/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/compat] Error 2
Makefile:2012: recipe for target '/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5' failed
make[6]: *** [/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5] Error 2
Makefile.build:13: recipe for target 'modules' failed
make[5]: *** [modules] Error 2
Makefile.real:93: recipe for target 'modules' failed
make[4]: *** [modules] Error 2
Makefile:120: recipe for target 'modules' failed
make[3]: *** [modules] Error 2
make[3]: Leaving directory '/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5'
Makefile:395: recipe for target '/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/.built' failed
make[2]: *** [/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/backports-6.5/.built] Error 2
make[2]: Leaving directory '/home/lin/ax6-6.1/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#4.09#2.31#5.66
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
package/Makefile:118: recipe for target 'package/kernel/mac80211/compile' failed
make[1]: *** [package/kernel/mac80211/compile] Error 1
make[1]: Leaving directory '/home/lin/ax6-6.1'
/home/lin/ax6-6.1/include/toplevel.mk:230: recipe for target 'package/mac80211/compile' failed
make: *** [package/mac80211/compile] Error 2
````

### OpenWrt version

commit：f861292abcb77655a9792ead47d397d31f1c356e

### OpenWrt target/subtarget

mac80211

### Device

AX6

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

zxlhhyccc, 2023-09-11:
> first issue is fixed for me by changing `LINUX_VERSION_IS_LESS(6,2,0)` to `LINUX_VERSION_IS_LESS(6,1,4)` in line 26 of backport-include/net/random.h. The mentioned function declarations are included in upstream kernel since 6.1.4

Modified to 6.1.4, the second problem is ath11 build failure!

zxlhhyccc, 2023-09-11:
patch：
`````
--- a/backport-include/linux/random.h
+++ b/backport-include/linux/random.h
@@ -23,7 +23,7 @@ static inline u16 get_random_u16(void)
 }
 #endif
 
-#if LINUX_VERSION_IS_LESS(6,2,0)
+#if LINUX_VERSION_IS_LESS(6,1,4)
 static inline u32 __get_random_u32_below(u32 ceil)
 {
 	/*
--- a/drivers/net/wireless/ath/ath11k/ahb.c
+++ b/drivers/net/wireless/ath/ath11k/ahb.c
@@ -1035,7 +1035,7 @@ static int ath11k_ahb_fw_resources_init(struct ath11k_base *ab)
 
 	ret = iommu_map(iommu_dom, ab_ahb->fw.msa_paddr,
 			ab_ahb->fw.msa_paddr, ab_ahb->fw.msa_size,
-			IOMMU_READ | IOMMU_WRITE, GFP_KERNEL);
+			IOMMU_READ | IOMMU_WRITE);
 	if (ret) {
 		ath11k_err(ab, "failed to map firmware region: %d\n", ret);
 		goto err_iommu_detach;
@@ -1043,7 +1043,7 @@ static int ath11k_ahb_fw_resources_init(struct ath11k_base *ab)
 
 	ret = iommu_map(iommu_dom, ab_ahb->fw.ce_paddr,
 			ab_ahb->fw.ce_paddr, ab_ahb->fw.ce_size,
-			IOMMU_READ | IOMMU_WRITE, GFP_KERNEL);
+			IOMMU_READ | IOMMU_WRITE);
 	if (ret) {
 		ath11k_err(ab, "failed to map firmware CE region: %d\n", ret);
 		goto err_iommu_unmap;
--- a/net/mac80211/drop.h
+++ b/net/mac80211/drop.h
@@ -26,18 +26,18 @@ typedef unsigned int __bitwise ieee80211_rx_result;
 /* having two enums allows for checking ieee80211_rx_result use with sparse */
 enum ___mac80211_drop_reason {
 /* if we get to the end of handlers with RX_CONTINUE this will be the reason */
-	___RX_CONTINUE	= SKB_CONSUMED,
+	___RX_CONTINUE	= GRO_CONSUMED,
 
 /* this never gets used as an argument to kfree_skb_reason() */
 	___RX_QUEUED	= SKB_NOT_DROPPED_YET,
 
 #define ENUM(x) ___ ## x,
 	___RX_DROP_MONITOR = SKB_DROP_REASON_SUBSYS_MAC80211_MONITOR <<
-		SKB_DROP_REASON_SUBSYS_SHIFT,
+		SKB_DROP_REASON_SUBSYS_NUM,
 	MAC80211_DROP_REASONS_MONITOR(ENUM)
 
 	___RX_DROP_UNUSABLE = SKB_DROP_REASON_SUBSYS_MAC80211_UNUSABLE <<
-		SKB_DROP_REASON_SUBSYS_SHIFT,
+		SKB_DROP_REASON_SUBSYS_NUM,
 	MAC80211_DROP_REASONS_UNUSABLE(ENUM)
 #undef ENUM
 };
--- a/net/mac80211/rx.c
+++ b/net/mac80211/rx.c
@@ -4074,7 +4074,7 @@ static void ieee80211_rx_handlers_result(struct ieee80211_rx_data *rx,
 			rx->link_sta->rx_stats.dropped++;
 	}
 
-	if (u32_get_bits((__force u32)res, SKB_DROP_REASON_SUBSYS_MASK) ==
+	if (u32_get_bits((__force u32)res, SKB_DROP_REASON_SUBSYS_NUM) ==
 			SKB_DROP_REASON_SUBSYS_MAC80211_UNUSABLE) {
 		kfree_skb_reason(rx->skb, (__force u32)res);
 		return;
``````

nbd168, 2023-09-11:
Pushed a fix to master -  should work now.

CharlesMengCA, 2023-09-11:
Package kmod-iwlwifi is missing dependencies for the following libraries:
ptp.ko

Same error as here: https://github.com/openwrt/openwrt/actions/runs/6149399713/job/16685246472

nbd168, 2023-09-12:
@CharlesMengCA fixed now

CharlesMengCA, 2023-09-12:
Thanks. confirmed working

closed 2023-09-12 by nbd168