openwrt/openwrt #13437: MT7688 (MediaTek LinkIt Smart 7688) reboot fails on OpenWRT 22.03.5

biboc, 2023-09-12:
### Describe the bug

Hi,
On OpenWRT 22.03.5,  MT7688 (MediaTek LinkIt Smart 7688) reboot command from ssh (and serial) fails
(It works on OpenWRT 19)

When reboot command is run:
```
[  398.389276] br-lan: port 1(eth0) entered disabled state
[  398.417446] device eth0 left promiscuous mode
[  398.426170] br-lan: port 1(eth0) entered disabled state
[  402.306830] reboot: Restarting system
```

Ethernet led stops to blink and nothing after

```
root@OpenWrt:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.5'
DISTRIB_REVISION='r20134-5f15225c1e'
DISTRIB_TARGET='ramips/mt76x8'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.5 r20134-5f15225c1e'
DISTRIB_TAINTS='no-all'
```

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ramips/mt76x8

### Device

MediaTek LinkIt Smart 7688

### Image kind

Self-built image

### Steps to reproduce

```
git clone https://github.com/openwrt/openwrt && cd openwrt
git checkout tags/v22.03.5 -b v22.03.5
make menuconfig
# Select target (MediaTek LinkIt Smart 7688)
# Change password
ssh [BOARD]
reboot
```

### Actual behaviour

```
root@OpenWrt:~# reboot
[  398.389276] br-lan: port 1(eth0) entered disabled state
[  398.417446] device eth0 left promiscuous mode
[  398.426170] br-lan: port 1(eth0) entered disabled state
[  402.306830] reboot: Restarting system
```

Ethernet led stops to blink and nothing after, board does not reboot

### Expected behaviour

board reboots

### Diffconfig

```text
$ ./scripts/diffconfig.sh
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt76x8=y
CONFIG_TARGET_ramips_mt76x8_DEVICE_mediatek_linkit-smart-7688=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

DragonBluep, 2023-09-12:
Adding `broken-flash-reset;` property to the `flash@0` node can solve this issue.
https://github.com/openwrt/openwrt/blob/main/target/linux/ramips/dts/mt7628an_mediatek_linkit-smart-7688.dts

biboc, 2023-09-12:
@DragonBluep 
Like this?
```
flash@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <40000000>;
                broken-flash-reset;
```

biboc, 2023-09-12:
Awesome it works
I see this line in some other boards
Shouldn't we include it in all mt7xxx boards dts?

DragonBluep, 2023-09-12:
> Awesome it works I see this line in some other boards Shouldn't we include it in all mt7xxx boards dts?

No, only 16+ MiB size **SPI Flash** needs this fix. If $addr + $size > 0x1000000 in the last partition, then we need it.
For example, this one: 0x50000 + 0x1fb0000 > 0x1000000
```
			partition@50000 {
				compatible = "denx,uimage";
				label = "firmware";
				reg = <0x50000 0x1fb0000>;
			};
```

biboc, 2023-09-12:
Ok so I have the problem because I use a Linkit with 32MB, alright!
Thanks
I close the issue then

closed 2023-09-12 by biboc