openwrt/openwrt #13438: mac80211: v6.5 breaks mesh networking

Codemonkey1973, 2023-09-12:
### Describe the bug

Building with f861292abcb77655a9792ead47d397d31f1c356e or anything up to and including 09d13cd8d87cc50fde67bbe81c6cca4b799b2724 results in mesh network not working.

Building with 69f12c2f23c6d32a5fdda2385b4156c9b8bfae22 works fine.

I have the dev board I'm using configured with 2 led's showing 2.4GHz wifi tx and rx. Plenty of tx led blinking but no rx.

### OpenWrt version

r23908-09d13cd8d8

### OpenWrt target/subtarget

ipq40xx/generic

### Device

8devices Habanero DVK

### Image kind

Self-built image

### Steps to reproduce

Boot device

### Actual behaviour

Device boots but no other mesh nodes are reachable.

### Expected behaviour

Other mesh nodes should be reachable.

### Diffconfig

```text
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_ipq40xx_generic=y
CONFIG_TARGET_ipq40xx_generic_DEVICE_8dev_habanero-dvk=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BUILD_PATENTED=y
# CONFIG_BUSYBOX_CONFIG_ASH_RANDOM_SUPPORT is not set
# CONFIG_BUSYBOX_CONFIG_NTPD is not set
CONFIG_BUSYBOX_CONFIG_RENICE=y
# CONFIG_BUSYBOX_CONFIG_RESIZE is not set
CONFIG_BUSYBOX_CONFIG_STTY=y
CONFIG_BUSYBOX_CONFIG_SU=y
# CONFIG_BUSYBOX_CONFIG_TASKSET is not set
CONFIG_BUSYBOX_CONFIG_WATCH=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_KERNEL_ENABLE_DEFAULT_TRACERS=y
CONFIG_KERNEL_FTRACE=y
CONFIG_KERNEL_IO_URING=y
# CONFIG_LUA_ECO_MBEDTLS is not set
CONFIG_LUA_ECO_OPENSSL=y
CONFIG_LUCI_SRCDIET=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_PACKAGE_ATH_DEBUG=y
CONFIG_PACKAGE_alsa-lib=y
CONFIG_PACKAGE_arp-scan=y
# CONFIG_PACKAGE_ath10k-firmware-qca4019-ct is not set
CONFIG_PACKAGE_ath10k-firmware-qca4019-ct-htt=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_fdk-aac=y
CONFIG_PACKAGE_ffmpeg=y
CONFIG_PACKAGE_firewall=y
# CONFIG_PACKAGE_firewall4 is not set
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_iperf3=y
CONFIG_PACKAGE_irqbalance=y
# CONFIG_PACKAGE_iw is not set
CONFIG_PACKAGE_iw-full=y
# CONFIG_PACKAGE_jansson is not set
# CONFIG_PACKAGE_kmod-ath10k-ct is not set
CONFIG_PACKAGE_kmod-ath10k-ct-smallbuffers=y
CONFIG_PACKAGE_kmod-br-netfilter=y
CONFIG_PACKAGE_kmod-crypto-ecb=y
CONFIG_PACKAGE_kmod-crypto-ecdh=y
CONFIG_PACKAGE_kmod-crypto-kpp=y
CONFIG_PACKAGE_kmod-crypto-sha256=y
CONFIG_PACKAGE_kmod-hid=y
CONFIG_PACKAGE_kmod-input-core=y
CONFIG_PACKAGE_kmod-input-evdev=y
CONFIG_PACKAGE_kmod-ip6tables=y
CONFIG_PACKAGE_kmod-ipt-conntrack=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-nat=y
CONFIG_PACKAGE_kmod-ipt-offload=y
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_kmod-mii=y
CONFIG_PACKAGE_kmod-netlink-diag=y
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nf-ipt6=y
# CONFIG_PACKAGE_kmod-nfnetlink is not set
# CONFIG_PACKAGE_kmod-nft-core is not set
# CONFIG_PACKAGE_kmod-nft-fib is not set
# CONFIG_PACKAGE_kmod-nft-nat is not set
# CONFIG_PACKAGE_kmod-nft-offload is not set
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_kmod-sound-core=y
CONFIG_PACKAGE_kmod-usb-acm=y
CONFIG_PACKAGE_kmod-usb-net=y
CONFIG_PACKAGE_lame-lib=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libbz2=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libffmpeg-full=y
CONFIG_PACKAGE_libgmp=y
CONFIG_PACKAGE_libgnutls=y
CONFIG_PACKAGE_libip4tc=y
CONFIG_PACKAGE_libip6tc=y
CONFIG_PACKAGE_libiperf3=y
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
# CONFIG_PACKAGE_libmnl is not set
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnettle=y
# CONFIG_PACKAGE_libnftnl is not set
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libopenssl-conf=y
CONFIG_PACKAGE_libopenssl-legacy=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libpcre=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libuuid=y
CONFIG_PACKAGE_libxml2=y
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci-app-commands=y
CONFIG_PACKAGE_luci-app-diag-core=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-nginx=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_luci-theme-openwrt-2020=y
CONFIG_PACKAGE_mediamtx=y
CONFIG_PACKAGE_mesh11sd=y
CONFIG_PACKAGE_netperf=y
# CONFIG_PACKAGE_nftables-json is not set
CONFIG_PACKAGE_nginx=y
CONFIG_PACKAGE_nginx-full=y
CONFIG_PACKAGE_nginx-mod-luci=y
CONFIG_PACKAGE_nginx-mod-luci-ssl=y
CONFIG_PACKAGE_nginx-mod-ubus=y
CONFIG_PACKAGE_nginx-ssl-util=y
CONFIG_PACKAGE_nginx-util=y
# CONFIG_PACKAGE_ppp is not set
# CONFIG_PACKAGE_procd-ujail is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ss=y
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_tedd-mesh=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uwsgi=y
CONFIG_PACKAGE_uwsgi-cgi-plugin=y
CONFIG_PACKAGE_uwsgi-luci-support=y
CONFIG_PACKAGE_uwsgi-syslog-plugin=y
CONFIG_PACKAGE_wireless-tools=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mesh-openssl=y
CONFIG_PACKAGE_zlib=y
# CONFIG_WOLFSSL_HAS_ECC25519 is not set
# CONFIG_WOLFSSL_HAS_OPENVPN is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
CONFIG_PACKAGE_kmod-crypto-crc32c=y
CONFIG_PACKAGE_kmod-lib-crc-ccitt=y
CONFIG_PACKAGE_kmod-lib-crc32c=y
# CONFIG_PCRE_JIT_ENABLED is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Codemonkey1973, 2023-09-12:
logread
```
Mon Sep 11 19:42:20 2023 kern.info kernel: [    2.827523] DSA: tree 0 setup
Mon Sep 11 19:42:20 2023 kern.info kernel: [    2.845270] VFS: Mounted root (squashfs filesystem) readonly on device 31:11.
Mon Sep 11 19:42:20 2023 kern.info kernel: [    2.847338] Freeing unused kernel image (initmem) memory: 1024K
Mon Sep 11 19:42:20 2023 kern.info kernel: [    2.869281] Run /sbin/init as init process
Mon Sep 11 19:42:20 2023 kern.debug kernel: [    2.869324]   with arguments:
Mon Sep 11 19:42:20 2023 kern.debug kernel: [    2.869332]     /sbin/init
Mon Sep 11 19:42:20 2023 kern.debug kernel: [    2.869341]   with environment:
Mon Sep 11 19:42:20 2023 kern.debug kernel: [    2.869347]     HOME=/
Mon Sep 11 19:42:20 2023 kern.debug kernel: [    2.869355]     TERM=linux
Mon Sep 11 19:42:20 2023 user.info kernel: [    4.020401] init: Console is alive
Mon Sep 11 19:42:20 2023 user.info kernel: [    4.020926] init: - watchdog -
Mon Sep 11 19:42:20 2023 user.info kernel: [    6.050647] kmodloader: loading kernel modules from /etc/modules-boot.d/*
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.340921] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.341021] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 1
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.345591] xhci-hcd xhci-hcd.0.auto: hcc params 0x0228f665 hci version 0x100 quirks 0x0000000002010010
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.353043] xhci-hcd xhci-hcd.0.auto: irq 108, io mem 0x08a00000
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.362538] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.368527] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 2
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.373849] xhci-hcd xhci-hcd.0.auto: Host supports USB 3.0 SuperSpeed
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.387203] hub 1-0:1.0: USB hub found
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.388000] hub 1-0:1.0: 1 port detected
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.392220] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.396364] hub 2-0:1.0: USB hub found
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.403921] hub 2-0:1.0: 1 port detected
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.407977] xhci-hcd xhci-hcd.1.auto: xHCI Host Controller
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.411537] xhci-hcd xhci-hcd.1.auto: new USB bus registered, assigned bus number 3
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.417019] xhci-hcd xhci-hcd.1.auto: hcc params 0x0220f665 hci version 0x100 quirks 0x0000000002010010
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.424454] xhci-hcd xhci-hcd.1.auto: irq 109, io mem 0x06000000
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.433934] xhci-hcd xhci-hcd.1.auto: xHCI Host Controller
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.439991] xhci-hcd xhci-hcd.1.auto: new USB bus registered, assigned bus number 4
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.445258] xhci-hcd xhci-hcd.1.auto: Host supports USB 3.0 SuperSpeed
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.454581] hub 3-0:1.0: USB hub found
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.459474] hub 3-0:1.0: 1 port detected
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.463612] usb usb4: We don't know the algorithms for LPM for this host, disabling LPM.
Mon Sep 11 19:42:20 2023 kern.info kernel: [    6.467784] hub 4-0:1.0: USB hub found
Mon Sep 11 19:42:20 2023 kern.err kernel: [    6.475341] hub 4-0:1.0: config failed, hub doesn't have any ports! (err -19)
Mon Sep 11 19:42:20 2023 user.info kernel: [    6.481600] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
Mon Sep 11 19:42:20 2023 user.info kernel: [    6.495721] init: - preinit -
Mon Sep 11 19:42:20 2023 kern.notice kernel: [    8.767301] random: jshn: uninitialized urandom read (4 bytes read)
Mon Sep 11 19:42:20 2023 kern.notice kernel: [    9.038588] random: jshn: uninitialized urandom read (4 bytes read)
Mon Sep 11 19:42:20 2023 kern.notice kernel: [    9.109345] random: jshn: uninitialized urandom read (4 bytes read)
Mon Sep 11 19:42:20 2023 kern.info kernel: [    9.392874] ipqess-edma c080000.ethernet eth0: configuring for fixed/internal link mode
Mon Sep 11 19:42:20 2023 kern.info kernel: [    9.393772] qca8k-ipq4019 c000000.switch lan1: configuring for phy/psgmii link mode
Mon Sep 11 19:42:20 2023 kern.info kernel: [    9.399880] ipqess-edma c080000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Mon Sep 11 19:42:20 2023 kern.info kernel: [   10.547082] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Mon Sep 11 19:42:20 2023 kern.notice kernel: [   10.570825] random: procd: uninitialized urandom read (4 bytes read)
Mon Sep 11 19:42:20 2023 kern.notice kernel: [   13.133448] jffs2: notice: (250) jffs2_build_xattr_subsystem: complete building xattr subsystem, 20 of xdatum (11 unchecked, 4 orphan) and 23 of xref (4 dead, 0 orphan) found.
Mon Sep 11 19:42:20 2023 user.info kernel: [   13.139496] mount_root: switching to jffs2 overlay
Mon Sep 11 19:42:20 2023 kern.warn kernel: [   13.166347] overlayfs: upper fs does not support tmpfile.
Mon Sep 11 19:42:20 2023 kern.notice kernel: [   13.168940] random: crng init done
[... cut ...]
Mon Sep 11 19:43:32 2023 daemon.notice wpa_supplicant[1012]: m-11s-0: MESH-SAE-AUTH-FAILURE addr=c4:93:00:40:ba:83
Mon Sep 11 19:43:34 2023 daemon.notice wpa_supplicant[1012]: m-11s-0: MESH-SAE-AUTH-FAILURE addr=c4:93:00:40:c6:73
Mon Sep 11 19:43:34 2023 daemon.notice hostapd: handle_probe_req: send failed
Mon Sep 11 19:43:34 2023 daemon.notice hostapd: handle_probe_req: send failed
Mon Sep 11 19:43:35 2023 daemon.notice hostapd: handle_probe_req: send failed
Mon Sep 11 19:43:35 2023 daemon.notice hostapd: handle_probe_req: send failed
```

Codemonkey1973, 2023-09-12:
cat /etc/config/wireless

```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '1'
        option path 'platform/soc/a000000.wifi'
        option country 'GB'
        option htmode 'HT20'
        option hwmode '11g'
        option cell_density '2'
        option txpower '20'
        option short_preamble '0'

config wifi-device 'radio1'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'platform/soc/a800000.wifi'
        option htmode 'VHT80'
        option disabled '1'

config wifi-iface 'wifinet1'
        option device 'radio0'
        option mode 'mesh'
        option encryption 'sae'
        option mesh_id 'tedd-mesh'
        option mesh_fwding '1'
        option key 'XXXXXXX'
        option network 'wan'
        option short_preamble '0'
        option ifname 'm-11s-0'
        option mesh_rssi_threshold '-80'

config wifi-iface 'wifinet2'
        option device 'radio0'
        option mode 'ap'
        option key 'XXXXXXX'
        option network 'lan'
        option ssid 'tedd-mesh-lan-ap'
        option encryption 'psk2'
        option disabled '1'

config wifi-iface 'wifinet3'
        option device 'radio0'
        option mode 'ap'
        option ssid 'tedd-mesh-wan-ap'
        option key 'XXXXXXX'
        option ieee80211r '1'
        option ft_psk_generate_local '1'
        option network 'wan'
        option ft_over_ds '0'
        option encryption 'psk2'
        option mobility_domain '5445'

```

cat /etc/config/network

```
config interface 'loopback'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'
        option device 'lo'

config globals 'globals'
        option ula_prefix 'fd8b:59e7:0201::/48'
        option packet_steering '1'

config interface 'lan'
        option proto 'static'
        option ipaddr '192.168.40.254'
        option netmask '255.255.255.0'
        option device 'br-lan'

config interface 'wan'
        option proto 'static'
        option device 'br-wan'
        option ipaddr '10.26.17.186'
        option netmask '255.0.0.0'
        option ip6ifaceid 'eui64'
        option ip6assign '64'

config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option ports '1 2 3 4 0'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth0'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        option ipv6 '0'
        option bridge_empty '1'

config device
        option name 'br-wan'
        option type 'bridge'
        list ports 'eth0'
        list ports 'wan'
        option bridge_empty '1'

config device
        option name 'eth0'

config device
        option name 'wan'

config device
        option name 'phy0-ap0'

```

raenye, 2023-09-12:
@Codemonkey1973 Not related to the mesh bug, but why do you have `eth0` in both `br-lan` and `br-wan`?

Codemonkey1973, 2023-09-12:
> @Codemonkey1973 Not related to the mesh bug, but why do you have `eth0` in both `br-lan` and `br-wan`?

well spotted! Looks like a cut & paste error from some recent tinkering. Shouldn't be in br-wan. I'll remove it. Currently building with the latest commits so will retry soon.

nbd168, 2023-09-12:
The AP+mesh issue that broke the AP (log message: handle_probe_req: send failed) is fixed in 499ca4cbe0ca2ee126c18a294f11d4c07a1699b4

The mesh connection issue is fixed in 20bd3502d3a841457cc06b3aa45646258fdcba9e

Latest master should work again, please test

closed 2023-09-13 by Codemonkey1973