openwrt/openwrt #13439: mac80211 update v6.5  breaks WIFI on RPi 4b

leobsky, 2023-09-13:
### Describe the bug

RPi 4b failed to load the wifi firmware with backpots v6.5

### OpenWrt version

r23908-09d13cd8d8

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.1

### Image kind

Official downloaded image

### Actual behaviour

[    5.228092] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin failed with error -2
[    5.239801] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin
[    5.252391] kmodloader: done loading kernel modules from /etc/modules.d/*
[    5.428785] brcmfmac: brcmf_fwvid_request_module: mod=wcc: failed 65280
[    5.435424] ieee80211 phy0: brcmf_attach: brcmf_fwvid_attach failed
[    5.498735] brcmfmac: brcmf_sdio_firmware_callback: brcmf_attach failed

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-13:
Errno -2 means file not found.
Do you have brcmfmac-firmware-43455-sdio-rpi-4b package installed?

nbd168, 2023-09-13:
Should be fixed in 21485791161a5b158b114659672b513059e7a035, please test

Entropy512, 2023-09-13:
That fixed it for me.

nbd168, 2023-09-13:
What outstanding issue?

closed 2023-09-13 by nbd168

Entropy512, 2023-09-13:
https://github.com/openwrt/openwrt/issues/12634

It is getting more and more difficult to maintain reverts of hostapd back to the last known working version.

Per https://github.com/openwrt/openwrt/issues/13156#issuecomment-1653484950 - bisecting hostapd is a real nightmare due to the massive pile of local patches

I'm GUESSING that the reason I haven't seen anyone else report it is:
1)  People using client mode is rarer than AP mode
2)  People using client mode on an AP that broadcasts an SSID on both bands simultaneously is even rarer

I'm suspecting item 2 combined with all of the upstream MLO/MLD work is the culprit - the dual-band specific behaviors in 13156 made a lightbulb pop in my head, I'll be taking this thing home and seeing if the problem goes away on my home network (separate SSIDs for 2.4 GHz and 5 GHz).  I'm trying to find how something in userspace could cause the driver to stop reporting proper link status but have failed so far.

nbd168, 2023-09-13:
@Entropy512, please compile both working and non-working hostapd with CONFIG_WPA_MSG_MIN_PRIORITY=0 in your .config, and edit /etc/init.d/wpad to add -dd to the wpa_supplicant command line, and give me a full log for both.

I think the most likely reason why this hasn't been reported much is the fact that the issue is probably completely specific to brcmfmac + client mode, since (IIRC) that combination offloads the WPA handshake to the firmware.
On other platforms, the handshake is performed in software by wpa_supplicant, and it does not wait for this particular notification. Looking at the code, I couldn't find any obvious reason for the regression, so I'm hoping that more detailed logs might uncover more clues.

Entropy512, 2023-09-13:
OK, thanks for the hint on adding -dd - that's the missing piece of getting better logging.  :)

I'm going to see if there's any difference with an AP that only has a single band per SSID tonight, and then try to get your logs tomorrow or Friday.