openwrt/openwrt #13441: 23.05.0-rc3 MT6721 switch issues on Ubiquiti EdgeRouter-X

simonvetter, 2023-09-13:
### Describe the bug

My EdgeRouter-X freshly updated to 23.05.0-rc3 (from 22.03.5 stable) floods unicast frames destined for the router itself out on all ports belonging to the same VLAN, resulting in high network load.

Relevant network configuration:
```
config device
	option name 'br0'
	option type 'bridge'
	option vlan_filtering '1'
	option igmp_snooping '0'
	option stp '0'
	list ports 'eth1'
	list ports 'eth2'
	list ports 'eth3'
	list ports 'eth4'

config interface 'vlan2'
        option device 'br0.2'
	option force_link '1'
        option macaddr 'd8:7e:19:48:fc:02'
        option proto 'static'
        option ip6assign '64'
        option ip6hint '02'

config bridge-vlan
        option device 'br0'
        option vlan '2'
        list ports 'eth1:t'
        list ports 'eth2'
        list ports 'eth4:t'
```

which translates in the following:
```bash
2: dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP group default qlen 1000
    link/ether f0:9f:c2:44:88:dd brd ff:ff:ff:ff:ff:ff
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether f0:9f:c2:44:88:dd brd ff:ff:ff:ff:ff:ff
4: eth1@dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br0 state UP group default qlen 1000
    link/ether f0:9f:c2:44:88:de brd ff:ff:ff:ff:ff:ff
5: eth2@dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br0 state UP group default qlen 1000
    link/ether f0:9f:c2:44:88:df brd ff:ff:ff:ff:ff:ff
6: eth3@dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br0 state UP group default qlen 1000
    link/ether f0:9f:c2:44:88:e0 brd ff:ff:ff:ff:ff:ff
7: eth4@dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br0 state UP group default qlen 1000
    link/ether f0:9f:c2:44:88:e1 brd ff:ff:ff:ff:ff:ff
8: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether f0:9f:c2:61:72:de brd ff:ff:ff:ff:ff:ff
9: br0.2@br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether d8:7e:19:48:fc:02 brd ff:ff:ff:ff:ff:ff
```
Incoming traffic going to d8:7e:19:48:fc:02 (br0.2, a VLAN interface created on top of the bridge) is flooded on all ports of the same VLAN, tagged or not e.g. a frame sent on LAN2 (vlan 2, untagged) can be seen on LAN1 and LAN4 (both trunked).

The bridge MAC address table doesn't seem to ever learn that address, even though it is the most active (since it's a router port forwarding traffic).
```bash
# brctl showmacs br0
port no	mac addr		is local?	ageing timer
  2	00:1e:06:c3:7e:b8	no		   0.00
  3	00:e0:4c:68:06:66	no		   0.00
  1	02:01:0b:66:93:ed	no		   0.02
  1	02:01:0b:66:93:ed	no		  64.29
  4	02:05:81:34:ff:f8	no		 289.08
  4	78:d7:71:9e:cc:9a	no		  26.89
  4	ba:ed:ed:fc:e3:27	no		 100.73
  4	d2:fa:13:a0:7c:54	no		  20.65
  1	f0:9f:c2:61:72:de	yes		   0.00
  2	f0:9f:c2:61:72:df	yes		   0.00
  2	f0:9f:c2:61:72:df	yes		   0.00
  3	f0:9f:c2:61:72:e0	yes		   0.00
  3	f0:9f:c2:61:72:e0	yes		   0.00
  4	f0:9f:c2:61:72:e1	yes		   0.00
  4	f0:9f:c2:61:72:e1	yes		   0.00
  4	f0:9f:c2:61:72:e1	yes		   0.00
```

The MAC addresses of the physical member ports can be seen (as static entries), but note how d8:7e:19:48:fc:02 (br0.2) isn't present. Having the CPU port not learning MAC addresses would probably be fine if VLAN MAC addresses were added ad permanent entries, but that doesn't seem to happen.

This is having great impact on network performance, since:
1) traffic is sent on a lot more ports than necessary, filling Ethernet links only to have the frames dropped at the MAC address filter of the device network interface,
2) bandwidth going to the CPU port (br0.2) is limited to that of the slowest Ethernet link e.g. a VLAN with two devices, one negotiating at 1 Gb/s and the other at 100Mb/s, the 1Gb/s device won't be able to push more than 100Mb/s to the switch.

Note that d8:7e:19:48:fc:02 is always sending traffic out on that VLAN, so it's not just a case of a MAC entry aging out (even more so since it probably should be a static entry).

I don't remember seeing this behavior on 22.03.5.

### OpenWrt version

r23389-5deed175a5

### OpenWrt target/subtarget

ramips/mt7621

### Device

Ubiquiti EdgeRouter-X

### Image kind

Official downloaded image

### Steps to reproduce

1. Use the network config snippet above (or equivalent) to place multiple ports into a VLAN
2. Connect a device (device A) to one of the ports, send traffic to the CPU port in that VLAN (e.g. using ping)
3. Connect a second device (device B) to another port assigned to this VLAN and run a packet capture (in promiscuous mode)
4. Notice how traffic sent from device A to the CPU port's MAC address is received by device B.

### Actual behaviour

Ethernet frames sent to a VLAN interface created on top of a bridge are flooded on all interfaces belonging to this VLAN.

### Expected behaviour

Ethernet frames sent to a VLAN interfaces created on top of a bridge should only be delivered to that VLAN interface.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

simonvetter, 2023-09-25:
FYI I've managed to work around this issue by manually updating the bridge's forwarding information base.

As it turns out, the MAC addresses in question are present in the FIB but lack many entries :

Before altering anything:
```
# bridge fdb list | grep d8:7e:19:48:fc:02
d8:7e:19:48:fc:02 dev br0 self permanent
```

Here, I'm updating the existing entry with the *same exact parameters*:
```
# bridge fdb replace d8:7e:19:48:fc:02 dev br0 self permanent
```

After which multiple entries do show up, some marked as VLAN-specific and most importantly with the "offload" attribute, which according to the bridge man page means that the entry is present in the MT6721's hardware FIB.
```
# bridge fdb list | grep d8:7e:19:48:fc:02
d8:7e:19:48:fc:02 dev dsa self permanent
d8:7e:19:48:fc:02 dev br0 self permanent
d8:7e:19:48:fc:02 dev br0 vlan 8 offload master br0 permanent
d8:7e:19:48:fc:02 dev br0 vlan 3 offload master br0 permanent
d8:7e:19:48:fc:02 dev br0 vlan 2 offload master br0 permanent
d8:7e:19:48:fc:02 dev br0 offload master br0 permanent
```

Note that the bridge command is part of the ip-bridge package (opkg install ip-bridge).

I've made that workaround persistent by adding the following to /etc/rc.local:
```
for macaddr in d8:7e:19:48:fc:02 d8:7e:19:48:fc:03 d8:7e:19:48:fc:08
do
	/usr/sbin/bridge fdb replace ${macaddr} dev br0 self permanent || logger -t bootup.sh "failed to replace ${macaddr} in bridge fdb"
done
```
which while doing the trick is a kludge I'd rather not have to do.

I'm not familiar enough with OpenWRT internals to tell if that looks like a kernel/DSA or netifd bug so I don't really know where to go from here... if anyone could give me some pointers I'd appreciate it.

In the meantime I'll try and take a look at what netifd does when configuring VLAN interfaces on top of a bridge, see if I can find anything.

simonvetter, 2023-10-03:
Just updated to 23.05.0-rc4 where the bug is still present. The workaround listed above still works.

henielssen, 2023-10-06:
Not same issue, but [(https://github.com/openwrt/openwrt/issues/13613)] indicates that you perhaps can find a solution for EdgeRouter X by looking at the EdgeRouter X SFP 23.05.0-rc4 distribution.

simonvetter, 2023-10-06:
@henielssen yeah that's a different thing for sure. I just replied on your issue to explain why.

andriej, 2023-10-13:
Is it still present in stable `23.05` released?

simonvetter, 2023-10-20:
> Is it still present in stable 23.05 released?
Yep, just did a first round of upgrades and I can confirm it is still present in the final release.

plat2on1, 2023-11-11:
Is there a reason the VLAN interface has a different MAC from its parent? Seems when you create a VLAN interface the MAC is inherited from the parent by default so I haven't seen this issue on my setup.

simonvetter, 2023-11-11:
Two use cases I can think of:- the ISP requires cloning a MAC address on the WAN side of things (if the WAN port is just a VLAN on top of a DSA switch rather than a dedicated Ethernet interface),- old/buggy managed switches or APs get confused and cause weird connectivity issues when a single MAC is present in multiple VLANs

howels, 2026-10-30:
This target is dropping raw ethernet frames between eth0 and other ports even when they are part of the same switch and vlan. Target is broken on 25.12 as traffic is disrupted.