openwrt/openwrt #13444: Hostapd fails to bring up wifi interfaces

Soromeister, 2023-09-13:
### Describe the bug

Getting the below error for all radios:

```
Wed Sep 13 19:01:18 2023 daemon.notice netifd: radio0 (2620): Syntax error: Failed to parse JSON string: unexpected character
Wed Sep 13 19:01:18 2023 daemon.notice netifd: radio0 (2620): In /usr/share/hostap/wdev.uc, line 133, byte 28:
Wed Sep 13 19:01:18 2023 daemon.notice netifd: radio0 (2620):
Wed Sep 13 19:01:18 2023 daemon.notice netifd: radio0 (2620):  `new_config = json(new_config);`
Wed Sep 13 19:01:18 2023 daemon.notice netifd: radio0 (2620):   Near here ------------------^
Wed Sep 13 19:01:18 2023 daemon.notice netifd: radio0 (2620):
Wed Sep 13 19:01:18 2023 daemon.notice netifd: radio0 (2620):
```

This bug was recently introduced in [13c1080](https://github.com/openwrt/openwrt/commit/13c1080a3f6cbe4e8a999e77a7688b2367b8adf9)

### OpenWrt version

r23928-7365e8f1bb

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX9000

### Image kind

Official downloaded image

### Steps to reproduce

Have build r23928-7365e8f1bb
Wi-fi won't come up and system log shows the mentioned error for all radio units.

### Actual behaviour

Hostapd fails to start

### Expected behaviour

Hostapd should start and bring up radios

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

nbd168, 2023-09-13:
Please show me the wifi config that you're using

Soromeister, 2023-09-13:
@nbd168 Here is my config:

```
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'soc/10000000.pci/pci0001:00/0001:00:00.0/0001:01:00.0'
	option channel 'auto'
	option channels '100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140, 144'
	option band '5g'
	option htmode 'VHT80'
	option he_bss_color '44'
	option country 'US'
	option cell_density '3'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/c000000.wifi'
	option channel '161'
	option channels '165, 161, 157, 153, 149'
	option band '5g'
	option htmode 'HE80'
	option he_bss_color '11'
	option country 'US'
	option cell_density '3'
	option txpower '30'

config wifi-device 'radio2'
	option type 'mac80211'
	option path 'platform/soc/c000000.wifi+1'
	option channels '13, 11, 6, 1'
	option channel '11'
	option band '2g'
	option htmode 'HE20'
	option he_bss_color '22'
	option country 'US'
	option cell_density '3'
	option txpower '15'

config wifi-device 'radio3'
	option type 'mac80211'
	option path 'soc/20000000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
	option channels '165, 161, 157, 153, 149, 64, 60, 56, 52, 48, 44, 40, 36'
	option channel '36'
	option band '5g'
	option htmode 'HE160'
	option he_bss_color '33'
	option country 'US'
	option cell_density '3'
	option txpower '30'

config wifi-iface 'guest_radio0'
	option device 'radio0'
	option network 'guest'
	option mode 'ap'
	option ssid 'Guest'
	option encryption 'sae-mixed'
	option isolate '1'
	option ifname 'r0-guest'
	option key 'myguestpass'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'Main'
	option encryption 'sae-mixed'
	option isolate '1'
	option ifname 'r1-main'
	option key 'mypass'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'

config wifi-iface 'default_radio2'
	option device 'radio2'
	option network 'lan'
	option mode 'ap'
	option ssid 'Main'
	option encryption 'sae-mixed'
	option isolate '1'
	option ifname 'r2-main'
	option key 'mypass'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'

config wifi-iface 'default_radio3'
	option device 'radio3'
	option network 'lan'
	option mode 'ap'
	option ssid 'WiFi (G)'
	option encryption 'sae-mixed'
	option ifname 'r3-main'
	option key 'mypass'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'
```

This works on build `r23900-7f54d9ba1a`. Hope this helps.

nbd168, 2023-09-13:
Did you update individual packages, or did you build a fresh image? The only way I could see this occur is if the files from different packages are out of sync, e.g. hostapd-common vs kmod-cfg80211.

Soromeister, 2023-09-13:
I first attempted an upgrade via sysupgrade to `r23928-7365e8f1bb` and noticed that the wi-fi doesn't come up. I plugged via ethernet and checked the logs seeing the error that I included in my first post.

I realized something is broken so I downgraded back to `r23900-7f54d9ba1a` and wanted to upgrade the packages instead (to get the new fw). Rebooted and noticed the same thing so I re-flashed `r23900-7f54d9ba1a` and reported this.

nbd168, 2023-09-13:
Could you please show me a full log from the wifi bring up with the latest version?

Code-TT, 2023-09-13:
The same for me after sysupgrade the latest snapshot on Xiaomi AX3600. 
I'm using these additional modules: 
"wpad-mesh-openssl luci luci-app-attendedsysupgrade nano luci-mod-dashboard mesh11sd luci-ssl"

Here the logfile:

```
root@OpenWrt:~# logread -l 20 -f
Wed Sep 13 15:07:38 2023 daemon.notice netifd: Interface 'wan' is now up
Wed Sep 13 15:07:38 2023 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Sep 13 15:07:38 2023 daemon.info dnsmasq[1]: using nameserver 192.168.31.1#53
Wed Sep 13 15:07:38 2023 daemon.info dnsmasq[1]: using nameserver fe80::2ad1:27ff:fee9:8495%wan#53
Wed Sep 13 15:07:38 2023 daemon.info dnsmasq[1]: using only locally-known addresses for test
Wed Sep 13 15:07:38 2023 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Wed Sep 13 15:07:38 2023 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Wed Sep 13 15:07:38 2023 daemon.info dnsmasq[1]: using only locally-known addresses for local
Wed Sep 13 15:07:38 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Wed Sep 13 15:07:38 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Wed Sep 13 15:07:38 2023 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Wed Sep 13 15:07:39 2023 user.notice firewall: Reloading firewall due to ifup of wan (wan)
Wed Sep 13 15:07:39 2023 daemon.warn odhcpd[2183]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Wed Sep 13 22:18:19 2023 daemon.notice mesh11sd[2933]: mesh11sd v2.0.0 has started: mesh management mode 1
Wed Sep 13 22:18:19 2023 daemon.warn odhcpd[2183]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Wed Sep 13 22:18:23 2023 kern.info kernel: [   33.113523] l11: disabling
Wed Sep 13 22:18:32 2023 daemon.warn odhcpd[2183]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Wed Sep 13 22:18:44 2023 daemon.err uhttpd[1285]: [info] luci: accepted login on / for root from 192.168.1.234
Wed Sep 13 22:19:01 2023 authpriv.info dropbear[4249]: Child connection from 192.168.1.234:54269
Wed Sep 13 22:19:03 2023 authpriv.notice dropbear[4249]: Auth succeeded with blank password for 'root' from 192.168.1.234:54269
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503): WARNING: Variable 'data' does not exist or is not an array/object
Wed Sep 13 22:20:13 2023 daemon.notice hostapd: Set new config for phy phy1:
Wed Sep 13 22:20:13 2023 daemon.notice wpa_supplicant[2003]: Set new config for phy phy1
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503): Syntax error: Failed to parse JSON string: unexpected character
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503): In /usr/share/hostap/wdev.uc, line 133, byte 28:
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503):
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503):  `new_config = json(new_config);`
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503):   Near here ------------------^
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503):
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503):
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503): Syntax error: Failed to parse JSON string: unexpected character
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503): In /usr/share/hostap/wdev.uc, line 133, byte 28:
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503):
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503):  `new_config = json(new_config);`
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503):   Near here ------------------^
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503):
Wed Sep 13 22:20:13 2023 daemon.notice netifd: radio1 (4503):
Wed Sep 13 22:20:14 2023 daemon.notice wpa_supplicant[2003]: Set new config for phy phy1
Wed Sep 13 22:20:14 2023 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Wed Sep 13 22:20:14 2023 daemon.notice hostapd: Restart interface for phy phy1
Wed Sep 13 22:20:14 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 hw_mode=a beacon_int=100 chanlist=36 tx_queue_data2_burst=2.0 mbssid=0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=42 vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][VHT160-80PLUS80][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=1 he_oper_centr_freq_seg0_idx=42 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu_edca_qos_info_txop_request
Wed Sep 13 22:20:14 2023 daemon.err hostapd: Line 77: invalid bssid item
Wed Sep 13 22:20:14 2023 daemon.err hostapd: 1 errors found in configuration file '<inline>'
[... cut ...]
Wed Sep 13 22:20:16 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Wed Sep 13 22:20:16 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Wed Sep 13 22:20:17 2023 daemon.warn odhcpd[2183]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Wed Sep 13 22:30:14 2023 daemon.warn odhcpd[2183]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
```

Soromeister, 2023-09-13:
@nbd168 Attached log files from `r23928-7365e8f1bb`:
[Kernel(dmesg)log.txt](https://github.com/openwrt/openwrt/files/12602348/Kernel.dmesg.log.txt)
[Syslog.txt](https://github.com/openwrt/openwrt/files/12602349/Syslog.txt)

illumiN8i, 2023-09-14:
Same issue here on R7800 r23929-02b4dc93b6

nbd168, 2023-09-14:
I just took a closer look at the log messages. Everything I see clearly indicates that for some reason the upgrade didn't work properly and it's still using wdev.uc from the old version. The error messages indicate that the error happens in this line: `new_config = json(new_config);` and that this is line 133.
This is only true for wdev.uc from before the commit indicated as being at fault.
Could you please sysupgrade to the latest snapshot again and check if any leftovers from the old packages remain in /overlay/upper/usr/share/hostap?
By the way, I did download and unpack the snapshot binary for AX9000 and the file looks fine in the rootfs, so it can't be an issue on the buildbot side.
My guess is that either stale files were preserved on your device for some reason, or sysupgrade is failing in other weird ways on your device. Either way, I don't see any fault in the commit pointed out here.

nbd168, 2023-09-14:
Not sure. I'm not very familiar with it, but it could possibly be related if its package lists are somehow out of sync

Soromeister, 2023-09-16:
I just installed `r23939-a33f1d3515` and seems to have been fixed. It's rather strange that such issues can happen. ASU uses `https://sysupgrade.openwrt.org` to build the images.

closed 2023-09-16 by Soromeister