openwrt/openwrt #13449: STA+AP does not work with automatic channel

castillofrancodamian, 2023-09-14:
### Describe the bug

STA+AP with automatic channel does not work, even with having the channel width configured at 20 MHz. If I enter the channel manually, it works as expected.

### OpenWrt version

r23930-6cf27094e9

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C60 v1

### Image kind

Self-built image

### Steps to reproduce

1. Add a STA interface to connect to an upstream AP.
2. Add an AP in the same radius.
3. Restart the radio with the "Restart" button on LuCI.
4. Press the "Restart" button again on LuCI.

### Actual behaviour

Logs when adding the STA interface.

```
root@archer_c60:~# logread -l 100 -f
Wed Sep 13 22:02:09 2023 daemon.notice netifd: Network device '5g_sta' link is up
Wed Sep 13 22:02:09 2023 daemon.notice netifd: Interface 'telecentro' has link connectivity
Wed Sep 13 22:02:09 2023 daemon.notice netifd: Interface 'telecentro' is setting up now
Wed Sep 13 22:02:09 2023 daemon.notice wpa_supplicant[1277]: 5g_sta: CTRL-EVENT-CONNECTED - Connection to 22:b8:2b:30:95:d9 completed [id=1 id_str=]
Wed Sep 13 22:02:09 2023 kern.info kernel: [  414.170460] IPv6: ADDRCONF(NETDEV_CHANGE): 5g_sta: link becomes ready
Wed Sep 13 22:02:09 2023 daemon.notice wpa_supplicant[1277]: 5g_sta: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Wed Sep 13 22:02:09 2023 daemon.notice netifd: telecentro (6730): udhcpc: started, v1.36.1
Wed Sep 13 22:02:10 2023 daemon.notice netifd: telecentro (6730): udhcpc: broadcasting discover
Wed Sep 13 22:02:10 2023 daemon.notice netifd: telecentro (6730): udhcpc: broadcasting select for 10.197.186.223, server 10.192.0.1
Wed Sep 13 22:02:10 2023 daemon.notice netifd: telecentro (6730): udhcpc: lease of 10.197.186.223 obtained from 10.192.0.1, lease time 300
Wed Sep 13 22:02:11 2023 daemon.notice netifd: Interface 'telecentro' is now up
Wed Sep 13 22:02:11 2023 kern.info kernel: [  416.557959] br-lan: port 2(2g_lag) entered blocking state
Wed Sep 13 22:02:11 2023 kern.info kernel: [  416.563609] br-lan: port 2(2g_lag) entered disabled state
Wed Sep 13 22:02:11 2023 kern.info kernel: [  416.569449] device 2g_lag entered promiscuous mode
Wed Sep 13 22:02:11 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=AR ieee80211d=1 hw_mode=g supported_rates=60 90 120 180 240 360 480 540 basic_rates=60 120 240 beacon_int=100 chanlist=1 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][DSSS_CCK-40] channel=1  interface=2g_lag bssid=50:c7:bf:75:81:b0 ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa_passphrase=masterofpuppets wpa_psk_file=/var/run/hostapd-2g_lag.psk auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=LAG bridge=br-lan wds_bridge= snoop_iface=br-lan wpa_disable_eapol_key_retries=0 wpa_key_mgmt=WPA-PSK okc=0 disable_pmksa_caching=1 dynamic_vlan=0 vlan_naming=1 vlan_no_bridge=1 vlan_file=/var/run/hostapd-2g_lag.vlan qos_
Wed Sep 13 22:02:11 2023 kern.info kernel: [  416.634135] br-lan: port 2(2g_lag) entered blocking state
Wed Sep 13 22:02:11 2023 kern.info kernel: [  416.639741] br-lan: port 2(2g_lag) entered forwarding state
Wed Sep 13 22:02:11 2023 kern.info kernel: [  416.645802] br-lan: port 2(2g_lag) entered disabled state
Wed Sep 13 22:02:11 2023 daemon.notice hostapd: 2g_lag: interface state UNINITIALIZED->COUNTRY_UPDATE
Wed Sep 13 22:02:12 2023 kern.info kernel: [  417.577757] IPv6: ADDRCONF(NETDEV_CHANGE): 2g_lag: link becomes ready
Wed Sep 13 22:02:12 2023 kern.info kernel: [  417.584718] br-lan: port 2(2g_lag) entered blocking state
Wed Sep 13 22:02:12 2023 kern.info kernel: [  417.590329] br-lan: port 2(2g_lag) entered forwarding state
Wed Sep 13 22:02:12 2023 daemon.notice netifd: Network device '2g_lag' link is up
Wed Sep 13 22:02:12 2023 kern.info kernel: [  417.619582] br-lan: port 3(2g_invitados) entered blocking state
Wed Sep 13 22:02:12 2023 kern.info kernel: [  417.625771] br-lan: port 3(2g_invitados) entered disabled state
Wed Sep 13 22:02:12 2023 kern.info kernel: [  417.632150] device 2g_invitados entered promiscuous mode
Wed Sep 13 22:02:12 2023 kern.info kernel: [  417.641988] br-lan: port 3(2g_invitados) entered blocking state
Wed Sep 13 22:02:12 2023 kern.info kernel: [  417.648194] br-lan: port 3(2g_invitados) entered forwarding state
Wed Sep 13 22:02:12 2023 kern.info kernel: [  417.663428] br-lan: port 3(2g_invitados) entered disabled state
Wed Sep 13 22:02:13 2023 kern.info kernel: [  418.444338] IPv6: ADDRCONF(NETDEV_CHANGE): 2g_invitados: link becomes ready
Wed Sep 13 22:02:13 2023 kern.info kernel: [  418.451714] br-lan: port 3(2g_invitados) entered blocking state
Wed Sep 13 22:02:13 2023 kern.info kernel: [  418.457905] br-lan: port 3(2g_invitados) entered forwarding state
Wed Sep 13 22:02:13 2023 daemon.notice netifd: Network device '2g_invitados' link is up
Wed Sep 13 22:02:13 2023 kern.info kernel: [  418.485519] br-lan: port 4(2g_f) entered blocking state
Wed Sep 13 22:02:13 2023 kern.info kernel: [  418.490973] br-lan: port 4(2g_f) entered disabled state
Wed Sep 13 22:02:13 2023 kern.info kernel: [  418.522697] device 2g_f entered promiscuous mode
Wed Sep 13 22:02:13 2023 kern.info kernel: [  418.528853] br-lan: port 4(2g_f) entered blocking state
[... 610 lines cut ...]
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_terminfo=y
# CONFIG_PACKAGE_uclient-fetch is not set
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wireguard-tools=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mini=y
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
CONFIG_TESTING_KERNEL=y
CONFIG_ZRAM_DEF_COMP_LZ4=y
# CONFIG_ZRAM_DEF_COMP_LZORLE is not set
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-14:
Actually it completes DFS and activating AP the driver restarts?
It should be very flaky by definition like 2min cuts when upstream DFS-es itself.
Try swapping -ct to no-ct driver and firmware (just try other)

nbd168, 2023-09-14:
Please test the latest version, it should work now.

castillofrancodamian, 2023-09-15:
Still not working.

```
root@archer_c60's password: 

BusyBox v1.36.1 (2023-09-15 12:25:27 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r23937-50e16efd41
 -----------------------------------------------------
root@archer_c60:~# logread
Fri Sep 15 14:32:31 2023 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Fri Sep 15 14:32:31 2023 kern.info kernel: [  270.829964] IPv6: ADDRCONF(NETDEV_CHANGE): 2g_invitados: link becomes ready
Fri Sep 15 14:32:31 2023 kern.info kernel: [  270.837372] br-lan: port 3(2g_invitados) entered blocking state
Fri Sep 15 14:32:31 2023 kern.info kernel: [  270.843569] br-lan: port 3(2g_invitados) entered forwarding state
Fri Sep 15 14:32:31 2023 daemon.notice netifd: Network device '2g_invitados' link is up
Fri Sep 15 14:32:31 2023 kern.info kernel: [  270.860525] br-lan: port 4(2g_f) entered blocking state
Fri Sep 15 14:32:31 2023 kern.info kernel: [  270.865984] br-lan: port 4(2g_f) entered disabled state
Fri Sep 15 14:32:31 2023 kern.info kernel: [  270.871732] device 2g_f entered promiscuous mode
Fri Sep 15 14:32:31 2023 kern.info kernel: [  270.899695] br-lan: port 4(2g_f) entered blocking state
Fri Sep 15 14:32:31 2023 kern.info kernel: [  270.905142] br-lan: port 4(2g_f) entered forwarding state
Fri Sep 15 14:32:31 2023 kern.info kernel: [  271.357980] br-lan: port 4(2g_f) entered disabled state
Fri Sep 15 14:32:32 2023 daemon.notice wpa_supplicant[1483]: 5g_sta: SME: Trying to authenticate with 22:b8:2b:30:95:d9 (SSID='TeleCentro Wifi' freq=5785 MHz)
Fri Sep 15 14:32:32 2023 kern.info kernel: [  271.489031] 5g_sta: authenticate with 22:b8:2b:30:95:d9
Fri Sep 15 14:32:32 2023 kern.info kernel: [  271.640099] IPv6: ADDRCONF(NETDEV_CHANGE): 2g_f: link becomes ready
Fri Sep 15 14:32:32 2023 kern.info kernel: [  271.646736] br-lan: port 4(2g_f) entered blocking state
Fri Sep 15 14:32:32 2023 kern.info kernel: [  271.652211] br-lan: port 4(2g_f) entered forwarding state
Fri Sep 15 14:32:32 2023 daemon.notice netifd: Network device '2g_f' link is up
Fri Sep 15 14:32:32 2023 daemon.notice hostapd: 2g_lag: interface state COUNTRY_UPDATE->ENABLED
Fri Sep 15 14:32:32 2023 daemon.notice hostapd: 2g_lag: AP-ENABLED
Fri Sep 15 14:32:32 2023 kern.info kernel: [  271.686808] 5g_sta: send auth to 22:b8:2b:30:95:d9 (try 1/3)
Fri Sep 15 14:32:32 2023 kern.info kernel: [  271.700695] 5g_sta: authenticated
Fri Sep 15 14:32:32 2023 daemon.notice wpa_supplicant[1483]: 5g_sta: Trying to associate with 22:b8:2b:30:95:d9 (SSID='TeleCentro Wifi' freq=5785 MHz)
Fri Sep 15 14:32:32 2023 kern.info kernel: [  271.727921] 5g_sta: associate with 22:b8:2b:30:95:d9 (try 1/3)
Fri Sep 15 14:32:32 2023 kern.info kernel: [  271.747924] 5g_sta: RX AssocResp from 22:b8:2b:30:95:d9 (capab=0x1 status=0 aid=23)
Fri Sep 15 14:32:32 2023 kern.warn kernel: [  271.776547] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
Fri Sep 15 14:32:32 2023 kern.warn kernel: [  271.784014] ath10k_pci 0000:00:00.0: failed to enable peer stats info: -122
Fri Sep 15 14:32:32 2023 kern.info kernel: [  271.791349] 5g_sta: associated
Fri Sep 15 14:32:32 2023 kern.debug kernel: [  271.838765] 5g_sta: Limiting TX power to 30 (30 - 0) dBm as advertised by 22:b8:2b:30:95:d9
Fri Sep 15 14:32:32 2023 daemon.notice wpa_supplicant[1483]: 5g_sta: Associated with 22:b8:2b:30:95:d9
Fri Sep 15 14:32:32 2023 daemon.notice wpa_supplicant[1483]: 5g_sta: CTRL-EVENT-CONNECTED - Connection to 22:b8:2b:30:95:d9 completed [id=0 id_str=]
Fri Sep 15 14:32:32 2023 daemon.notice netifd: Network device '5g_sta' link is up
Fri Sep 15 14:32:32 2023 daemon.notice netifd: Interface 'telecentro' has link connectivity
Fri Sep 15 14:32:32 2023 daemon.notice netifd: Interface 'telecentro' is setting up now
Fri Sep 15 14:32:32 2023 daemon.notice wpa_supplicant[1483]: 5g_sta: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Fri Sep 15 14:32:32 2023 kern.info kernel: [  271.876506] IPv6: ADDRCONF(NETDEV_CHANGE): 5g_sta: link becomes ready
Fri Sep 15 14:32:32 2023 kern.warn kernel: [  271.919574] ath10k_pci 0000:00:00.0: Invalid peer id 0 or peer stats buffer, peer: 00000000  sta: 00000000
Fri Sep 15 14:32:33 2023 daemon.notice netifd: telecentro (5063): udhcpc: started, v1.36.1
Fri Sep 15 14:32:33 2023 daemon.notice netifd: Wireless device 'radio1' is now up
Fri Sep 15 14:32:33 2023 daemon.notice netifd: telecentro (5063): udhcpc: broadcasting discover
Fri Sep 15 14:32:33 2023 daemon.notice netifd: telecentro (5063): udhcpc: broadcasting select for 10.196.124.182, server 10.192.0.1
Fri Sep 15 14:32:34 2023 daemon.notice netifd: telecentro (5063): udhcpc: lease of 10.196.124.182 obtained from 10.192.0.1, lease time 300
Fri Sep 15 14:32:34 2023 user.notice https-dns-proxy: Restarting dnsmasq ✓
Fri Sep 15 14:32:34 2023 daemon.info dnsmasq[1]: started, version 2.89 cachesize 10000
Fri Sep 15 14:32:34 2023 daemon.info dnsmasq[1]: DNS service limited to local subnets
Fri Sep 15 14:32:34 2023 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
[... cut ...]
                0       0       0       0       0       0       0       0               0
root@archer_c60:~# iw 5g_ap info
Interface 5g_ap
        ifindex 31
        wdev 0x7
        addr 50:c7:bf:75:81:af
        ssid AP
        type AP
        wiphy 0
        channel 165 (5825 MHz), width: 20 MHz, center1: 5825 MHz
        txpower 30.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                0       0       0       0       0       0       0       0               0

```

nbd168, 2023-09-15:
Ah, I think I finally understand the issue. It seems that the ubus acl file needs to be updated, otherwise wpa_supplicant and hostapd don't have permission to talk to each other. I didn't see this issue myself, because I had ujail disabled in my config. Will make a fix soon, thanks for your patience.

nbd168, 2023-09-15:
Fix pushed

castillofrancodamian, 2023-09-19:
STA+AP already works with an automatic channel, and I didn't expect 40 and 80 MHz too. DFS and non-DFS channels. But other problems arose.

This happened to me on September 17th. I don't remember what I was doing at the time. The `eth1` device disappeared from the `br-wan` bridge and it takes the MAC of the `5g_invitados` interface. After this, I connected to an upstream AP with the 5GHz radio and the bridge was left without interfaces. I didn't realize to capture the logs.

[image]

[image]

[image]

Two days later a similar error appeared. This was when I was changing the channel on the 5 GHz radio.

[image]

[image]

```
root@archer_c60:~# logread
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 10.1.0.100 -- 10.1.0.249, lease time 5m
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq[1]: using nameserver 192.168.50.2#5300
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5053
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for test
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for local
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Mon Sep 18 13:16:40 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Mon Sep 18 13:16:41 2023 user.err ddns-scripts[28601]: duckdns: cURL Error: '6'
Mon Sep 18 13:16:41 2023 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Mon Sep 18 13:16:42 2023 user.warn ddns-scripts[28601]: duckdns: Transfer failed - retry 1/0 in 60 seconds
Mon Sep 18 13:16:44 2023 user.notice ddns-scripts[29259]: duckdns: PID '29259' started at 2023-09-18 13:16
Mon Sep 18 13:16:46 2023 user.notice ddns-scripts[28601]: duckdns: PID '28601' terminated by 'SIGTERM' at 2023-09-18 13:16
Mon Sep 18 13:16:47 2023 user.notice https-dns-proxy: Stopping service ✓
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: started, version 2.89 cachesize 10000
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: DNS service limited to local subnets
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 10.1.0.100 -- 10.1.0.249, lease time 5m
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: using nameserver 192.168.50.2#5300
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5053
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: using only locally-known addresses for test
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: using only locally-known addresses for local
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Mon Sep 18 13:16:47 2023 daemon.warn dnsmasq[1]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mon Sep 18 13:16:47 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Mon Sep 18 13:16:48 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Mon Sep 18 13:16:49 2023 user.notice firewall: Reloading firewall due to ifup of wg0 (wg0)
Mon Sep 18 14:28:45 2023 daemon.warn dnsmasq-dhcp[1]: no address range available for DHCP request via br-wan
Mon Sep 18 16:20:33 2023 kern.info kernel: [55292.351838] ath10k_pci 0000:00:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
Mon Sep 18 16:20:33 2023 kern.info kernel: [55292.366930] ath10k_pci 0000:00:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
Mon Sep 18 16:20:33 2023 daemon.notice hostapd: 5g_invitados: STA ca:f0:bf:96:b8:18 IEEE 802.11: did not acknowledge authentication response
Mon Sep 18 16:20:33 2023 daemon.notice hostapd: 5g_invitados: STA ca:f0:bf:96:b8:18 IEEE 802.11: did not acknowledge authentication response
Mon Sep 18 16:20:33 2023 kern.info kernel: [55292.386949] ath10k_pci 0000:00:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
Mon Sep 18 16:35:28 2023 daemon.err uhttpd[1511]: [info] luci: accepted login on / for root from 192.168.50.225
Mon Sep 18 18:01:20 2023 daemon.err uhttpd[1511]: [info] luci: accepted login on / for root from 192.168.50.225
[... cut ...]
Tue Sep 19 01:06:39 2023 daemon.err hostapd: Failed to set beacon parameters
Tue Sep 19 01:06:45 2023 daemon.err hostapd: Failed to set beacon parameters
Tue Sep 19 01:06:50 2023 authpriv.info dropbear[14359]: Child connection from 10.1.0.100:35898
Tue Sep 19 01:06:51 2023 daemon.err hostapd: Failed to set beacon parameters
Tue Sep 19 01:06:54 2023 authpriv.notice dropbear[14359]: Password auth succeeded for 'root' from 10.1.0.100:35898
Tue Sep 19 01:06:57 2023 daemon.err hostapd: Failed to set beacon parameters

```

brada4, 2023-09-20:
Duplicate MAC was rejected in kernel 4, it is not a problem nowadays. Must be another issue., not directly related to wifi one? Try ip lonk and brctl show if they are reflected correctly in luci to decide if it is cosmetic or technical issue.

castillofrancodamian, 2023-09-20:
@brada4 I didn't mention earlier that I couldn't access it from the bridge that was down. It is not a cosmetic error. The STA+AP thing that didn't work was solved, but this thing about MAC addresses appeared.

These are the logs that I managed to capture.

[image]

[image]

```
root@archer_c60:~# logread
Wed Sep 20 18:40:23 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Wed Sep 20 18:40:23 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Wed Sep 20 18:40:24 2023 daemon.notice hostapd: Set new config for phy phy1:
Wed Sep 20 18:40:24 2023 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 7 crashes, 65 seconds since last crash
Wed Sep 20 18:40:25 2023 daemon.notice hostapd: Remove interface '2g_lag'
Wed Sep 20 18:40:25 2023 daemon.notice hostapd: 2g_lag: interface state ENABLED->DISABLED
Wed Sep 20 18:40:25 2023 daemon.notice hostapd: 2g_f: AP-DISABLED
Wed Sep 20 18:40:25 2023 daemon.notice hostapd: 2g_f: CTRL-EVENT-TERMINATING
Wed Sep 20 18:40:25 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Wed Sep 20 18:40:25 2023 daemon.notice hostapd: 2g_invitados: AP-DISABLED
Wed Sep 20 18:40:25 2023 daemon.notice hostapd: 2g_invitados: CTRL-EVENT-TERMINATING
Wed Sep 20 18:40:25 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Wed Sep 20 18:40:25 2023 daemon.notice hostapd: 2g_lag: AP-DISABLED
Wed Sep 20 18:40:25 2023 daemon.notice hostapd: 2g_lag: CTRL-EVENT-TERMINATING
Wed Sep 20 18:40:25 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Wed Sep 20 18:40:25 2023 daemon.notice hostapd: nl80211: deinit ifname=2g_lag disabled_11b_rates=0
Wed Sep 20 18:40:25 2023 user.notice https-dns-proxy: Starting service instances ✓
Wed Sep 20 18:40:30 2023 daemon.notice wpa_supplicant[1276]: Set new config for phy phy1
Wed Sep 20 18:40:30 2023 daemon.notice wpa_supplicant[1276]: Remove interface 2g_sta
Wed Sep 20 18:40:30 2023 user.notice firewall: Reloading firewall due to ifup of wan (br-wan)
Wed Sep 20 18:40:33 2023 kern.info kernel: [ 1311.630877] 2g_sta: deauthenticating from 22:b8:2b:30:96:df by local choice (Reason: 3=DEAUTH_LEAVING)
Wed Sep 20 18:40:33 2023 daemon.notice wpa_supplicant[1276]: 2g_sta: CTRL-EVENT-DISCONNECTED bssid=22:b8:2b:30:96:df reason=3 locally_generated=1
Wed Sep 20 18:40:33 2023 daemon.notice wpa_supplicant[1276]: 2g_sta: CTRL-EVENT-DSCP-POLICY clear_all
Wed Sep 20 18:40:33 2023 daemon.notice wpa_supplicant[1276]: 2g_sta: CTRL-EVENT-DSCP-POLICY clear_all
Wed Sep 20 18:40:33 2023 daemon.notice wpa_supplicant[1276]: nl80211: deinit ifname=2g_sta disabled_11b_rates=0
Wed Sep 20 18:40:33 2023 daemon.err wpa_supplicant[1276]: rmdir[ctrl_interface=/var/run/wpa_supplicant]: Permission denied
Wed Sep 20 18:40:35 2023 daemon.notice wpa_supplicant[1276]: Set new config for phy phy0
Wed Sep 20 18:40:36 2023 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Wed Sep 20 18:40:36 2023 daemon.notice hostapd: Restart interface for phy phy0
Wed Sep 20 18:40:36 2023 kern.info kernel: [ 1315.163191] br-lan: port 2(5g_lag) entered blocking state
Wed Sep 20 18:40:36 2023 kern.info kernel: [ 1315.168839] br-lan: port 2(5g_lag) entered disabled state
Wed Sep 20 18:40:36 2023 kern.info kernel: [ 1315.174665] device 5g_lag entered promiscuous mode
Wed Sep 20 18:40:36 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=AR ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 noscan=1 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=-6 vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-2][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] channel=acs_survey  interface=5g_lag bssid=50:c7:bf:75:81:af ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa_passphrase=masterofpuppets wpa
Wed Sep 20 18:40:37 2023 daemon.notice wpa_supplicant[1276]: Set new config for phy phy1
Wed Sep 20 18:40:40 2023 user.notice firewall: Reloading firewall due to ifup of wg0 (wg0)
Wed Sep 20 18:40:41 2023 kern.warn kernel: [ 1319.546640] ath10k_pci 0000:00:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Wed Sep 20 18:40:41 2023 kern.warn kernel: [ 1319.554484] ath10k_pci 0000:00:00.0: msdu-desc: 2500  skid: 32
Wed Sep 20 18:40:41 2023 kern.info kernel: [ 1319.584724] ath10k_pci 0000:00:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Wed Sep 20 18:40:41 2023 kern.info kernel: [ 1319.596236] ath10k_pci 0000:00:00.0: wmi print 'free: 114572 iram: 12644 sram: 29508'
Wed Sep 20 18:40:41 2023 kern.info kernel: [ 1319.909031] ath10k_pci 0000:00:00.0: rts threshold -1
Wed Sep 20 18:40:41 2023 kern.warn kernel: [ 1319.918154] ath10k_pci 0000:00:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[... cut ...]
60: br-wan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 50:c7:bf:75:81:b0 brd ff:ff:ff:ff:ff:ff
root@archer_c60:~# brctl show
bridge name     bridge id               STP enabled     interfaces
br-lan          7fff.50c7bf7581b0       no              2g_f
                                                        eth0.1
br-wan          7fff.50c7bf7581b0       no              5g_invitados
                                                        eth1
                                                        2g_invitados
```

brada4, 2023-09-21:
Nothing apparent in logread in regard to bridge flopping.
Is it intntional that you bridge radio-s to WAN and not to LAN? i.e does your WAN support multiple DHCP clients at your home?

castillofrancodamian, 2023-09-21:
@brada4 This device is behind a main router. But at the same time it works as a simple AP. And the wireless interfaces are on the `br-wan` bridge so they are on the same upstream network. The script is to create a WISP router and have all wireless interfaces except the STA be on the `br-lan` bridge and the other script to reverse this. These problems appeared after the solutions for STA+AP, therefore the originators are `hostapd`, `netifd` or `mac80211`.

brada4, 2023-09-22:
There is nothing broken in your setup, no breakage shows in the logs either.
To shake the things somewhat
- mac802 - try setting fixed other channel on AP ipo auto - it will accept STA channel anyway
- sometimes vlan1 is silently consumed for some special use inside switches with basically anything above working fine (up to 16 or 64 in cheapest incarnations)

Neustradamus, 2024-11-03:
@castillofrancodamian: Have you solved your problem?

castillofrancodamian, 2024-11-05:
@Neustradamus I just retested this device with `SNAPSHOT r28018-6490c88c75` and I don't see the problem. STA+AP works fine with auto channel. With or without DFS.

closed 2024-11-05 by castillofrancodamian