openwrt/openwrt #13457: The inbound traffic statistics of the LAN interface are incorrect.

dongc123, 2023-09-15:
### Describe the bug

**The inbound traffic statistics of the LAN interface are incorrect.**
[image]

### OpenWrt version

OpenWrt 23.05.0-rc3 r23389-5deed175a5

### OpenWrt target/subtarget

x86/64

### Device

x86/64

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-15:
How is it incorrect? It is quite credible packets and megabytes numbers?

dongc123, 2023-09-15:
> 怎么会不正确呢？这是相当可信的数据包和兆字节数字吗？

[image]
[image]
The lan bridge has three network ports eth0 eth1 eht2. The inbound traffic statistics of eth2 are 26GB. The lan bridge shows that the inbound traffic is 91MB.

brada4, 2023-09-15:
What is in /sys/class/net/br-lan/statistics/{tx|rx}_{bytes|packets} ?

dongc123, 2023-09-15:
> /sys/class/net/br-lan/statistics/{tx|rx}_{bytes|packets} 中有什么？

[image]

brada4, 2023-09-15:
Luci reflects what kernel accounted?

dongc123, 2023-09-15:
Are you talking about a luci problem?

KanjiMonster, 2023-09-15:
Just an educated guess, but the rx/tx counters of `lan` is the amount of traffic sent to the bridge interface `br-lan`, so the OpenWrt router itself, and not the aggregated traffic over all bridge ports.

So packets that just get forwarded for example between `eth2` and `eth3` do not count. But if you access luci, then the request and the reply get counted.

brada4, 2023-09-15:
@KanjiMonster you are absolutely correct, the counters are for  bridge's own MAC+IP, excluding forwarded traffic.
If one disables all aggregating offloads in all network cards input/output sums match.

dongc123, 2023-09-15:
> 只是一个有根据的猜测，但 rx/tx 计数器`lan`是发送到桥接接口的流量量`br-lan`，因此是 OpenWrt 路由器本身，而不是所有桥接端口上的聚合流量。
> 
> `eth2`因此，仅在和之间转发的数据包`eth3`不计入在内。但如果您访问 luci，则请求和回复都会被计数。

I remember that the OpenWrt 23.05.0-rc2 version lan bridge tx rx statistics are correct

dongc123, 2023-09-15:
> @KanjiMonster你是完全正确的，计数器是针对网桥自己的MAC+IP，不包括转发的流量。 如果禁用所有网卡输入/输出总和匹配中的所有聚合卸载。

Then why are the statistics sent from the wan port and the inbound lan correct?

brada4, 2023-09-15:
If packet is offloaded in firewall then it is transported from one interface with cable to another, no intermediary counters involved.
If interface supports some bulk offloads like gro tso etc then 1 packet may match multiple physical frames.

KanjiMonster, 2023-09-15:
> > @KanjiMonster你是完全正确的，计数器是针对网桥自己的MAC+IP，不包括转发的流量。 如果禁用所有网卡输入/输出总和匹配中的所有聚合卸载。
> 
> Then why are the statistics sent from the wan port and the inbound lan correct?

I don't fully understand the question, likely the answer is layer 2 forwarding versus layer 3 routing.

So on br-lan between eth2 and eth3 the packets go directly between both ports and the br-lan interface in the kernel, and are never sent to the br-lan interface.

But for wan -> eth3 the packets are first received at the wan port, then the src and dst macs get changed to br-lan's mac and eth3's mac, and then the packet gets sent from br-lan to eth3.

So basically this is wan -> br-lan -> eth3.

KanjiMonster, 2023-09-15:
> > But for ... wan ...
> 
> ... nftables offload bypasses all that br-lan rituals.

But the question was why are they correct, not incorrect, so I assumed no offloading shenanigans involved there.

And obviously offloading will screw up counters in many ways. Which is why AFAICT there is support for explicit offloaded stats.

brada4, 2023-09-15:
`bridge vlan show -s` 
Shows some incrementing numbers with physical adapters and zeroes with dsa (no DSA in OP)

LGA1150, 2023-09-18:
Two reasons:

1. A bridge device does not account packets forwarded to another bridge port.
2. A bridge device is a virtual interface, which can be set up and torn down at runtime. When this happens, the software counter will be reset.