openwrt/openwrt #13458: ipv4 routing rules are invalid

dongc123, 2023-09-15:
### Describe the bug

The ipv4 routing rule is invalid. Please tell me how to configure and use this ipv4 routing rule.
[image]
[image]

### OpenWrt version

OpenWrt 23.05.0-rc3 r23389-5deed175a5

### OpenWrt target/subtarget

x86/64

### Device

x86/64

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dongc123, 2023-09-15:
The IP address 172 is for testing. By capturing packets with tcpdump, you can see that the data does not go out from the iptv port.

brada4, 2023-09-15:
`ip ro sh` (wipe 2 middle bytes of recognisable public IP addresses)

It is absolutely unlikely that iptv multicast  network will be globally routable as for the web browser.

Default routing table (last field) should be main (254) if you do not have any pbr like mwan enabled.

brada4, 2023-09-15:
First things first - lets learn to operate putty:
Run `ip ro sh`
Then select the output and sommand WITHOUT RIGHT CLICK
Now start text block with 3 backticks \`\`\`
Paste text that putty marked and copied.
Finish with 3 more backticks.

I would take exactly opposite approach - try to get route right with command line `ip route` then try to repeat experiment in luci.
Manual is here https://www.linux.org/docs/man8/ip-route.html , default `ip` is from busybox and it excludes some corner-case options, but those implemented follow the manual.

You understand iptv network uses IP addresses in its private range and probably filters all other addresses off.

brada4, 2023-09-15:
But show the routing table, more eyes see better like overlapping routes or not bound to interface etc.

LGA1150, 2023-09-21:
Do NOT set `传入接口名称`. Packets originating from your router are not from `lan`.

jow-, 2023-10-19:
User configuration issue

closed 2023-10-19 by jow-