openwrt/openwrt #13459: Wifi module rtl8822be not working

DragonGamees, 2023-09-15:
### Describe the bug

Hello, I have a home server (os: debian 12) and I installed openwrt x86_64 on qemu/kvm (Image: openwrt-22.03.5-x86-64-generic-ext4-combined.img). In the VM settings, I'm passing my wifi module (rtl8822be) as a PCI-e device.

I installed 2 packages in Openwrt. (rtl8822be-firmware, kmod-rtw88) And after the reboot, the wireless tab appeared in LuCi.

The network scan button works and displays my networks, but an error occurs when trying to connect in the kernel module.
This error also appears when trying to create an AP at any frequency and with any other settings.

Kernel message: 
```
[  217.651858] ------------[ cut here ]------------
[  217.654813] write RF mode table fail
[  217.657434] WARNING: CPU: 2 PID: 3705 at 0xffffffffa03df3cb [rtw88_8822b@00000000bf0fde9e+0x35000]
[  217.662078] Modules linked in: rtw88_8822ce rtw88_8822c rtw88_8822be rtw88_8822b rtw88_8723de rtw88_8723d pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet rtw88_pci rtw88_core pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 cfg80211 slhc r8169 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c igc forcedeth e1000e crc_ccitt compat bnx2 i2c_dev ixgbe e1000 amd_xgbe mdio vfat fat nls_utf8 nls_iso8859_1 nls_cp437 ena sha256_ssse3 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac igb button_hotplug tg3 ptp realtek pps_core mii
[  217.697086] CPU: 2 PID: 3705 Comm: iw Not tainted 5.10.176 #0
[  217.700336] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 1.16.2-debian-1.16.2-1 04/01/2014
[  217.704412] RIP: 0010:0xffffffffa03df3cb [rtw88_8822b@00000000bf0fde9e+0x35000]
[  217.707738] Code: 88 f5 40 a0 e8 36 91 61 e1 0f 0b 48 83 c4 18 5b 41 5c 41 5d 41 5e 41 5f 5d e9 01 15 a2 e1 48 c7 c7 1e f5 40 a0 e8 15 91 61 e1 <0f> 0b e9 55 fd ff ff 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 55
[  217.714876] RSP: 0018:ffffc9000040b910 EFLAGS: 00010282
[  217.717306] RAX: 0000000000000018 RBX: 00000000000aeaea RCX: 0000000000000027
[  217.720084] RDX: ffff88800f91ca28 RSI: 0000000000000001 RDI: ffff88800f91ca20
[  217.722799] RBP: ffffc9000040b950 R08: 0000000000000000 R09: ffffc9000040b6f0
[  217.725357] R10: ffffc9000040b6e8 R11: ffffffff824aac28 R12: ffff88800d9628c0
[  217.727860] R13: ffff88800d967188 R14: 0000000000000216 R15: 0000000000000000
[  217.730315] FS:  00007ff0cdfa9b48(0000) GS:ffff88800f900000(0000) knlGS:0000000000000000
[  217.732853] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  217.734965] CR2: 00007ff0cdf0f080 CR3: 0000000000226006 CR4: 0000000000170ee0
[  217.737232] Call Trace:
[  217.738683]  0xffffffffa03df943 [rtw88_8822b@00000000bf0fde9e+0x35000]
[  217.740865]  0xffffffffa0385ab9 [rtw88_core@00000000a8d26f0d+0x1a000]
[  217.742993]  0xffffffffa0279111 [mac80211@000000003cb4b3bd+0x7b000]
[  217.745028]  0xffffffffa0239092 [cfg80211@00000000a951c4b5+0x46000]
[  217.747046]  ? 0xffffffff81875059
[  217.748518]  0xffffffff818c8ee5
[  217.749949]  0xffffffff818c924c
[  217.751368]  ? 0xffffffffa0238de0 [cfg80211@00000000a951c4b5+0x46000]
[  217.753244]  ? 0xffffffff818c9180
[  217.754647]  0xffffffff818c8261
[  217.755961]  0xffffffff818c88e4
[  217.757269]  0xffffffff818c7a08
[  217.758576]  0xffffffff818c7ce0
[  217.759875]  0xffffffff8183f805
[  217.761177]  ? 0xffffffff818c7ad0
[  217.762500]  ? 0xffffffff813b972b
[  217.763767]  0xffffffff81840f94
[  217.764994]  ? 0xffffffff81207d3a
[  217.766247]  ? 0xffffffff811b3bd2
[  217.767446]  ? 0xffffffff811d691a
[  217.768649]  0xffffffff81841090
[  217.769810]  0xffffffff818410e8
[  217.770982]  0xffffffff81a3cea8
[  217.772140]  0xffffffff81c000a9
[  217.773278] RIP: 0033:0x7ff0cdf8836c
[  217.774479] Code: c3 8b 07 85 c0 75 24 49 89 fb 48 89 f0 48 89 d7 48 89 ce 4c 89 c2 4d 89 ca 4c 8b 44 24 08 4c 8b 4c 24 10 4c 89 5c 24 08 0f 05 <c3> e9 ff d0 ff ff 41 54 b8 02 00 00 00 49 89 f4 be 00 88 08 00 55
[  217.778518] RSP: 002b:00007ffdf39925a8 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
[  217.780383] RAX: ffffffffffffffda RBX: 00007ff0cdfa9b48 RCX: 00007ff0cdf8836c
[  217.782106] RDX: 0000000000000000 RSI: 00007ffdf39925f8 RDI: 0000000000000003
[  217.783967] RBP: 0000000000000003 R08: 0000000000000000 R09: 0000000000000000
[  217.785815] R10: 0000000000000000 R11: 0000000000000246 R12: 000000000000002e
[  217.787647] R13: 00007ff0cdf30de0 R14: 00007ff0cdfaaec0 R15: 00007ff0cdf25b10
[  217.789457] ---[ end trace 177712424e9eab4a ]---
```

Full log:
[dmesg.log](https://github.com/openwrt/openwrt/files/12621272/dmesg.log)

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009), (C612 and Xeon 2666v3, host os: Debian 12, QEMU/KVM)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-15:
qemu IRQ mapping heavily depends on how BIOS set them up
some report starting VM multiple times improves matters.
some get success in different PCIE slot, like device being alone on PCIE BUS, or on 0th PCIE BUS. 

First rmmod rtw88_pci (and dependent -be module)
Then reload  _pci module setting these parameters to 0
parm:           disable_msi:Set Y to disable MSI interrupt support (bool)
parm:           disable_aspm:Set Y to disable PCI ASPM support (bool)
Then modprobe parameterless -be module

Try all config, like restart wpad service.

Locate file in /etc/modules.d/ loading dependency rtw88_pci
Add working parameter combo after driver name
Add edited file name to /etc/sysupgrade.conf

DragonGamees, 2023-09-15:
I tried all the options for the parameters of the kernel modules and not one of them did not lead to a result. I also moved the wifi module to different slots both from the processor and the chipset

brada4, 2023-09-15:
Does it work when booting OpenWRT from USB stick on pure hardware? Especially pay attention to driver init messages. 
Say in VM bus 0a does not get BAR allocated, pay attention if same number and sizes of  memory mappings are allocated to device and the interrupt mode.

brada4, 2023-09-15:
There is a message one of wired adapters does not get to some optional config space. It may help to boot EFI (no secureboot) guest.

DragonGamees, 2023-09-15:
> Does it work when booting OpenWRT from USB stick on pure hardware? Especially pay attention to driver init messages.
> Say in VM bus 0a does not get BAR allocated, pay attention if same number and sizes of memory mappings are allocated to device and the interrupt mode.

I tried to download directly from a usb flash drive, the problem still remains
[dmesg_usb.log](https://github.com/openwrt/openwrt/files/12623527/dmesg_usb.log)

brada4, 2023-09-15:
https://github.com/lwfinger/rtw88/issues/51#issuecomment-823361690

brada4, 2023-09-15:
Bios update from vendor advised there (the tsc part)

DragonGamees, 2023-09-16:
> [lwfinger/rtw88#51 (comment)](https://github.com/lwfinger/rtw88/issues/51#issuecomment-823361690)

In a file along the path /etc/modules.d/rtw88 with the original content:
```
rtw88_8723de
rtw88_8822be
rtw88_8822ce
```

I changed it to
```
option rtw88_pci disable_msi=y disable_aspm=y
rtw88_8822be
```
there have also been attempts with
```
option rtw88_pci disable_msi=1 disable_aspm=1
rtw88_8822be
```

And this again did not lead to results.

I also tried connecting the wifi module to the motherboard with b550 and 5800, which also did not change anything.
[dmesg_b550_5800.log](https://github.com/openwrt/openwrt/files/12640473/dmesg_b550_5800.log)

brada4, 2023-09-16:
Debian instructions do not irradiate kvm guest systems.

It has to be without `option` a the first line in that file.
i.e:
```
rtw88_pci disable_msi=1 disable_aspm=1
rtw88_8822be
...
```

brada4, 2023-09-16:
Sometimes module options are dynamically changeable by writing to supposedly
`echo 1 > /sys/module/rtw88_pci/parameters/disable_aspm`

brada4, 2023-09-16:
chipset supports x2apic which may help with much more msi interrupts. Check for same text in debian? 
Basically 'auto' msi means enabling it if mainboard claims to support them and particular wifi chipset is msi capable.

DragonGamees, 2023-09-16:
> Debian instructions do not irradiate kvm guest systems.
> 
> It has to be without `option` & the first line in that file. i.e:
> 
> ```
> rtw88_pci disable_msi=1 disable_aspm=1
> rtw88_8822be
> ...
> ```

> Sometimes module options are dynamically changeable by writing to supposedly `echo 1 > /sys/module/rtw88_pci/parameters/disable_aspm`

Both options did not lead to results

> chipset supports x2apic which may help with much more msi interrupts. Check for same text in debian? Basically 'auto' msi means enabling it if mainboard claims to support them and particular wifi chipset is msi capable.

Latest tests I load OpenWrt directly from a USB flash drive

I also disabled/enabled msi on pcie ports in bios, which also didn’t change anything

brada4, 2023-09-16:
`cat /proc/interrupts` should confirm that wifi irq is not MSI. It cannot be that card can not produce any interrupt.

Does card work in debian? I think you need some firmware from sa,e repo as intel microcode.

DragonGamees, 2023-09-16:
> `cat /proc/interrupts` should confirm that wifi irq is not MSI. It cannot be that card can not produce any interrupt.

[interrupts.txt](https://github.com/openwrt/openwrt/files/12641033/interrupts.txt)

> Does card work in debian? I think you need some firmware from sa,e repo as intel microcode.

Yes, hostapd creates a point and I can connect to it 

```
root@server:~# hostapd wpa.conf
wlp10s0: interface state UNINITIALIZED->HT_SCAN
20/40 MHz operation not permitted on channel pri=6 sec=10 based on overlapping BSSes
Fallback to 20 MHz
wlp10s0: interface state HT_SCAN->ENABLED
wlp10s0: AP-ENABLED
wlp10s0: STA 6e:c4:ec:87:9a:90 IEEE 802.11: authenticated
wlp10s0: STA 6e:c4:ec:87:9a:90 IEEE 802.11: associated (aid 1)
wlp10s0: AP-STA-CONNECTED 6e:c4:ec:87:9a:90
wlp10s0: STA 6e:c4:ec:87:9a:90 RADIUS: starting accounting session F655F961787D4143
wlp10s0: STA 6e:c4:ec:87:9a:90 WPA: pairwise key handshake completed (RSN)
wlp10s0: EAPOL-4WAY-HS-COMPLETED 6e:c4:ec:87:9a:90
```

I also compared the hash of the firmware files on debian and on OpenWrt, they are the same

brada4, 2023-09-16:
parameter did not work out?
[image]
KVM - thats debian or wrt?

brada4, 2023-09-16:
If debian works without any additional configuration - try shortly 23.05 or snapshot openwrt. Maybe respective quirks for mainboard are applied.

DragonGamees, 2023-09-16:
> KVM - thats debian or wrt? 

Debian host, openwrt on virtual machine, interrupts from openwrt.

> If debian works without any additional configuration - try shortly 23.05 or snapshot openwrt. Maybe respective quirks for mainboard are applied.

Then I'll wait for the update or try to rebuild the kernel module myself. 
Thanks for the help.

brada4, 2023-09-16:
The intent of module parameters was to fprce wifi interrupt to be ioapic... Kvm  os very sensitive to minor bios glitches.