openwrt/openwrt #13460: ath9k: failed (err=-2) to del object (id=3)

M95D, 2023-09-15:
### Describe the bug

The device is a Linksys WRT1900AC v1. It is modded. I replaced the mwlwifi with ath9k: AR9380 for 5GHz and AR9285 for 2.4GHz. Details here: https://forum.openwrt.org/t/157786/

I built an image with the new testing kernel 6.1 for mvebu. I get these errors in the log:
```
Sat Sep 16 02:16:17 2023 kern.err kernel: [  422.874332][ T8458] ath9k 0000:03:00.0 wlan0: failed (err=-2) to del object (id=3)
Sat Sep 16 02:16:17 2023 kern.err kernel: [  422.885493][ T8458] ath9k 0000:03:00.0 wlan0: failed (err=-2) to del object (id=3)
Sat Sep 16 02:16:17 2023 kern.err kernel: [  422.896996][ T8458] ath9k 0000:03:00.0 wlan0: failed (err=-2) to del object (id=3)
Sat Sep 16 02:16:17 2023 kern.err kernel: [  422.908295][ T8458] ath9k 0000:03:00.0 wlan0: failed (err=-2) to del object (id=3)
Sat Sep 16 02:16:17 2023 kern.err kernel: [  422.919572][ T8458] ath9k 0000:03:00.0 wlan0: failed (err=-2) to del object (id=3)
Sat Sep 16 02:16:17 2023 kern.err kernel: [  422.930842][ T8458] ath9k 0000:03:00.0 wlan0: failed (err=-2) to del object (id=3)
Sat Sep 16 02:16:17 2023 kern.err kernel: [  422.941734][ T8458] ath9k 0000:03:00.0 wlan0: failed (err=-2) to del object (id=3)
Sat Sep 16 02:16:38 2023 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 70:03:9f:63:df:7a
```

I also saw another (possibly related) error on the other card during a wifi restart commanded from Luci:
`daemon.notice netifd: radio1 (1517): WARNING: Variable 'data' does not exist or is not an array/object`

Wifi appears to be working, but these errors remind me of the old key removal error I had with mwlwifi. I think it's a bug.
Can anyone confirm?

### OpenWrt version

r23885

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1900AC v1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-16:
What is in the logread right  before error chain?

M95D, 2023-09-17:
Nothing interesting. There are a couple of minutes of normal operation before these messages, then it continues normal operation for at least one hour with no other errors. I didn't test longer than that.

brada4, 2023-09-17:
Like disassociation that mandates per-sta queue cleanup?

M95D, 2023-09-17:
I don't think so. The clients are ESP8266 smart-plugs. They are always on.

M95D, 2023-09-17:
The error comes actually from switchdev driver:
```
$ grep -R "to del object"
net/switchdev/switchdev.c:              netdev_err(dev, "failed (err=%d) to del object (id=%d)\n",
```

closed 2024-12-04 by M95D