openwrt/openwrt #13461: umount error during sysupgrade

M95D, 2023-09-16:
### Describe the bug

While performing sysupgrade, after writing the new image, but before the reboot, I noticed an umount command that outputs usage information. There is probably something missing or unsupported by busybox in that command.

Since this error comes after writing the new image, when Luci and ssh are already disconnected, it can only be viewed if a serial is used.
Sysupgrade is a success, but this umount might be needed to prevent filesystem corruption.

```
Unlocking kernel2 ....
.
Writing from <stdin> to kernel2 ...     .
sysupgrade successful.
BusyBox v1.36.1 (2023-07-22 19:23:53 UTC) multi-call binary..
.
Usage: umount [-rlfd] [-t FSTYPE] FILESYSTEM|DIRECTORY.
.
Unmount filesystems.
.
        -r      Remount devices read-only if mount is busy.
        -l      Lazy umount (detach filesystem).
        -f      Force umount (i.e., unreachable NFS server).
        -d      Free loop device if it has been used.
        -t FSTYPE[,...] Unmount only these filesystem type(s).
[  276.722007][ T6303] sd 0:1:0:0: [sdd] Synchronizing SCSI cache.
[  276.728045][ T6303] sd 0:0:0:0: [sdc] Synchronizing SCSI cache.
---- more lines ----
[  277.172487][ T6303] reboot: Restarting system.
```

PS: There are some extra dots that are scattered over the log. They came from the router.

### OpenWrt version

r23611

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1900AC v1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2023-09-16:
Probably a result of your other busybox issues. `umount -a` is called after the success message. Your output shows the -a parameter missing from the usage help.

It depends on `BUSYBOX_DEFAULT_FEATURE_UMOUNT_ALL`. Was this in your build config?

M95D, 2023-09-16:
You are correct. My mistake.

closed 2023-09-16 by M95D