openwrt/openwrt #13462: Build error mac80221 with backports 6.5

InBomBi, 2023-09-16:
### Describe the bug

I got this error when i build image for Xiaomi Router R3. Please guide me how to fix

`<2023-09-14T13:05:37.7878548Z```       |                             ^~~~~~~~~~~~~~~~~~~ 2023-09-14T13:05:37.7879165Z /workdir/xwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/backports-6.5/net/mac80211/drop.h:32:27: note: in expansion of macro 'SKB_NOT_DROPPED_YET' 2023-09-14T13:05:37.7880076Z    32 |         ___RX_QUEUED    = SKB_NOT_DROPPED_YET, 2023-09-14T13:05:37.7880363Z       |                           ^~~~~~~~~~~~~~~~~~~ 2023-09-14T13:05:37.8922299Z make[9]: *** [scripts/Makefile.build:286: /workdir/xwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/backports-6.5/net/mac80211/main.o] Error 1 2023-09-14T13:05:37.8923430Z make[8]: *** [scripts/Makefile.build:503: /workdir/xwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/backports-6.5/net/mac80211] Error 2 2023-09-14T13:05:37.8925259Z make[7]: *** [Makefile:1829: /workdir/xwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/backports-6.5] Error 2 2023-09-14T13:05:37.8927677Z make[6]: *** [Makefile.build:13: modules] Error 2 2023-09-14T13:05:37.8928246Z make[5]: *** [Makefile.real:93: modules] Error 2 2023-09-14T13:05:37.8931820Z make[4]: Leaving directory '/workdir/xwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/backports-6.5' 2023-09-14T13:05:37.8933059Z make[4]: *** [Makefile:121: modules] Error 2 2023-09-14T13:05:37.8933548Z make[3]: Leaving directory '/workdir/xwrt/package/kernel/mac80211' 2023-09-14T13:05:37.8941134Z make[3]: *** [Makefile:401: /workdir/xwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/backports-6.5/.built] Error 2 2023-09-14T13:05:37.8941607Z time: package/kernel/mac80211/regular/compile#1.72#0.31#1.93`>

### OpenWrt version

22.03

### OpenWrt target/subtarget

Ramips mt7620

### Device

xiaomi miwifi r3

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

castiel652, 2023-09-16:
xwrt?

robimarko, 2023-09-18:
Completely irrelevant for OpenWrt, again unsupported fork.
Even the terms list that and its not ticked.

closed 2023-09-18 by ynezz