openwrt/openwrt #13466: Kernel warning due to mac80211 on Netgear WAC124 in AP+Mesh configuration

gca31, 2023-09-16:
### Describe the bug

Fri Sep 15 21:53:56 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:53:57 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:53:58 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:53:59 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:00 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:01 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:02 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:03 2023 daemon.notice mesh11sd[2162]: m-11s-0 is not up - giving up for now.
Fri Sep 15 21:54:03 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:04 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:05 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:06 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:07 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:08 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:09 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:10 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:11 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:12 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:13 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:14 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:15 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:16 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Fri Sep 15 21:54:17 2023 daemon.notice hostapd: phy1-ap0: DFS-CAC-COMPLETED success=1 freq=5500 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5530 cf2=0
Fri Sep 15 21:54:18 2023 kern.info kernel: [  111.234463] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
Fri Sep 15 21:54:18 2023 kern.info kernel: [  111.247999] br-lan: port 6(phy1-ap0) entered blocking state
Fri Sep 15 21:54:18 2023 kern.info kernel: [  111.259153] br-lan: port 6(phy1-ap0) entered forwarding state
Fri Sep 15 21:54:18 2023 daemon.notice netifd: Network device 'phy1-ap0' link is up
Fri Sep 15 21:54:18 2023 daemon.notice hostapd: phy1-ap0: interface state DFS->ENABLED
Fri Sep 15 21:54:18 2023 daemon.notice hostapd: phy1-ap0: AP-ENABLED
Fri Sep 15 21:54:23 2023 daemon.notice mesh11sd[2162]: m-11s-0 is not up - giving up for now.
Fri Sep 15 21:54:28 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: interface state UNINITIALIZED->COUNTRY_UPDATE
Fri Sep 15 21:54:28 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: interface state COUNTRY_UPDATE->ENABLED
Fri Sep 15 21:54:28 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: AP-ENABLED
Fri Sep 15 21:54:28 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: joining mesh Nuts
Fri Sep 15 21:54:28 2023 daemon.notice netifd: Network device 'm-11s-0' link is up
Fri Sep 15 21:54:28 2023 kern.info kernel: [  121.633653] IPv6: ADDRCONF(NETDEV_CHANGE): m-11s-0: link becomes ready
Fri Sep 15 21:54:28 2023 kern.info kernel: [  121.647099] br-lan: port 7(m-11s-0) entered blocking state
Fri Sep 15 21:54:28 2023 kern.info kernel: [  121.658113] br-lan: port 7(m-11s-0) entered forwarding state
Fri Sep 15 21:54:28 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: CTRL-EVENT-CONNECTED - Connection to 00:00:00:00:00:00 completed [id=1 id_str=]
Fri Sep 15 21:54:28 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: MESH-GROUP-STARTED ssid="Nuts" id=1
Fri Sep 15 21:54:28 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: new peer notification for d2:xx:xx:xx:xx:xx
Fri Sep 15 21:54:28 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: new peer notification for 02:yy:yy:yy:yy:yy
Fri Sep 15 21:54:28 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: new peer notification for d2:xx:xx:xx:xx:xx
Fri Sep 15 21:54:29 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: new peer notification for d2:xx:xx:xx:xx:xx
Fri Sep 15 21:54:29 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: mesh plink with 02:yy:yy:yy:yy:yy established
Fri Sep 15 21:54:29 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: MESH-PEER-CONNECTED 02:yy:yy:yy:yy:yy
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.302522] ------------[ cut here ]------------
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.311848] WARNING: CPU: 0 PID: 376 at backports-6.5/net/mac80211/main.c:283 ieee80211_link_info_change_notify+0x14c/0x15c [mac80211]
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.336396] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7615e mt7615_common mt7603e mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat ledtrig_usbport sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 crypto_hw_eip93 hmac des_generic libdes cmac authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore nls_base usb_common crc32c_generic
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.490748] CPU: 0 PID: 376 Comm: kworker/0:2 Not tainted 5.15.131 #0
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.503612] Workqueue: events cfg80211_stop_iface [cfg80211]
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.514993] Stack : 808b0000 80083cec 00000000 00000004 807d43f0 81847cc4 8185e480 8004b710
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.531641]         808b0000 81847cfb 00000001 81847b10 83080d40 00000001 81847cd0 80c455c0
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.548285]         00000000 00000000 807d43f0 81847b10 ffffefff 00000000 ffffffea 00000000
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.564933]         81847b1c 00000178 808b6a08 ffffffff 807d43f0 00000001 00000000 832025b8
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.581592]         00000009 00000000 81005c05 8185e4c0 00000018 804135d0 00000000 80a70000
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.598246]         ...
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.603103] Call Trace:
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.607951] [<80007db4>] show_stack+0x28/0xf0
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.616640] [<8038511c>] dump_stack_lvl+0x60/0x80
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.626006] [<8002df44>] __warn+0x9c/0x124
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.634165] [<8002e028>] warn_slowpath_fmt+0x5c/0xac
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.644053] [<832025b8>] ieee80211_link_info_change_notify+0x14c/0x15c [mac80211]
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.659166] [<83276848>] ieee80211_mesh_work+0x1a8/0x30c [mac80211]
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.671736] [<83080e00>] cfg80211_stop_iface+0x1bc/0x208 [cfg80211]
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.684288]
Fri Sep 15 21:54:29 2023 kern.warn kernel: [  122.687672] ---[ end trace 21fe5193619b6a4b ]---
Fri Sep 15 21:54:29 2023 daemon.notice wpa_supplicant[1518]: m-11s-0: new peer notification for d2:xx:xx:xx:xx:xx
[... 15 lines cut ...]
### Image kind

Official downloaded image

### Steps to reproduce

Install the snapshot and configure AP+Mesh
Kernel warning should occur at the end of the startup sequence.

### Actual behaviour

At the end of the startup sequence, got this kernel warning.
Don't know yet what is the consequence as the router in AP+Mesh seems to work as expected.

### Expected behaviour

no kernel warning

### Additional info

Occurs also without mesh11sd package

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-16:
Please attach full logread and try to mask just 2 octets in MAC  to let ppl see they are unique
10:23:45:XX:XX:AB

gca31, 2023-09-16:
Full logread in this file:
[wac124.txt](https://github.com/openwrt/openwrt/files/12641405/wac124.txt)

brada4, 2023-09-16:
The weird thing is that radio goes up, then goes down and repeats ACS/DFS routine.
Can you trigger the bug with selectively booting with mesh or ap up and then starting other when DFS/ACS is complete?

nbd168, 2023-09-16:
Are you sure that you posted the right version number of the image? I'm asking because several things in that log look like symptoms of bugs that I already fixed. Can you please also show me your /etc/config/wireless, so that I can try to reproduce this?

gca31, 2023-09-16:
> Are you sure that you posted the right version number of the image? I'm asking because several things in that log look like symptoms of bugs that I already fixed. Can you please also show me your /etc/config/wireless, so that I can try to reproduce this?

Yep. The firmware was generated with openwork firmware selector (https://firmware-selector.openwrt.org) at 1:35 pm on Sept 16, French Time (UTC+2 by now) with 3 additional packages (iperf3, luci and luci-app-attendedsysupgraded)

gca31, 2023-09-16:
> The weird thing is that radio goes up, then goes down and repeats ACS/DFS routine.
> Can you trigger the bug with selectively booting with mesh or ap up and then starting other when DFS/ACS is complete?

I can trigger this issue with mesh only enabled.

gca31, 2023-09-16:
> Are you sure that you posted the right version number of the image? I'm asking because several things in that log look like symptoms of bugs that I already fixed. Can you please also show me your /etc/config/wireless, so that I can try to reproduce this?

Forgot to say the package wpad-mbetls was changed to wpad-openssl for the firmware generation and mesh11sd was installed after firmware installation on the netgear WAC124
Here is /etc/config/wireless with ssid(s), key(s) and domain changed

config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option channel '11'
	option band '2g'
	option htmode 'HT20'
	option country 'FR'
	option cell_density '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'Aaaaaaaa'
	option encryption 'psk2'
	option key 'AaaaaaaaPasswd'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option channel '100'
	option band '5g'
	option htmode 'VHT80'
	option country 'FR'
	option cell_density '0'

config wifi-iface 'wifinet2'
	option device 'radio1'
	option mode 'mesh'
	option network 'lan'
	option encryption 'sae'
	option key 'MeshBbbbPasswd'
	option mesh_id 'Bbbb'
	option mesh_fwding '1'
	option mesh_rssi_threshold '0'

config wifi-iface 'wifinet3'
	option device 'radio1'
	option mode 'ap'
	option ssid 'Aaaaaaaa'
	option encryption 'psk2'
	option key 'AaaaaaaaPasswd'
	option ieee80211r '1'
	option mobility_domain '1234'
	option ft_over_ds '0'
	option ft_psk_generate_local '1'
	option network 'lan'
	option ieee80211k '1'
	option time_advertisement '2'
	option time_zone 'CET-1CEST,M3.5.0,M10.5.0/3'

gca31, 2023-09-16:
I did some more testing and here is what I found with the same configuration used. Hope this could help:

If the WAC124 is booted with Mesh and AP enabled, throughput is just awful (from 10Kbps to 5Mbps from a smartphone)
[image]

If the WAC124 is booted with only Mesh enabled, then once the mesh link is established, AP is enabled, then throughput is very good (reaching over 200Mbps from the same smartphone).
[image]

nbd168, 2023-09-17:
I think there may be some issues with attendedsysupgrade at the moment - it might be mixing packages from different versions, pulling in older code into newer images.

gca31, 2023-09-17:
> I think there may be some issues with attendedsysupgrade at the moment - it might be mixing packages from different versions, pulling in older code into newer images.

I rebuild a firmware from openwork firmware selector today. And I still have the issue (kernel warning and throughput issue)
So I build my own firmware with a dev platform, kernel warning is still there, but throughput issue has vanished.

So yes, there might be something with attendedsysupgrade at the moment. Therefore I'll assume I have to try later in some days.

But what about the kernel warning which is still there ?

gca31, 2023-09-22:
I reopen this issue as I found out what may be the consequence of this kernel warning:
The AP linked to the same radio providing the Mesh is in disabled state and it seems impossible to enable it (see below)
[image]

If I click the "Disable button", the following messages appear in the system log and the enable button is now available
[image]

Then enabling the AP crashes the radio and a reboot is mandatory

brada4, 2023-09-22:
You can (try to) enable meshpoint in /etc/config/wireless (disabled = '0') and restart wifi radio here.
Meshpoint needs fixed radio channel matching mesh. That will override anything else on that radio hopefully.

gca31, 2023-10-30:
Changed WAC124 to AP only

closed 2023-10-30 by gca31