openwrt/openwrt #13487: [23.05-rc3] upgrading with auc fails

solsticedhiver, 2023-09-19:
### Describe the bug

When attempting to upgrade from 22.03 to 23.05-rc3 with `auc -b 23.05`, one gets that following error:
```
Are you sure you want to continue the upgrade process? [N/y] Y
Requesting build.....Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.0-rc3/targets/ath79/generic/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.0-rc3/targets/ath79/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0-rc3/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.0-rc3/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0-rc3/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.0-rc3/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0-rc3/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.0-rc3/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0-rc3/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.0-rc3/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0-rc3/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.0-rc3/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /home/aparcar/asu/worker1/cache/23.05.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
Error: Impossible package selection
Bad message (77)
```

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ath79/generic

### Device

D-link DAP-2695-A1

### Image kind

Official downloaded image

### Steps to reproduce

1. install 22.03.5
2. install luci-ssl (might not be needed)
3. run auc -b 23.05

### Actual behaviour

Error posted above

### Expected behaviour

No error, and upgrade completed

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-19:
Expected behaviour: user reads release notes and removes wolfssl packages from submitted list via web ui and gets mbedtls replacements instead.

brada4, 2023-09-22:
Ping @aparcar @hauke

solsticedhiver, 2023-09-23:
@brada4 I tried and this is not trivial
I removed libusream-wolfssl, and I ended up with a non functionnal auc because it missed ca-bundle certificates and couldn't load any new packages.

I had to download packages from a browser and upload it to my openwrt device. It was ca-bundle, ca-certificates, libustream-mebdtls and may be some other mebdtls package I don't remember the names right now.

I was able to get out of this sh.. situation and upgrade to 23.05 with auc but this is not user-friendly, not even admin friendly

👎

brada4, 2023-09-23:
I somehow succeeded removing most and renaming those obvious leaf packages like wpad. Thats why called auc bosses if they can help.

aparcar, 2023-09-23:
@dangowrt do you have an idea how to solve this dependency tracking once and for all? I'm happy to implement whatever but ideally it doesn't involve opkg

qral, 2024-02-27:
> Expected behaviour: user reads release notes and removes wolfssl packages from submitted list via web ui and gets mbedtls replacements instead.

https://openwrt.org/releases/23.05/notes-23.05.0 is saying only that there was a switch from woflssl to mbedtls, but nothing about how to correctly update. If I manually uninstall wolfssl I will not be able to download mbedtls packages because of cert issues. So it's really not nice. 
But manual upload of sysimage worked so I am happy.

brada4, 2024-02-27:
@qral hope some day auc server can substitute packages, no doubt somebody will complain eithet way.
 for now it is trial and error or restoring conf on clean install.